adafruit/circuitpython #8903: Broken link to UF2 binary on CircuitPython.org

henrygab, 2024-02-11:
### CircuitPython version

```python
N/A
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

### Description

If this is not the right depot, please redirect or inform where to file this report.

Reviewing new adabox21, which is based on Momento.  The 

Starting at [CircuitPython.org](https://circuitpython.org/board/adafruit_esp32s3_camera/) website for the board.

Near the bottom is the link to [download](https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-adafruit_esp32s3_camera-0.18.1.zip) the UF2 binary.

However, that link is incorrect.  It is currently:

`https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-adafruit_esp32s3_camera-0.18.1.zip`

Where it should be:

`https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-adafruit_camera_esp32s3-0.18.1.zip`

In particular, the relative ordering between `camera` and `esp32s3` differs between the build system and the website.

jepler, 2024-02-12:
(I modified the initial message slightly to avoid spoilers)

RetiredWizard, 2024-02-12:
The link is no longer broken, I believe this can be closed :grin:

closed 2024-02-12 by makermelissa

henrygab, 2024-02-12:
I appreciate the spoiler-avoidance.  I'm not sure what was not public.   A simple web search for the next numerical version gives multiple pages of information.   Also, emails to subscribers already went out, with shipping information.

I will defer to your greater knowledge.  Perhaps it was because I said only "reviewing" instead of "reviewing instructions for"?

Either way, thanks for the quick fix!