adafruit/circuitpython #8907: Repeated Socket Failures with recent builds

jerryneedell, 2024-02-12:
### CircuitPython version

```python
This issue first occurs with this build and is present on all later builds.
Adafruit CircuitPython 9.0.0-beta.0-19-g7ba62bb1cb on 2024-02-07; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3

which was from this commit
https://github.com/adafruit/circuitpython/commit/7ba62bb1cb6a52eec7d91d4b48ce3063d0502de7
```

### Code/REPL

```python
import binascii
import io
import busio
import os
import ssl
import time
import board
import socketpool
import wifi
import adafruit_requests
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError

aio_username = os.getenv('AIO_USERNAME')
aio_key = os.getenv('AIO_KEY')
i2c_bus=busio.I2C(scl=board.SCL,sda=board.SDA)

pool = socketpool.SocketPool(wifi.radio)

requests = adafruit_requests.Session(pool, ssl.create_default_context())

# Initialize an Adafruit IO HTTP API object
io = IO_HTTP(os.getenv("AIO_USERNAME"), os.getenv("AIO_KEY"), requests)

# Adafruit IO feed configuration
try:
    # Get the 'camera' feed from Adafruit IO
    feed_camera = io.get_feed("esp32s3eye")
except AdafruitIO_RequestError:
    # If no 'camera' feed exists, create one
    feed_camera = io.create_new_feed("esp32s3eye")
print("got feeds")
```

### Behavior

Adafruit CircuitPython 9.0.0-beta.0-19-g7ba62bb1cb on 2024-02-07; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
>>> import aio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "aio.py", line 29, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 716, in request
OutOfRetries: Repeated socket failures
>>> 

### Description

The failure occurs on this line:
```    feed_camera = io.get_feed("esp32s3eye")```

The test code works for earlier builds but fails for any after this time.
There don't appear to be any corresponding changes to the adafruit_io or adafruit_requests libraries.

Also tested with same results on a memento (adafruit_esp32s3_camera) and espressif_esp32s3_eye

jerryneedell, 2024-02-12:
I also tried using a feather_esp32s2 and it fails the same way with the build reported above, but it also fails with memory errors for the previous build...
```

10.0.0.107 | REPL | 9.0.0-beta.0-19-g7ba62bb1cbreload.
Adafruit CircuitPython 9.0.0-beta.0-19-g7ba62bb1cb on 2024-02-07; Adafruit Feather ESP32S2 with ESP32S2
>>> 
>>> import aio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "aio.py", line 29, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 716, in request
OutOfRetries: Repeated socket failures
>>> 
[05:55:18.187] Disconnected
[05:55:19.187] Warning: Could not open tty device (No such file or directory)
[05:55:19.187] Waiting for tty device..
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

10.0.0.107 | REPL | 9.0.0-beta.0-18-g09fcc5997freload.
Adafruit CircuitPython 9.0.0-beta.0-18-g09fcc5997f on 2024-02-07; Adafruit Feather ESP32S2 with ESP32S2
>>> import aio
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
espidf.MemoryError: 

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "aio.py", line 29, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 691, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed
>>> 

```

jerryneedell, 2024-02-12:
The feather esp32s3 NoPSRAM behaves the same as the esp32s3 with PSRAM above.

```

10.0.0.202 | REPL | 9.0.0-beta.0-19-g7ba62bb1cbreload.
Adafruit CircuitPython 9.0.0-beta.0-19-g7ba62bb1cb on 2024-02-07; Adafruit Feather ESP32S3 No PSRAM with ESP32S3
>>> import aio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "aio.py", line 29, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 716, in request
OutOfRetries: Repeated socket failures
>>> 

10.0.0.202 | REPL | 9.0.0-beta.0-17-gcc235061efreload.
Adafruit CircuitPython 9.0.0-beta.0-17-gcc235061ef on 2024-02-07; Adafruit Feather ESP32S3 No PSRAM with ESP32S3
>>> import aio
got feeds
>>> 

```

jerryneedell, 2024-02-12:
Just to clarify -- there is also a build 09ffc59 that is not available for all boards.
It works OK with the AIO example on the esp32s3_eye

The first build that fails with the "repeated requests" for all boards is 7ba62bb

On the esp32s2  -- the ESPIDF memory error appears to be a different issue, but it also fails with the requests error for the newer builds....

sigh...

justmobilize, 2024-02-12:
Tweaked test code:

```py
import binascii
import io
import busio
import os
import ssl
import time
import board
import socketpool
import wifi
import adafruit_requests
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError

ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")
aio_username = os.getenv('AIO_USERNAME')
aio_key = os.getenv('AIO_KEY')

print("Connecting to WiFi...")
while not wifi.radio.connected:
  wifi.radio.connect(ssid, password)

pool = socketpool.SocketPool(wifi.radio)
ssl_context = ssl.create_default_context()

requests = adafruit_requests.Session(pool, ssl_context)

#io = IO_HTTP(aio_username, aio_key, requests)

try:
    feed = io.get_feed("test")
except AdafruitIO_RequestError:
    feed = io.create_new_feed("test")

print("got feed")

urls = [
  "http://wifitest.adafruit.com/testwifi/index.html",
  "https://www.adafruit.com/api/quotes.php",
  "https://api.github.com",
]

for url in urls:
  print(url)
  r = requests.get(url)
  r.close()

print("Done!")
```

Tested on a MagTag - `8.2.9` works fine, but `9.0.0-beta.0` fails with `OSError: Failed SSL handshake`:

```
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit MagTag with ESP32S2
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WiFi...
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 30, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 691, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Strangely on a `MEMENTO` running `9.0.0-beta.0` I the only URL I get a handshake error is `https://api.github.com` (and also made sure to re-test it being the first URL called...

```
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit Camera with ESP32S3
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WiFi...
got feed
http://wifitest.adafruit.com/testwifi/index.html
https://www.adafruit.com/api/quotes.php
https://api.github.com
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 45, in <module>
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 691, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

jerryneedell, 2024-02-12:
It looks like you are running your tests on the "official 9.0-beta.0" release, not on the latest builds in S3 -- is that correct?

justmobilize, 2024-02-12:
It was. Testing with `Adafruit CircuitPython 9.0.0-beta.0-24-g52c3cf6448 on 2024-02-09`:

`MagTag` - worked fine
`Adafruit Camera with ESP32S3` errored in the same spot, same way
`FeatherS3 with ESP32S3` errored the same way as the `MEMENTO`

justmobilize, 2024-02-12:
Extra info, added some logging. This is on a `FeatherS3 - ESP32-S3 Development Board by Unexpected Maker`

```
Adafruit CircuitPython 9.0.0-beta.0-24-g52c3cf6448 on 2024-02-09; FeatherS3 with ESP32S3
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WiFi...
https://api.github.com
_get_socket: start
_get_socket: try: 1
_get_socket: set timeout: 300
_get_socket: socket.connect - OSError: Failed SSL handshake
_get_socket: try: 2
Sending request failed
Time: 10.605 seconds
Done!
```
So it's the call to socket connect. Works for some secure sites, but not all. Failed after about 10 seconds.

jerryneedell, 2024-02-12:
SO this is getting strange. The example I have been using to test access to AIO has been failing with the Repeated Socket failures reliable witht the latest builds BUT I just found taht if I change the name of the AIO feed I am trying to access or create, the it works fine:
```
Adafruit CircuitPython 9.0.0-beta.0-22-g58b1ce46f1 on 2024-02-09; Adafruit Metro ESP32S3 with ESP32S3
>>> 
>>> import aio
feed exists
got feeds
>>> 

```
here is the code
```
 import binascii
import io
import busio
import os
import ssl
import time
import board
import socketpool
import wifi
import adafruit_requests
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError

aio_username = os.getenv('AIO_USERNAME')
aio_key = os.getenv('AIO_KEY')

pool = socketpool.SocketPool(wifi.radio)

requests = adafruit_requests.Session(pool, ssl.create_default_context())

# Initialize an Adafruit IO HTTP API object
io = IO_HTTP(os.getenv("AIO_USERNAME"), os.getenv("AIO_KEY"), requests)

# Adafruit IO feed configuration
try:
    # Get the 'camera' feed from Adafruit IO
    feed_camera = io.get_feed("sockettest")
    print("feed exists")
except AdafruitIO_RequestError:
    # If no 'camera' feed exists, create one
    feed_camera = io.create_new_feed("sockettest")
    print("created feed")
print("got feeds")

```
this works fine but if I change the feed name to "esp32s3eye"  it fails !!
```
Adafruit CircuitPython 9.0.0-beta.0-22-g58b1ce46f1 on 2024-02-09; Adafruit Metro ESP32S3 with ESP32S3
>>> 
>>> import aio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "aio.py", line 27, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 716, in request
OutOfRetries: Repeated socket failures

```

It does not matter if the feed exists or not -- as long as it is not named esp32s3eye ....

jerryneedell, 2024-02-12:
I am now finding that the test above works OK -- most of the time -- I deleted the existing esp32s3eye feed and then reran it successfully.  This is quite strange, but this may not actually be a CP issue...yet

I think the issue I am seeing is not related to the SSL issue @justmobilize is reporting .

jerryneedell, 2024-02-12:
There do appear to be a few different things going on. The simple test I have been using seems to work OK BUT when I have mode complex code using the camera, it breaks --- It may well be memory issues.
I will work on trying to create some better test examples and try to exonerate CP from the issue with AIO.
Then I'll get back to why the camera breaks it....

tannewt, 2024-02-12:
@justmobilize Please file a new issue for the api.github.com issue. It isn't memory related like I suspect this camera + IO issue is.

jerryneedell, 2024-02-13:
After further testing, it appears that the "repeated socket failures" errors are misleading and not reproducible with a simple test case.  I am able to reliably send data to AIO from the esp32s3_eye and metro_esp32s3.  The issues arise when attempting to send camera images to AIO and this need further review. I am closing this issue and will reopen an issue regarding sending images to AIO when I have a better example.

jerryneedell, 2024-02-13:
I am re-opening this since I have found that there is definitely an issue starting with build  https://github.com/adafruit/circuitpython/commit/7ba62bb1cb6a52eec7d91d4b48ce3063d0502de7
that is still present 

I have been working with the MEMENTo Doorbell guide code https://learn.adafruit.com/diy-iot-doorbell-camera-with-memento/code  and it works fine with commit https://github.com/adafruit/circuitpython/commit/09fcc5997f2359761c57bf68aeb89684ad9acf42  but fails with anything newer.

the failure is the same as reported above:
```
Adafruit CircuitPython 9.0.0-beta.0-26-gc6d195a39f on 2024-02-12; Adafruit Camera with ESP32S3
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
CircuitPython Doorbell Camera
Connecting to Needell Airport
Connected to Needell Airport!
Traceback (most recent call last):
  File "code.py", line 46, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 716, in request
OutOfRetries: Repeated socket failures

Code done running.
```  but as I found, it is not reproducible wit a simple AIO test.  Something more complex is going on.  If I can isomer it, I will create a new issue but I did not want this to get lost for now since it does break the MEMENTO which will be getting a lot of attention.

jerryneedell, 2024-02-13:
see previous comment for re-open rational

tannewt, 2024-02-13:
I've reproduced this with IOT doorbell code.

jerryneedell, 2024-02-13:
> I've reproduced this with IOT doorbell code.

I’m glad it is not just me!

PR #8913 (2024-02-14): Disable variable length SSL buffers
Fixes #8907