adafruit/circuitpython #8910: Unable to connect to api.github.com

justmobilize, 2024-02-13:
### CircuitPython version

```python
Most recent 9.0.0 build
```

### Code/REPL

```python
import os
import ssl
import time
import board
import socketpool
import wifi
import adafruit_requests

print("------------------------------")
print(os.uname())

ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

print("Connecting to WiFi...")
while not wifi.radio.connected:
  wifi.radio.connect(ssid, password)

pool = socketpool.SocketPool(wifi.radio)
ssl_context = ssl.create_default_context()

requests = adafruit_requests.Session(pool, ssl_context)

urls = [
  "http://wifitest.adafruit.com/testwifi/index.html",
  "http://www.adafruit.com/api/quotes.php",
  "https://api.github.com",
]

import time

s = time.monotonic()

try:
  for url in urls:
    print(url)
    r = requests.get(url, timeout=300)
    r.close()
except Exception as e:
  print(e)

e = time.monotonic()

print(f"Time: {e-s} seconds")
  
print("Done!")
```

### Behavior

Added some print statements to `adafruit_requests`:

```
------------------------------
(sysname='ESP32S3', nodename='ESP32S3', release='9.0.0', version='9.0.0-beta.0-26-gc6d195a39f on 2024-02-12', machine='FeatherS3 with ESP32S3')
Connecting to WiFi...
http://wifitest.adafruit.com/testwifi/index.html
_get_socket: start
_get_socket: try: 1
_get_socket: set timeout: 300
http://www.adafruit.com/api/quotes.php
_get_socket: start
_get_socket: try: 1
_get_socket: set timeout: 300
_get_socket: start
_get_socket: try: 1
_get_socket: set timeout: 300
https://api.github.com
_get_socket: start
_get_socket: try: 1
_get_socket: set timeout: 300
_get_socket: socket.connect - OSError: Failed SSL handshake
_get_socket: try: 2
_get_socket: _free_sockets
_get_socket: set timeout: 300
_get_socket: socket.connect - OSError: Failed SSL handshake
_get_socket: try: 3
Sending request failed
Time: 25.471 seconds
Done!

Code done running.
```

### Description

On a ESP32S3 chip will fail with `OSError: Failed SSL handshake` when trying to `socket.connect` to `api.github.com`

`8.2.9`: works every time. Average time ~5 seconds

`9.0.0-alpha.2.uf2`: happens half the time: Average time ~15 seconds

`9.0.0-alpha.4.uf2`: `MemoryError: memory allocation failed, allocating 1784 bytes` on line 7

`9.0.0-alpha.5.uf2`: never errored: Average time ~12 seconds

`9.0.0-alpha.6.uf2`: always errored: Average time ~20 seconds
`9.0.0-beta.0`: always errored: Average time ~20 seconds
`9.0.0-beta.0 from PR merge 8898`: always errored: Average time ~20 seconds

tannewt, 2024-02-13:
Here is the MBEDTLS log. I'm not sure why it is failing. Maybe outgoing buffer size?

```
I (265484) mbedtls: ssl_tls.c:3931 => handshake

I (265484) mbedtls: ssl_msg.c:2370 => flush output

I (265484) mbedtls: ssl_msg.c:2379 <= flush output

I (265484) mbedtls: ssl_tls.c:3850 client state: MBEDTLS_SSL_HELLO_REQUEST

I (265494) mbedtls: ssl_msg.c:2370 => flush output

I (265504) mbedtls: ssl_msg.c:2379 <= flush output

I (265504) mbedtls: ssl_tls.c:3850 client state: MBEDTLS_SSL_CLIENT_HELLO

I (265514) mbedtls: ssl_client.c:938 => write client hello

I (265524) mbedtls: ssl_msg.c:2800 => write handshake message

I (265524) mbedtls: ssl_msg.c:2960 => write record

I (265534) mbedtls: ssl_msg.c:3097 <= write record

I (265534) mbedtls: ssl_msg.c:2921 <= write handshake message

I (265544) mbedtls: ssl_client.c:1026 <= write client hello

I (265554) mbedtls: ssl_msg.c:2370 => flush output

I (265554) mbedtls: ssl_msg.c:2384 message length: zu, out_left: zu

I (265564) mbedtls: ssl_msg.c:2391 ssl->f_send() returned 215 (-0xffffff29)

I (265574) mbedtls: ssl_msg.c:2418 <= flush output

I (265584) mbedtls: ssl_tls.c:3850 client state: MBEDTLS_SSL_SERVER_HELLO

I (265584) mbedtls: ssl_tls12_client.c:1205 => parse server hello

I (265594) mbedtls: ssl_msg.c:4134 => read record

I (265594) mbedtls: ssl_msg.c:2172 => fetch input

I (265604) mbedtls: ssl_msg.c:2312 in_left: zu, nb_want: zu

I (265614) mbedtls: ssl_msg.c:2332 in_left: zu, nb_want: zu

I (265614) mbedtls: ssl_msg.c:2335 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (265624) mbedtls: ssl_msg.c:2357 <= fetch input

I (265634) mbedtls: ssl_msg.c:2172 => fetch input

I (265634) mbedtls: ssl_msg.c:2312 in_left: zu, nb_want: zu

I (265644) mbedtls: ssl_msg.c:2332 in_left: zu, nb_want: zu

I (265644) mbedtls: ssl_msg.c:2335 ssl->f_recv(_timeout)() returned 65 (-0xffffffbf)

I (265654) mbedtls: ssl_msg.c:2357 <= fetch input

I (265664) mbedtls: ssl_msg.c:4206 <= read record

I (265674) mbedtls: ssl_tls12_client.c:1457 server hello, total extension length: zu

I (265674) mbedtls: ssl_tls12_client.c:1673 <= parse server hello

I (265684) mbedtls: ssl_msg.c:2370 => flush output
[... 178 lines cut ...]
I (276454) mbedtls: ssl_tls.c:3850 client state: MBEDTLS_SSL_CLIENT_CHANGE_CIPHER_SPEC

I (276464) mbedtls: ssl_msg.c:5136 => write change cipher spec

I (276474) mbedtls: ssl_msg.c:2800 => write handshake message

I (276474) mbedtls: ssl_msg.c:2960 => write record

I (276484) mbedtls: ssl_msg.c:2370 => flush output

I (276494) mbedtls: ssl_msg.c:2384 message length: zu, out_left: zu

I (276494) mbedtls: ssl_msg.c:2391 ssl->f_send() returned -80 (-0x0050)

W (276504) mbedtls: ssl_msg.c:3093 mbedtls_ssl_flush_output() returned -80 (-0x0050)

W (276514) mbedtls: ssl_msg.c:2916 ssl_write_record() returned -80 (-0x0050)

W (276524) mbedtls: ssl_msg.c:5145 mbedtls_ssl_write_handshake_msg() returned -80 (-0x0050)

I (276534) mbedtls: ssl_tls.c:3942 <= handshake

E (276534) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x0050
E (276544) esp-tls: Failed to open new connection
```

jepler, 2024-02-13:
Just a note that "%zu" is intended to print a value of type size_t but this is not supported and so just "zu" is printed.

"I (265634) mbedtls: ssl_msg.c:2312 in_left: zu, nb_want: zu"

 This problem almost certainly stems from the hacky way I connect esp-idf's debug printing to use circuitpython's printf implementation:
```
#if CIRCUITPY_CONSOLE_UART
static int vprintf_adapter(const char *fmt, va_list ap) {
    return mp_vprintf(&mp_plat_print, fmt, ap);
}   

void port_serial_early_init(void) {
    esp_log_set_vprintf(vprintf_adapter);
}
#endif                                   
```

tannewt, 2024-02-13:
Ya, that seemed suspect to me. I'll fix it now and see what I get.

jepler, 2024-02-13:
could try hacking in %z support.

```diff
diff --git a/py/mpprint.c b/py/mpprint.c
index 467371bec2..5bc920bb62 100644
--- a/py/mpprint.c
+++ b/py/mpprint.c
@@ -464,7 +464,7 @@ int mp_vprintf(const mp_print_t *print, const char *fmt, va_list args) {
         const
         #endif
         bool long_arg = false;
-        if (*fmt == 'l') {
+        if (*fmt == 'l' || *fmt == 'z') {
             ++fmt;
             #ifdef __LP64__
             long_arg = true;
```

dhalbert, 2024-02-13:
I checked the commit for `lib/certificates` and it is the same for `8.2.9` and `main`. So it is not an obvious missing root cert.

justmobilize, 2024-02-13:
I updated the test list to:
```
urls = [
  "http://wifitest.adafruit.com/testwifi/index.html",
  "http://www.adafruit.com/api/quotes.php",
  "https://api.github.com",
  "https://api.thingspeak.com",
  "https://opensky-network.org",
  "https://api.fitbit.com/",
  "https://discord.com/api",
  "https://id.twitch.tv",
  "https://api.twitter.com/"
  "https://youtube.googleapis.com",
]
```
And `github` is the only one that fails

tannewt, 2024-02-13:
I'm at a loss. It looks like the server is resetting the connection after we send the client key:

```
I (23072) mbedtls: ssl_msg.c:2379 <= flush output

I (23082) mbedtls: ssl_tls.c:3852 client state: MBEDTLS_SSL_SERVER_HELLO_DONE

I (23082) mbedtls: ssl_tls12_client.c:2686 => parse server hello done

I (23092) mbedtls: ssl_msg.c:4134 => read record

I (23102) mbedtls: ssl_msg.c:4202 reuse previously read message

I (23102) mbedtls: ssl_msg.c:4206 <= read record

I (23112) mbedtls: ssl_tls12_client.c:2714 <= parse server hello done

I (23122) mbedtls: ssl_msg.c:2370 => flush output

I (23122) mbedtls: ssl_msg.c:2379 <= flush output

I (23132) mbedtls: ssl_tls.c:3852 client state: MBEDTLS_SSL_CLIENT_CERTIFICATE

I (23142) mbedtls: ssl_tls.c:6929 => write certificate

I (23142) mbedtls: ssl_tls.c:6940 <= skip write certificate

I (23152) mbedtls: ssl_msg.c:2370 => flush output

I (23152) mbedtls: ssl_msg.c:2379 <= flush output

I (23162) mbedtls: ssl_tls.c:3852 client state: MBEDTLS_SSL_CLIENT_KEY_EXCHANGE

I (23172) mbedtls: ssl_tls12_client.c:2729 => write client key exchange

D (23272) mbedtls: ssl_tls12_client.c:2875 value of 'ECDH: Q(X)' (255 bits) is:

D (23272) mbedtls: ssl_tls12_client.c:2875  46 81 12 6c a1 d7 c8 58 e9 91 25 ad b0 f2 01 7f

D (23272) mbedtls: ssl_tls12_client.c:2875  88 e4 62 3e 81 a4 29 1b 57 02 4f 9e 3f 3f 0d 64

D (23282) mbedtls: ssl_tls12_client.c:2875 value of 'ECDH: Q(Y)' (0 bits) is:

D (23292) mbedtls: ssl_tls12_client.c:2875  00

D (23392) mbedtls: ssl_tls12_client.c:2903 value of 'ECDH: z' (255 bits) is:

D (23392) mbedtls: ssl_tls12_client.c:2903  57 db 22 3a 28 6e 23 08 3e a4 56 b8 0a f6 fd ee

D (23402) mbedtls: ssl_tls12_client.c:2903  2d f2 b1 2d b7 9a 97 76 b3 8b 10 8a b5 87 27 62

I (23412) mbedtls: ssl_msg.c:2800 => write handshake message

I (23412) mbedtls: ssl_msg.c:2960 => write record

D (23422) mbedtls: ssl_msg.c:3044 output record: msgtype = 22, version = [3:3], msglen = 37

I (23432) mbedtls: ssl_msg.c:2370 => flush output

I (23432) mbedtls: ssl_msg.c:2384 message length: 42, out_left: 42

D (23442) lwip: lwip_send(58, data=0x3fce5f7c, size=42, flags=0x0)

D (23452) lwip: ip4_output_if: st1

D (23452) lwip: IP header:

D (23452) lwip: +-------------------------------+

D (23462) lwip: | 4 | 5 |  0x00 |        82     | (v, hl, tos, len)
[... 142 lines cut ...]

W (23922) mbedtls: ssl_msg.c:3093 mbedtls_ssl_flush_output() returned -80 (-0x0050)

W (23932) mbedtls: ssl_msg.c:2916 ssl_write_record() returned -80 (-0x0050)

W (23942) mbedtls: ssl_msg.c:5145 mbedtls_ssl_write_handshake_msg() returned -80 (-0x0050)

I (23952) mbedtls: ssl_tls.c:3944 <= handshake

E (23952) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x0050
I (23962) esp-tls-mbedtls: (FFFFFFB0): UNKNOWN ERROR CODE (0050)
E (23972) esp-tls: Failed to open new connection
D (24022) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbb8ec
D (24022) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbb8ec
D (24032) lwip: lwip_shutdown(57, how=2)

D (24032) lwip: tcp_close: closing in 
D (24032) lwip: State: CLOSED

D (24042) lwip: lwip_close(57)

D (24042) lwip: lwip_shutdown(58, how=2)

D (24042) lwip: lwip_close(58)
```

justmobilize, 2024-02-13:
Is it possible to compare to `8.2.9` to see where it's different?

dhalbert, 2024-02-14:
Confirmed that 9.0.0-alpha.5 fails and 9.0.0-alpha.6 succeeds, so we can do a bisect in that range. Tested on both ESP32-S2 and ESP32-S3. Both show this behavior.

justmobilize, 2024-02-14:
@dhalbert that 5 fails and 6 succeeds, or the other way around? I had 5 being successful...

Also the other strange part was how much slower it was vs 8.2.9

dhalbert, 2024-02-14:
> @dhalbert that 5 fails and 6 succeeds, or the other way around? I had 5 being successful...

Sorry that was my late-night error. Fixed.

dhalbert, 2024-02-14:
Bisect shows api.github.com fails as of the first commit in #8686, 051ad4667f12a2dc0b2818072c30e72145d81b4d. So I'll look further at changes there, ESP-IDF issues, and maybe try advancing past there. The third commit in #8686 set it to exactly v5.1.2, which was needed to get it to work.

justmobilize, 2024-02-14:
ugh 5.1 -> 5.1.2: `Showing 478 changed files with 11,490 additions and 3,813 deletions.`

dhalbert, 2024-02-14:
I'm getting this to work by adjusting the MBEDTLS_ECP settings. Will have a PR later.

PR #8924 (2024-02-15): esp-idf-config/sdkconfig.defaults: Fix TLS access to api.github.com; remove BLE from 4MB ESP32-S3
- Fixes #8910
- Fixes #8919 

Both of the options below were not set (and therefore not turned on), and it caused TLS access to `api.github.com` to not work at all.
```
# CONFIG_MBEDTLS_ECP_NIST_OPTIM is not set
# CONFIG_MBEDTLS_ECP_FIXED_POINT_OPTIM is not set
```
Turning on either one fixes the problem. Turning on  `CONFIG_MBEDTLS_ECP_NIST_OPTIM` instead of `CONFIG_MBEDTLS_ECP_FIXED_POINT_OPTIM` is smaller (+2.2kB vs +6.9kB) and faster (1.6 seconds vs 7.6 seconds, at least some of the time).

I  [... cut]
closed 2024-02-15 by dhalbert