adafruit/circuitpython #8925: SSL error connecting to aeroapi.flightaware.com

justmobilize, 2024-02-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0-33-g264e68fe16 on 2024-02-15; FeatherS3 with ESP32S3
from `Merge pull request #8924 from dhalbert/tls-nist-optim`
```

### Code/REPL

```python
import os
import socketpool
import ssl
import wifi
import adafruit_requests

wifi_ssid = os.getenv("CIRCUITPY_WIFI_SSID")
wifi_password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

while not wifi.radio.connected:
    wifi.radio.connect(wifi_ssid, wifi_password)

requests = adafruit_requests.Session(socketpool.SocketPool(wifi.radio), ssl.create_default_context())
response = requests.get("https://aeroapi.flightaware.com/aeroapi/flights/search")
print(response.text)
```

### Behavior

In CircuitPython
```
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
  File "adafruit_requests_2.py", line 752, in get
  File "adafruit_requests_2.py", line 691, in request
  File "adafruit_requests_2.py", line 496, in _get_socket
RuntimeError: Sending request failed
```

in CPython:
```
In [1]: import requests

In [2]: response  = requests.get("https://aeroapi.flightaware.com/aeroapi/flights/search").text

In [3]: print(response.text)

        {"title": "Invalid API key", "reason": "INVALID_API_KEY", "detail": "Provided API key is not valid", "status": 401}
```

tannewt, 2024-02-16:
Does it work with 8.x?

dhalbert, 2024-02-16:
This appears to be a missing root cert: `SSL.com Root Certification Authority RSA`. I'll add it to certificates and re-test.

justmobilize, 2024-02-16:
It does not. Tested on a `M4+Airlift`, `Pico W` and `UM FeatherS3`

justmobilize, 2024-02-16:
@dhalbert feel free to ping me here or discord if you want me to test

dhalbert, 2024-02-16:
@justmobilize I tested with your test program with the new cert, and now get:
```
        {"title": "Invalid API key", "reason": "INVALID_API_KEY", "detail": "Provided API key is not valid", "status": 401}
```
which is what I expect. PR on the way.

justmobilize, 2024-02-16:
Awesome!

PR #8935 (2024-02-16): lib/certificates: update to include ssl.com RSA cert
- Fixes #8925.

I forgot that this was already reported in https://github.com/adafruit/certificates/issues/3.
closed 2024-02-16 by tannewt