adafruit/circuitpython #8942: espcamera ESPIDF Generic Failure and Memory Error on ESP32S2

jerryneedell, 2024-02-17:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut
import time
import espcamera

cam = espcamera.Camera(
    data_pins=[board.D6,board.A5,board.D9,board.A4,board.D10,board.A3,board.D11,board.A2],
    external_clock_pin=board.D12,
    pixel_clock_pin=board.A1,
    vsync_pin=board.A0,
    href_pin=board.D13,
    pixel_format=espcamera.PixelFormat.RGB565,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
    grab_mode=espcamera.GrabMode.WHEN_EMPTY,
    powerdown_pin=board.D5,
    reset_pin=board.RX
)
print(cam.sensor_name)
```

### Behavior

```
10.0.0.107 | REPL | 9.0.0-beta.0-52-g6c1e34e957reload.
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; Adafruit Feather ESP32S2 with ESP32S2
>>> 
>>> 
>>> import code_test_rgb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code_test_rgb.py", line 18, in <module>
espidf.IDFError: Generic Failure
10.0.0.107 | Done | 9.0.0-beta.0-52-g6c1e34e957soft reboot
```

### Description

This is being run on a feather esp32s2 wit an OV5460 breakout.

See the additional example below that fails with a Memory error when adafruit_requests is also used.
I suspect these failures are both dies to memory issues on the esp32s2 so I am reporting both here.

### Additional information

switching to PixelFormat = JPEG runs normally

```

import board
from digitalio import DigitalInOut
import time
import espcamera

cam = espcamera.Camera(
    data_pins=[board.D6,board.A5,board.D9,board.A4,board.D10,board.A3,board.D11,board.A2],
    external_clock_pin=board.D12,
    pixel_clock_pin=board.A1,
    vsync_pin=board.A0,
    href_pin=board.D13,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
    grab_mode=espcamera.GrabMode.WHEN_EMPTY,
    powerdown_pin=board.D5,
    reset_pin=board.RX
)
print(cam.sensor_name)
```

```
10.0.0.107 | REPL | 9.0.0-beta.0-52-g6c1e34e957reload.
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; Adafruit Feather ESP32S2 with ESP32S2
>>> import code_test_jpeg
OV5640
```

Both examples run as espected on an esp32s3 (memento or metro_esp32s3)

I suspect this is a memory issue and not really related to the Pixel Format.

If I extend the working code (JPEG format) to try to use the requests library to access AIO, then it fails with a Memory Error.

code:
```
# SPDX-FileCopyrightText: Copyright (c) 2022 Jeff Epler for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""
Upload a jpeg image to Adafruit IO at regular intervals

This example requires:
 * ESP32-S3-EYE development kit from Espressif

To use:
 * On io.adafruit.com, create a feed named "image" and turn OFF history
 * On io.adafruit.com, create a dashboard and add an "image" block
   using the feed "image" as its data
 * Set up CIRCUITPY/.env with WiFI and Adafruit IO credentials
 * Copy the project bundle to CIRCUITPY
"""

import binascii
import io
import busio
import os
import ssl
import time
import board
import espcamera
import socketpool
import wifi
import adafruit_requests
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError
from digitalio import DigitalInOut

aio_username = os.getenv('AIO_USERNAME')
aio_key = os.getenv('AIO_KEY')

cam = espcamera.Camera(
    data_pins=[board.D6,board.A5,board.D9,board.A4,board.D10,board.A3,board.D11,board.A2],
    external_clock_pin=board.D12,
    pixel_clock_pin=board.A1,
    vsync_pin=board.A0,
    href_pin=board.D13,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
    grab_mode=espcamera.GrabMode.WHEN_EMPTY,
    powerdown_pin=board.D5,
    reset_pin=board.RX
)
print(cam.sensor_name)

cam.vflip = True
print("camera initialized")

pool = socketpool.SocketPool(wifi.radio)

requests = adafruit_requests.Session(pool, ssl.create_default_context())

# Initialize an Adafruit IO HTTP API object
io = IO_HTTP(os.getenv("AIO_USERNAME"), os.getenv("AIO_KEY"), requests)
# Adafruit IO feed configuration
try:
    # Get the 'camera' feed from Adafruit IO
    feed_camera = io.get_feed("featheresp32s2")
except AdafruitIO_RequestError:
    # If no 'camera' feed exists, create one
    feed_camera = io.create_new_feed("featheresp32s2")

while True:
    frame = cam.take(1)
    if isinstance(frame, memoryview):
        jpeg = frame
        print(f"Captured {len(jpeg)} bytes of jpeg data")

        # b2a_base64() appends a trailing newline, which IO does not like
        encoded_data = binascii.b2a_base64(jpeg).strip()
        print(f"Expanded to {len(encoded_data)} for IO upload")

        # Send encoded_data to Adafruit IO camera feed
        print("Sending image to Adafruit IO...")
        io.send_data(feed_camera["key"], encoded_data)
        print("Sent image to IO!")

        time.sleep(10)
```

fails:
```
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; Adafruit Feather ESP32S2 with ESP32S2
>>> 
>>> import code_aio
OV5640
camera initialized
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code_aio.py", line 65, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 752, in get
  File "adafruit_requests.py", line 691, in request
  File "adafruit_requests.py", line 510, in _get_socket
MemoryError: 
>>> 

```
line 65 is:
```
    feed_camera = io.get_feed("featheresp32s2")
```

These may be unrelated issues, let me know if you would prefer them to be separated.

jerryneedell, 2024-02-17:
I am trying to make a build for the feather esp32s2 with imagecapture instead of esp32camera, but it is not building. What am I missing?

```

jerryneedell@Mac-mini espressif % cat boards/adafruit_feather_esp32s2/mpconfigboard.mk
USB_VID = 0x239A
USB_PID = 0x80EC
USB_PRODUCT = "Adafruit Feather ESP32S2"
USB_MANUFACTURER = "Adafruit"

IDF_TARGET = esp32s2

CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m
CIRCUITPY_ESP_FLASH_SIZE = 4MB

CIRCUITPY_ESP_PSRAM_SIZE = 2MB
CIRCUITPY_ESP_PSRAM_MODE = qio
CIRCUITPY_ESP_PSRAM_FREQ = 80m

CIRCUITPY_IMAGECAPTURE = 1
CIRCUITPY_ESPCAMERA = 0

```

```
jerryneedell@Mac-mini espressif % gmake BOARD=adafruit_feather_esp32s2 clean           
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
rm -rf build-adafruit_feather_esp32s2 
jerryneedell@Mac-mini espressif % gmake BOARD=adafruit_feather_esp32s2      
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
gmake: *** No rule to make target 'common-hal/imagecapture/ParallelImageCapture.c', needed by 'build-adafruit_feather_esp32s2/genhdr/qstr.i.last'.  Stop.
```

jepler, 2024-02-17:
That code has been removed. Scott removed part of it in 3b159317240a1ff2b56101d1c92aec4ab2d9f454 while I removed the rest in efed3c975897069943c7e409fd48670dc17e0a37. So you could travel back in time to efed3c975897069943c7e409fd48670dc17e0a37 with git to have that module possible to build, maybe -- it hadn't been used in a long time (not since the idf5 update), so I don't even know whether it would build before the related files were removed.

jerryneedell, 2024-02-17:
I think the real question is whether there is a bug causing the problems with espcamera on the esp32s2 or if esp32s2 simply can’t handle espcamera. If that is the case, do we just document that and move on or is it worth trying to revert to something lime imagecapture.  There does not appear to have been a great demand for it so I expect there are more important issues to resolve at this time.

tannewt, 2024-02-20:
> That code has been removed. Scott removed part of it in [3b15931](https://github.com/adafruit/circuitpython/commit/3b159317240a1ff2b56101d1c92aec4ab2d9f454) while I removed the rest in [efed3c9](https://github.com/adafruit/circuitpython/commit/efed3c975897069943c7e409fd48670dc17e0a37). So you could travel back in time to [efed3c9](https://github.com/adafruit/circuitpython/commit/efed3c975897069943c7e409fd48670dc17e0a37) with git to have that module possible to build, maybe -- it hadn't been used in a long time (not since the idf5 update), so I don't even know whether it would build before the related files were removed.

I don't think it would work before I removed it. It was relying on other stuff I also removed.

tannewt, 2024-02-20:
> I think the real question is whether there is a bug causing the problems with espcamera on the esp32s2 or if esp32s2 simply can’t handle espcamera. If that is the case, do we just document that and [...]

I'm not super interested in getting things going on the S2 if we don't have a project that uses it. The S3 is much better.

deshipu, 2024-02-20:
I have used the espcamera on an esp32-s2 without problems a while back, so I'm confident it's possible: https://hackaday.io/project/182342-camera-shield-for-s2-mini

jerryneedell, 2024-02-27:
> I'm not super interested in getting things going on the S2 if we don't have a project that uses it. The S3 is much better.

That is fine with me. I have migrated my camera projects to esp32s3 boards and I am happy to put the "kaluga" back on the shelf.

jerryneedell, 2024-02-27:
> I have used the espcamera on an esp32-s2 without problems a while back, so I'm confident it's possible: https://hackaday.io/project/182342-camera-shield-for-s2-mini

Have you tried this with espcamera and CP9? I'm just curious if you will have the same issues with memory.

deshipu, 2024-06-07:
> Have you tried this with espcamera and CP9? I'm just curious if you will have the same issues with memory.

Sorry for the delay, I didn't see your comment. I just tried with 9.1.0-beta on a Lolin S2 Mini, with this code:

```py
import busio
import espcamera
import board

i2c = busio.I2C(scl=board.IO35, sda=board.IO33)
data_pins = (
    board.IO21, board.IO17, board.IO16, board.IO18,
    board.IO37, board.IO34, board.IO36, board.IO39,
)
c = espcamera.Camera(
    data_pins=data_pins,
    external_clock_pin=board.IO11,
    pixel_clock_pin=board.IO12,
    vsync_pin=board.IO40,
    href_pin=board.IO38,
    pixel_format=espcamera.PixelFormat.RGB565,
    frame_size=espcamera.FrameSize.QVGA,
    i2c=i2c,
)
c.take()
```

and I'm getting ``espidf.IDFError: Operation or feature not supported`` error on the line where I try to create the camera object. Looks like a regression?