adafruit/circuitpython #8943: ESP32 missing `SOL_SOCKET` and SO_REUSEADDR`

bill88t, 2024-02-17:
### CircuitPython version

```python
Master
```

### Code/REPL

```python
import wifi
import socketpool
pool = socketpool.SocketPool(wifi.radio)
socket = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
socket.setsockopt(pool.SOL_SOCKET, pool.SO_REUSEADDR, 1)
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SocketPool' object has no attribute 'SOL_SOCKET'

dhalbert, 2024-02-17:
I do not see this problem on the tip of main, building for a QT Py ESP32:
```py
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; Adafruit QT Py ESP32 PICO with ESP32
>>> import wifi
>>> import socketpool
>>> pool = socketpool.SocketPool(wifi.radio)
>>> socket = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
>>> socket.setsockopt(pool.SOL_SOCKET, pool.SO_REUSEADDR, 1)
>>> 
```

bill88t, 2024-02-17:
Oh whoops! I guess the PRxxxx aws S3 artifacts are of the actually merged commits.
I thought I should be using `no-branch` ones.

closed 2024-02-17 by bill88t

dhalbert, 2024-02-17:
That's right, the `no-branch` ones are now named correctly. `no-branch` was just wrong. Only merge commits and releases are in S3 -- they are not test builds while a PR is in progress.