adafruit/circuitpython #8944: Large Files on MEMENTO Causes Device Reset & Eventual Hard Crash

FoamyGuy, 2024-02-18:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.1 on 2024-02-17; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
# Doesn't seem to matter
print("Hello World!")
```

### Behavior

I've noticed that putting files of around 900kb in size on the MEMENTO drive seems to cause problems for the device. Originally I noticed this with a .bmp file that is 900kb I can provide that if it's though to be important, but I've since replicated the same issue using this "dummy" text file that was filled with 900kb of As, spaces and newlines. 
[dummy.txt](https://github.com/adafruit/circuitpython/files/14320692/dummy.txt)

So far I have observed the same issues when pasting the file from an Ubuntu PC and a Raspberry Pi with standard OS.

Sometimes attempting to paste the dummy.txt file once causes the device to reset and then show a hard fault message if you ctrl-c / ctrl-d in the serial console.

Other times pasting a single copy of dummy.txt suceeds, but then if you attempt to paste a second one thus far I have not seen it succeed. It has always reset part way thru the paste and then come back up with a hard fault message in serial 

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

RetiredWizard, 2024-02-18:
While trying to reproduce this from Ubuntu, I noticed that once the 900kb file was placed on the Memento, any subsequent file copy attempts resulted in an immediate auto-reload and boot to safe mode. Removing the 900kb file allows smaller files to once again be copied to the device.

FoamyGuy, 2024-02-18:
I have reproduced the same issue when the host OS pasting the file is Win 10 as well. 

For testing that I did `storage.erase_filesystem()` first to start with a blank slate. 

Then plugged in the MEMENTO to Win10 PC and copied the dummy.txt file to the drive

First paste of that file was successful. Renaming the first copy of the file was successful.

Then when I attempt to paste a new copy of the file the device resets with the same hard fault message as posted above.

dhalbert, 2024-02-18:
I see the MEMENTO going into safe mode when copying the 900kB file the first time from Ubuntu. I removed the file after each copy -- I did not trying reformatting the filesystem. I reproduced on alpha.2 and all later builds.

I could _not_ reproduce copying the same file to a Metro ESP32-S3.

RetiredWizard, 2024-02-18:
While the initial attempt to copy the 900kb file takes some time and does create the directory entry, the file size is shown as zero bytes. Once this errant directory entry is on the device subsequent attempts to copy files (even small ones) cause the hard fault.

I also reproduced the error using web workflow rather than the host os copies.

FoamyGuy, 2024-02-18:
I was also able to reproduce without the host OS involved in the copying at all by setting storage to writable and using code like this in the REPL:
```
>>> f = open("dummy.txt", "w")
>>> row = "AAAA AAAA AAAA AAAA\n"
>>> f.write(row*45*1000)
900000
>>> f.close()
>>> f = open("dummy2.txt", "w")
>>> f.write(row*45*1000)
Disconnected
```
It reset to the same hardfault message during the second write call.

dhalbert, 2024-02-18:
The MEMENTO wants a larger CIRCUITPY and no OTA partition. This was not specified properly in `mpconfigboard.mk`. However, after fixing this I'm still getting the issue. Still working on this but I think I'm on track.

dhalbert, 2024-02-18:
I corrected the MEMENTO build `mpconfigboard.mk` to use `FLASH_SIZE_SDKCONFIG` instead of `FLASH_SDKCONFIG`, and setting `CIRCUITPY_DUALBANK = 0`. See https://github.com/dhalbert/circuitpython/tree/memento-storage-crash.

Now, starting from a clean filesystem, I copy the file once, and then when copying again, the filesystem has been set to read-only due to an error during the first copy. `var/log/syslog` shows:
```
Feb 17 23:37:20 cod udisksd[819]: Mounted /dev/sdb1 at /media/halbert/CIRCUITPY on behalf of uid 1000
Feb 17 23:37:20 cod kernel: [313657.172574] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
   [the error below appears very quickly during the first copy]
Feb 17 23:37:42 cod kernel: [313679.091388] FAT-fs (sdb1): error, fat_free_clusters: deleting FAT entry beyond EOF
Feb 17 23:37:42 cod kernel: [313679.091394] FAT-fs (sdb1): Filesystem has been set read-only
```
(The "not properly unmounted" is innocuous, and shows up on other boards that accept the copy without error).

RetiredWizard, 2024-02-18:
Dan, I built an image using the changes from your repo and had similar (but not exactly the same) issues. My Cirucitpy drive was still only 2.3M though. Is it possible you made a partition table change to /esp-idf-config/partitions-4MB-no-ota.csv locally that's not pushed to your repo?

dhalbert, 2024-02-19:
I have been working more on this, and have leads on what is really going on. Once DUALBANK is turned off and the proper partition files are set up, that exercises some bug that causes nothing to be written to the filesystem. So even reformatting the filesystem doesn't work, though it appears to, and what I'm seeing is the old leftover larger filesystem. I'll say more tomorrow.

dhalbert, 2024-02-19:
Here's what I think is going on:

The MEMENTO *aka adafruit_esp32s3_camerawas set up using an unusual partition scheme. There is a single 2.8MB 2816kB) firmware partition, instead of two 1.4MB firmware partitions (one for OTA update of firmware). The user filesystem partition is 960KB like other boards. The TinyUF2 bootloader `combined.bin` sets up this partition scheme.

There was an incorrect `FLASH_SDKCONFIG` setting in the `mpconfigboard.mk` for the MEMENTO. That config name is obsolete.

`CIRCUITPY_DUALBANK`, whose value is also used for `CIRCUITPY_STORAGE_EXTEND` was turned on for the MEMENTO, even though there is no dual bank and OTA update is not possible.

As noted by @jepler, `esp_ota_get_next_update_partition()` will return first ota partition, even if it's the only one. This means that CircuitPython thinks it can use the ota partition as the second filesystem partition, after the 960kB partition for use by the `CIRCUITPY_STORAGE_EXTEND` partition. The sum of these two sizes is 960kB + 2816kB, which corresponds to the 3.7MB CIRCUITPY we are seeing with `statvfs()` and `df -h`. When the filesystem is created by 9.0.0-beta.1 or earlier, sy by `storage.erase_filesystem()`, it will have the wrong size.

So this is wrong: the OTA partition should not be a second filesystem partition. We need to turn off DUALBANK and EXTEND, make a beta.2, and then also tell everyone to do `storage.erase_filesystem()` to fix up their CIRCUITPY to be the right size.

There is also a bug that prevents writing to the single-partition, non-EXTEND, CIRCUITPY. That causes there to be no CIRCUITPY if flash is erased before installing my non-EXTEND branch build listed above, or causes `storage.erase_filesystem(0` to not actually do anything in the branch build, which preserves the 2.7MB filesystem incorrectly. @jepler is diagnosing that at this moment. See https://discord.com/channels/327254708534116352/327298996332658690/1209158871944265748 and later.

We could also get rid of the special partition scheme on this board and get a bigger CIRCUITPY, but that requires also installing a new bootloader.

PR #8952 (2024-02-20): Espressif: fix handling of single-partition CIRCUITPY (non-CIRCUITPY_STORAGE_EXTEND) - fixes MEMENTO bug
- Fixes #8944

---

- Fix `supervisor_flash_write_blocks()` to write instead of read when handling a single firmware partition (thanks @jepler).
- Fix `adafruit_esp32s3_camera/mpconfigboard.mk` (MEMENTO)to use proper setting name for partition file.
- Rename partition files to have consistent names: no-ota` if only one firmware partition.
- Forbid `CIRCUITPY_DUALBANK = 1` and `CIRCUITPY_STORAGE_EXTEND = 1` if there is only one firmware partition. Thanks @jepler for initial version of valida [... cut]
closed 2024-02-20 by jepler