adafruit/circuitpython #8945: UART: unable to set value of Pull-up/down on defined pins, especially needed on RX

davs2rt, 2024-02-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4 

digitalio allows setting pin.pull values, however using busio.UART (and possibly others) the RX pin seems to be pulled up.

In my use case, the circuit driving RX is weak, and I'm not getting solid low voltages, causing data corruption.  I'd like to see the default on inputs be configurable, or at least not pulled up or down, i.e tri-stated or pin.pull=None.
This seems to be similar to #8941 "Configure internal pull up on MISO", an enhancement.
```

### Code/REPL

```python
import board, busio, time, digitalio
while True:
    rx = digitalio.DigitalInOut(board.RX)
    rx.direction = digitalio.Direction.INPUT
    rx.pull = None
    print("RX is floating")
    time.sleep(10)
    rx.deinit()

    uart = busio.UART(tx=board.TX, rx=board.RX, timeout=60)
    print("""RX line is pulled up.
Note that there is no way to disable the pull-up in the API.""")
    time.sleep(10)
    uart.deinit()
```

### Behavior

The code should run okay.  The problem shows up as electrical due to board configuration. If you use a serial port monitor with lights, you'll see RXD flash as it's pulled high, and then off when tri-stated.

### Additional information

This is probably easiest to observe with an oscilloscope, and the serial port _not connected_ to anything driving the RX line.  Compare the difference between digitalio pin.pull = None, pin.pull = digitalio.Pull.UP and what you get with busio.UART(rx=board.RX)

tannewt, 2024-02-20:
This is also related to https://github.com/adafruit/circuitpython/issues/1270#issuecomment-1711991127 which we consider splitting out `Pad` properties from `DigitalInOut`.

The easiest thing by far is to add the pull to your circuit. Internal resistors are often not the right value anyway.

dhalbert, 2025-02-01:
This could be done via #9845.