adafruit/circuitpython #8948: Enhancement: Support built-in keyboard in terminal on LilyGO T-Deck.

mirkokral, 2024-02-18:
It would be cool if the support for the T-Deck keyboard was present in the terminal.
It would add the ability to use the T-Deck as a standalone python device.

RetiredWizard, 2024-02-18:
This was discussed in #8684 but the original poster closed it. Tannewt did make a comment suggesting it might be in the future though, so maybe this issue (or a re-opened #8684) could be marked long-term.

The python scripts from [tdeck_repl](https://github.com/RetiredWizard/tdeck_repl) mentioned in #8684 (or [PyDOS](https://github.com/RetiredWizard/PyDOS)) both require a serial terminal to initially set up but once installed get you a standalone python T-Deck device.

tannewt, 2024-02-20:
> It would be cool if the support for the T-Deck keyboard was present in the terminal. It would add the ability to use the T-Deck as a standalone python device.

How is the keyboard connected? You'd need a C implementation of scanning the keyboard and then connect it up to `serial_read()`. See https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/serial.c#L241 for how USB host keyboard is connected in.

This isn't an Adafruit-funded priority because it isn't an Adafruit board. We'd be happy to review and merge a PR from LilyGo or the community to implement it.