adafruit/circuitpython #8971: INA228 needs a CP library

tannewt, 2024-02-22:
https://www.adafruit.com/product/5832

Forum request is here: https://forums.adafruit.com/viewtopic.php?p=1004716

dhalbert, 2024-03-25:
This is on @BlitzCityDIY's plate, but awaits some changes in the Arduino library.

fabh2o, 2024-09-27:
The Arduino version of the library has reached version `1.1.0` 3 months ago, are the required changes implemented yet? Or is there an issue to keep track on the progress?

Darkextratoasty, 2025-01-27:
> This is on [@BlitzCityDIY](https://github.com/BlitzCityDIY)'s plate, but awaits some changes in the Arduino library.

Almost a year later, is there any update on the status of this?

dhalbert, 2025-02-01:
Closed by https://github.com/adafruit/Adafruit_CircuitPython_INA228 !

closed 2025-02-01 by dhalbert