adafruit/circuitpython #8972: Circuitpython core code crash. Whoops!

kenidowa, 2024-02-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit QT Py RP2040 with rp2040
Board ID:adafruit_qtpy_rp2040 UID:DF611CF78F5F3D32
I got this message:
Hard fault: memory access of instuction error.
Please file and issue with your program here (github.com etc.)
Press reset to exit safe mode.
I have a SSD1306 display and a 24lc32 eeprom over stemma i2c
```

### Code/REPL

```python
""
This is code to write and read from an eeprom
"""
import gc
import board
import displayio
import terminalio
import digitalio
from adafruit_display_text import label
import adafruit_displayio_ssd1306
import adafruit_bus_device
import microcontroller
import time
import busio
#import adafruit_drv2605
import adafruit_24lc32

print("\nQTPY_Pico EEPROM")
print("Memory Free {}".format(gc.mem_free()))
start = 0
#print(dir(board))
#print(dir(bytearray))
#print(dir(adafruit_24lc32.EEPROM))

# Initialize I2C bus and DRV2605 module.
i2c = board.STEMMA_I2C()
eeprom = adafruit_24lc32.EEPROM_I2C(i2c)
display_bus = displayio.I2CDisplay(i2c, device_address=0x3D)
WIDTH = 128
HEIGHT = 64  # Change to 64 if needed
BORDER = 2
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=WIDTH, height=HEIGHT)
splash = displayio.Group(y=3)
display.show(splash)
inner_bitmap = displayio.Bitmap(WIDTH - BORDER * 2, HEIGHT - BORDER * 2, 1)
inner_palette = displayio.Palette(1)
inner_palette[0] = 0x000000  # Black
print("length: {}".format(len(eeprom)))
#4096/32=128  3072/32=96 8*96=768
print(eeprom[4095])

def store_eeprom(address,str_name):
    str = input(str_name)
    print("length: {}".format(len(str)))
    strlng = len(str)
    print(strlng)
    #strlng = bytes(strlng,'utf-8')
    for i in range(0,16):  #change back to 16
        eeprom[address+i] = 0
    if strlng > 16:
        strlng = 16
    byte_str = bytes(str,'utf-8')
    eeprom[address] = strlng
    print(byte_str)
    for i in range(1,strlng+1):
        eeprom[address+i] = byte_str[i-1]
    print(eeprom[address+1:address+strlng+1])
    print(eeprom[address:address+32])
    return strlng

def read_eeprom(addr):
    #print(eeprom[addr:addr+32])
    cnt = eeprom[addr]
    cnt1 = int.from_bytes( cnt,"big" )
    #print(cnt1)
    try:
        eestrgi = eeprom[addr+1:addr+cnt1+1]
        eestrg = eestrgi.decode("utf-8")
        return [eestrg,cnt1]
    except:
        return ["blank",0]

#cnt1 += cnt1
end_byte = eeprom[4095]
pointx = int.from_bytes(end_byte,"big")
r_indx = 0

while True:
    mode = input("Enter s for store or\n r for read\n i for index\n b for base \n e edit: ")
    base = start
    if mode == 's':
        s_indx = pointx*64
        print("Store index: ",s_indx)
        if s_indx > 4032:
            print("no more entries to store!!!")
        lng0 = store_eeprom(s_indx,"Enter Name: ")
        lng1 = store_eeprom(s_indx+lng0+1,"Enter User ID: ")  #add 1 to lng to account for first location is string length
        lng2 = store_eeprom(s_indx+lng0+lng1+2,"Enter PW: ")
        pointx = pointx+1
        eeprom[4095] = int(pointx)
        print(int(s_indx/64))
        if s_indx > 4032:
            print("no more entries to store!!!")
    elif mode == 'e':
        if e_indx > 4032:
            print("no more entries to store!!!")
        lng0 = store_eeprom(e_indx,"Enter Name: ")
        lng1 = store_eeprom(e_indx+lng0+1,"Enter User ID: ")  #add 1 to lng to account for first location is string length
        lng2 = store_eeprom(e_indx+lng0+lng1+2,"Enter PW: ")
        #s_indx = s_indx+64
        #eeprom[4095] = int(s_indx/64)
        print(int(e_indx/64))
        if e_indx > 4032:
            print("no more entries to store!!!")
    elif mode == 'r':
        print(r_indx)
        name_entry = read_eeprom(r_indx)
        print(name_entry[0])
        #print(name_entry[1])
        user_entry = read_eeprom(r_indx+name_entry[1]+1)
        print(user_entry[0])
        pw_entry = read_eeprom(r_indx+name_entry[1]+user_entry[1]+2)
        print(pw_entry[0])
    elif mode == 'i':
        index = 0
        for i0 in range(0,63,5):
            q1 =input("enter m for more or c to choose index")
            if q1 == 'm':
                for i1 in range(i0,i0+5):
                    name_entry = read_eeprom(index)
                    index = index + 64
                    icode = i1
                    if icode == 63:
                        print(icode,' ',name_entry[0],"Last")
                        break
                    else:
                        print(icode,' ',name_entry[0])
            else:
                break
        indx = int(input("Enter index number: "))
        r_indx = base + indx*64
        e_indx = r_indx
    elif mode == 'b':
        print(eeprom[4090:4096])
        rmode = input("Enter R to reset to 0 or x to exit: ")
        if rmode == 'R':
            offset = int(input("enter offset number 0-62"))
            eeprom[4095] = start+offset  # start is 0
            print(eeprom[4095])
```

### Behavior

It does nothing but leaves the message as stated above.
Hard fault: memory access of instuction error.
Please file and issue with your program here (github.com etc.)
Press reset to exit safe mode.

### Description

Happens immediately upon plugging in to power or after a reset

dhalbert, 2024-02-22:
Could you try with CircuitPython 9.0.0-beta.2?

kenidowa, 2024-02-22:
Well I am trying CircuitPython 9.0.0-beta.2 and it seemed to take care of the hard fault but there are alot of changes from 8.2.9 in the display area that need to be fix before I know if I have a solid fix.
thanks,
ken

dhalbert, 2024-02-22:
Mainly just change `display.show(splash)` to `display.root_group = splash`.

kenidowa, 2024-02-22:
I did that change but continued to have other changes to my code. It became a never ending battle.  All the imbeded modules related to displays i2c seemed to have changed.  Now I have another HARD failure.  Is there an adafruit learn or some tutorial.  That's what I followed in the past but they seemed to be all outdated.

dhalbert, 2024-02-23:
I'd say take out all the eeprom code temporarily until you get it working. It would be useful to know if the eeprom is causing you to go into safe mode.

kenidowa, 2024-02-23:
I have now got everything working, display and eeprom code together and displaying eeprom data to display. 
 I do get a bunch of FutureWarning messages at start, see below.
FutureWarning: Display moved from displayio to busdisplay
FutureWarning: Display renamed BusDisplay
FutureWarning: EPaperDisplay moved from displayio to epaperdisplay
FutureWarning: FourWire moved from displayio to fourwire
FutureWarning: I2CDisplay moved from displayio to i2cdisplaybus
FutureWarning: I2CDisplay renamed I2CDisplayBus
FutureWarning: I2CDisplay moved from displayio to i2cdisplaybus
FutureWarning: I2CDisplay renamed I2CDisplayBus

lots of changing coming?
Thanks,
Ken

dhalbert, 2024-02-23:
Yes, those changes are in 9.0.0, but there is still backward-compatible code there. In 10.0.0, the old names will be gone.

kenidowa, 2024-02-25:
Maybe I'll try with new structures after I get my functionality I want.

closed 2024-02-25 by kenidowa