adafruit/circuitpython #8977: SWAN R5: Sleep modes are broken

bablokb, 2024-02-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Swan R5 with STM32L4R5ZIY6
```

### Code/REPL

```python
def work():
  led.value = 1
  time.sleep(LED_TIME)
  led.value = 0

while True:
  # do some work
  work()

  # sleep
  if MODE == SPIN:
    next = time.monotonic() + INT_TIME
    while time.monotonic() < next:
      continue
  elif MODE == SLEEP:
    print(f"(normal) sleep for {INT_TIME}s")
    time.sleep(INT_TIME)
  elif MODE == LIGHT_SLEEP:
    print(f"(light) sleep for {INT_TIME}s")
    time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic()+INT_TIME)
    alarm.light_sleep_until_alarms(time_alarm)
  elif MODE == DEEP_SLEEP:
    print(f"(deep) sleep for {INT_TIME}s")
    time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic()+INT_TIME)
    alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

With light-sleep, I can see the following current (at 3.6V):
[image]

So in light-sleep mode, the SWAN needs about 30mA. When the LED is on and the device sleeps normally (during `work()`), current is only about 16mA.

Deep-sleep is even more broken:
[image]

You can see that the device tries to switch to deep-sleep, but it wakes up again without actually sleeping (note that this is not fake deep-sleep, there was no USB connection). After wake up, you can see the 16mA during `work()` again, then the try-to-go-to-deep-sleep-and-fail cycle starts over.

dhalbert, 2026-11-12:
@bablokb #11855 may fix this. Want to test?

bablokb, 2026-11-12:
Yes, I will update with the device and retest.

bablokb, 2026-11-13:
I tested with the 11855 artifacts. Neither light-sleep nor deep-sleep work, but the behavior is different now. Even when connected with USB the light-sleep sometimes works, sometimes it returns at once. Fake deep-sleep is similar: sometime it works, sometimes the device just resets (i.e. it disconnects and connects again to tio). I could not detect a pattern.

When connected with my PPK2, I see this behavior too. Instead of entering deep-sleep, it resets at once.

As far as I can see, there are a few more PRs in the queue, so it might be better to wait with further testing until the dust has settled.

[image]

vegardw, 2026-11-13:
Light sleep sometimes returning at once should be fixed by my PR #11862, if want you could test that and see how it behaves. I haven't tested it on other STM boards, so I wouldn't mind some extra testing

vegardw, 2026-11-13:
@bablokb All my PRs touching sleep and alarms is merged as of commit 168de89

bablokb, 2026-11-14:
Great. It will probably be Monday until I have time to test this.

bablokb, 2026-11-16:
Retested, but it still does not work.

Light-sleep does not return anymore at once, but I still see 31mA with light-sleep compared to 17mA with normal sleep (`time.sleep()`) as documented in my initial comment. Even with 100000 samples per second I don't see the system trying to reach a low-power state (it jumps from 17mA to 31mA and basically stays there).

Deep-sleep is still the same: current does dip down but it resets instantly.