adafruit/circuitpython #8978: Hard crash when writing to flash with low memory

furbrain, 2024-02-24:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0-24-g6022b94cf6-dirty on 2024-02-11; Seeed XIAO nRF52840 Sense with nRF52840
```

### Code/REPL

```python
import time
import gc
MEM_CUTOFF = 16000
time.sleep(4)
a = []
print("Eating memory:...")
while True:
    gc.collect()
    mem_free = gc.mem_free()
    if mem_free < MEM_CUTOFF:
        print(f"mem_free: {mem_free}")
        print("opening file")
        time.sleep(0.5)
        f = open("mem_test.txt", "w")
        print("writing file")
        time.sleep(0.5)
        f.write("testy test");
        print("flushing file");
        time.sleep(0.5)
        f.flush()
        print("closing file");
        time.sleep(0.5)
        f.close()
    a.append([12]*128)
```

### Behavior

```
Eating memory:...
mem_free: 15952
opening file

[tio 17:31:57] Disconnected
[tio 17:31:58] Connected
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

When `MEM_CUTOFF` is 20000, then code behaves as expected: mem_free drops by about 4000 after creating the file, and it then continues to reduce until an out of memory error is generated
When `MEM_CUTOFF` is 16000 or lower, I get a hard crash as above

Looking at the flash code, it looks like a scratch flash sector is used as a buffer when memory is low - could this be the issue?

### Additional information

This is `boot.py`
```python
import board
import digitalio
import storage

pin = digitalio.DigitalInOut(board.D9)
pin.pull = digitalio.Pull.DOWN
pin.switch_to_input()
print("Switch value:", not pin.value)
storage.remount("/",readonly= not pin.value)
```

dhalbert, 2024-02-24:
This _might_ be releated to #8961. Could you also retest with 9.0.0 beta.2? I don't expect a change but let's make sure.

furbrain, 2024-02-24:
I've retested with 9.0.0 beta.2 Same results (although different cutoff for remaining memory before crash occurs).

I don't think its related to #8961 as this is running on nrf, and I can't see the suspect code in that port.

tannewt, 2024-02-26:
> Looking at the flash code, it looks like a scratch flash sector is used as a buffer when memory is low - could this be the issue?

Ya, I suspect that is the issue or something similar.

tannewt, 2024-02-29:
I wasn't able to reproduce this on a Feather nrf52840 Express. If it can't allocate to RAM for the scratch buffer it will use the very end of the SPI flash. Maybe the flash on the XIAO can't do that?

Here is my output:
```
mem_free: 1552
opening file
writing file
flushing file
closing file
mem_free: 1024
opening file
writing file
flushing file
closing file
Traceback (most recent call last):
  File "code.py", line 24, in <module>
MemoryError: memory allocation failed, allocating 512 bytes
```

I tested with this code: https://github.com/adafruit/circuitpython/pull/8994

tannewt, 2024-03-05:
@furbrain I'm going to close this because I couldn't reproduce it. Please post here again if it is still happening. It may be board specific.

closed 2024-03-05 by tannewt