adafruit/circuitpython #8980: Can't Access SD over Web Workflow in Beta 2

makermelissa, 2024-02-24:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
import os
import storage

storage.remount("/", readonly=False)

try:
    os.mkdir("/sd")
except OSError:
    pass  # It's probably 'file exists', OK to ignore

storage.remount("/", readonly=True)
```

### Behavior

Above Code is in boot.py. When going to http://cpy-da7e84.local/fs/#/sd/, I get the following error in the console:
```
VM103:1 Uncaught (in promise) SyntaxError: Unexpected end of JSON input
    at refresh_list (directory.js:6:154)
```

### Description

I noticed this error in the full editor. Was going to attempt to fix, so I thought I'd see how it was done on the device only version and noticed the error happened here too. I tried both with CIRCUITPY and after I ejected CIRCUITPY with the same result. The only file in the SD folder is placeholder.txt.

tannewt, 2024-02-26:
@makermelissa What code.py are you running? It needs to mount the SD card using the native classes. We don't auto-mount SD cards yet. On reload, the SD card won't be available while the VM is stopped and restarted. The reload won't happen while a file is being uploaded.

I filed and issue for auto-mounting here: https://github.com/adafruit/circuitpython/issues/8678

makermelissa, 2024-03-04:
@tannewt pretty sure it's just fancy camera example: https://github.com/adafruit/Adafruit_CircuitPython_PyCamera/tree/main/examples/camera

jepler, 2024-03-04:
I tested with beta2 and did see some flakiness even though I initially thought it was working.

boot_out.txt:
```
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit Camera with ESP32S3
Board ID:adafruit_esp32s3_camera
UID:0740D1DC2B0D
[2K[0Gboot.py output:
```

boot.py:
```py
# SPDX-FileCopyrightText: 2023 Jeff Epler for Adafruit Industries
# SPDX-FileCopyrightText: 2023 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense

"""Automatically create the /sd mount point at boot time"""

import os
import storage

storage.remount("/", readonly=False)

try:
    os.mkdir("/sd")
except OSError:
    pass  # It's probably 'file exists', OK to ignore

storage.remount("/", readonly=True)
```

code.py:
```
import time
import adafruit_pycamera

cam = adafruit_pycamera.PyCameraBase()
cam.mount_sd_card()

print("SD card mounted. Check web workflow")
cam.display_message("SD\nmounted")

while True:
    time.sleep(1)
```

[image]

I also checked that the directory listing appears with the "fancy" camera demo.

**however** I noticed that there is definitely flakiness. I tried to load a small file at `http://..../fs/sd/config.txt` and many times it 404's instead of returning file content:
```
$ for i in `seq 12`; do curl -s 'http://10.0.3.69:8080/fs/sd/cmdline.txt' -H 'Authorization: Basic ...' || echo 'fail'; done
console=serial0,115200 console=tty1 root=PARTUUID=2fed7fee-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
console=serial0,115200 console=tty1 root=PARTUUID=2fed7fee-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
console=serial0,115200 console=tty1 root=PARTUUID=2fed7fee-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
console=serial0,115200 console=tty1 root=PARTUUID=2fed7fee-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
console=serial0,115200 console=tty1 root=PARTUUID=2fed7fee-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```
(7 of 12 requests failed in this case, and printed no output)

Getting boot_out.txt in the same way, 100% success on 60 tries so it's something that affects access at `/sd` but not at the root. 

It's seems to be related to mount points, not subdirectories: I created "/notsd/placeholder.txt" and also had no failures in >60 tests.

Getting the json directory data displays similar behavior, `/sd` fails about half the time:
```
for i in `seq 12`; do curl -s 'http://10.0.3.69:8080/fs/sd/' -H "Accept: application/json" -H 'Authorization: Basic ...' | wc ; done
      0     346    4985
      0       0       0
      0     346    4985
      0       0       0
      0     346    4985
      0       0       0
      0     346    4985
      0       0       0
      0     346    4985
      0       0       0
      0     346    4985
      0       0       0
```

jepler, 2024-03-04:
This is a locking + background tasks problem.

The camera demo is pretty frequently writing to the display bus, which is on the SPI bus shared with the SD card. During this long-running SPI request, background tasks are serviced.

If the SPI bus happens to be locked when the workflow http handler gets called from a background task, then `sdcardio_sdcard_readblocks` will indicate an error code because `lock_and_configure_bus` fails. This bubbles up into giving a 404 response for e.g., getting file content from the SD card.

That is why the problem did not manifest during trivial testing but makes access to `/sd` from the "fancy" camera demo extremely unreliable.

PR #9005 (2024-03-05): Postpone interacting with the web workflow if a SPI bus is locked
I tested this by fetching a .txt file repeatedly using curl while running the fancy camera demo. (100+ times without failure). I also repeatedly loaded the filesystem view http://.../fs/#/sd/ which worked 10+ times without failure, but does take some time (multiple seconds) to show a listing with a few dozen files.

(I suspect there's an accidentally quadratic behavior in oofatfs to stat every file in a directory, because it repeatedly does a linear search of the directory for the stat information of each file, but that's not an [... cut]
commit c0064dcf (2024-03-05): Postpone interacting with the web workflow if a SPI bus is locked

I tested this by fetching a .txt file repeatedly using curl while
running the fancy camera demo. (100+ times without failure). I also
repeatedly loaded the filesystem view http://.../fs/#/sd/ which worked
10+ times without failure, but does take some time (multiple seconds) to
show a listing with a few dozen files.

(I suspect there's an accidentally quadratic behavior in oofatfs to stat
every file in a directory, because it repeatedly does a linear search of
the directory for the stat information of each file, but that's not a [... cut]
closed 2024-03-05 by dhalbert