adafruit/circuitpython #8986: Web workflow on CP9 file browser is always read-only

VPTechOps, 2024-02-27:
### CircuitPython version

```python
9.0.0-beta.2-10-g54e78c89fc
9.0.0-beta.0
```

### Code/REPL

```python
Any, even Hello World.
```

### Behavior

Web workflow to http://192.168.0.66/fs/ results in file list with message "ℹ️ USB is using the storage. Only allowing reads. Try ejecting the CIRCUITPY drive. See [Getting Started with Web Workflow](https://learn.adafruit.com/3654/device-setup#disabling-usb-mass-storage-3125964) for details." at the top of the list **even though the device is not connected through USB**.

### Description

This happens on MatrixPortal S3 and Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM. Both device types work correctly on CP 8.2.10 and earlier, but none of the CP 9 versions. I think it also happens on MatrixPortal M4.

RetiredWizard, 2024-02-27:
@VPTechOps I was not able to reproduce this however I did notice that the default behavior of CircuitPython is for WebWorkflow to be read-only.

In order to use WebWorkflow without getting the message you are indicating you must create/edit a boot.py file on the device that contains the following:
```py
import storage
storage.remount("/",False)
```
That is obviously a problem if you don't have the device connected to a computer.

anecdata, 2024-02-27:
I am able to reproduce this. QT Py S3 and Feather S3 TFT both get read-only when not connected to host when using 9.0.0-beta.2. But I tried the Feather on 8.2.10, and it behaves as expected: when not connected to host, web workflow file editor files are writable, with no `storage.remount` necessary.

RetiredWizard, 2024-02-27:
I have the feeling that earlier versions of CP were not strictly enforcing the "writing from one source only" requirement which may have resulted in some web workflow file corruption issues.

RetiredWizard, 2024-02-27:
@anecdata Can you confirm the contents of boot.py on the boards you tested?

anecdata, 2024-02-27:
But if it's not connected to host, web workflow is the one source.

RetiredWizard, 2024-02-27:
I believe CP can't currently dynamically change the boot mode and you could plug the device in to a computer while CP is running so CP has to assume if the boot mode is not configured with the FALSE option you might write from two sources.

anecdata, 2024-02-27:
I removed code.py, boot.py, safemode.py, and repl.py. Same behavior: works (writable) on 8.2.10, but not on 9.0.0-beta.2 when not on host.

RetiredWizard, 2024-02-27:
That is the behavior I expected, that is without the existence of a boot.py containing `storage.remount("/",False)` neither code running on the device or web workflow would have write access to the flash on the device.

Maybe my expectation is wrong though :grin:

anecdata, 2024-02-27:
I tried with 8.2.10, first using web workflow unconnected to host with files writable. Then I plugged it into a computer without resetting the board and tried to edit and couldn't (timeout). Refreshing the file editor page showed that it recognized the host and was read-only.

VPTechOps, 2024-02-27:
I have no boot.py files in any of this testing. I'll also point out that when everything in /fs/ is read-only, the full editor /code/ still allows read-write.

PR #8994 (2024-03-04): Grab FS lock to test writability
Checking is_writable_from_python always follows USB policy even if it is unplugged. Fixes #8986. It also makes writing from Python work when not on USB. It'll be interesting to see if folks have trouble with code only working off-USB.

Also does bare minimum for #8961.
commit ddf9aec0 (2024-03-04): Grab FS lock to test writability

Checking is_writable_from_python always follows USB policy even if
it is unplugged. Fixes #8986

Also does bare minimum for #8961.
closed 2024-03-04 by dhalbert