adafruit/circuitpython #8987: Internal watchdog timer expired error on MatrixPortal S3

BlitzCityDIY, 2024-02-27:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit MatrixPortal S3 with ESP32S3 # also on beta.1
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2023 Liz Clark for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import os
import gc
import ssl
import time
import wifi
import socketpool
import adafruit_requests
import adafruit_display_text.label
import board
import terminalio
import displayio
import framebufferio
import rgbmatrix
import adafruit_json_stream as json_stream
import microcontroller
from adafruit_ticks import ticks_ms, ticks_add, ticks_diff
from adafruit_datetime import datetime, timedelta
import neopixel

displayio.release_displays()

# font color for text on matrix
font_color = 0xFFFFFF
# your timezone UTC offset and timezone name
timezone_info = [-4, "EDT"]
# the name of the sports you want to follow
sport_name = ["football", "baseball", "hockey", "basketball"]
# the name of the corresponding leages you want to follow
sport_league = ["nfl", "mlb", "nhl", "nba"]
# the team names you want to follow
# must match the order of sport/league arrays
# include full name and then abbreviation (usually city/region)
team0 = ["New England Patriots", "NE"]
team1 = ["Boston Red Sox", "BOS"]
team2 = ["Boston Bruins", "BOS"]
team3 = ["Boston Celtics", "BOS"]
# how often the API should be fetched
fetch_timer = 300 # seconds
# how often the display should update
display_timer = 30 # seconds

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness = 0.3, auto_write=True)

# matrix setup
base_width = 64
base_height = 32
chain_across = 2
tile_down = 2
DISPLAY_WIDTH = base_width * chain_across
DISPLAY_HEIGHT = base_height * tile_down
matrix = rgbmatrix.RGBMatrix(
    width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, bit_depth=3,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
[... 330 lines cut ...]
    # pylint: disable=broad-except
    except Exception as Error:
        print(Error)
        time.sleep(10)
        gc.collect()
        time.sleep(5)
        microcontroller.reset()
```

### Behavior

Initially the code runs properly for a few minutes but then it hard resets and goes into safe mode with this error in the serial monitor:
```
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

This occurs when connected to USB on a computer or a separate USB-PD power supply. This behavior does not occur with 9.0.0 beta.0. I was running that build for the past week and everything ran smoothly.

BlitzCityDIY, 2024-02-29:
i tested with absolute newest (`Adafruit CircuitPython 9.0.0-beta.2-11-gf4166ce3e5 on 2024-02-29; Adafruit MatrixPortal S3 with ESP32S3`)/PR8990 since that had fixed some MEMENTO issues but the error is still occurring

tannewt, 2024-03-01:
I was able to reproduce this yesterday. It isn't clear what is going wrong so I've got more debugging to do.

dhalbert, 2024-03-01:
I am reproducing watchdog timer crashes after five or so hard restarts due to errors caused by my not having the logo files. I am trying to trim to a small reproducible example

czei, 2024-03-01:
I was able to reproduce this bug just running a variation of the BitMap demo:  https://learn.adafruit.com/circuitpython-display-support-using-displayio/draw-pixels

dhalbert, 2024-03-01:
I am pruning the original-post example and still getting hard crashes (without printing of exceptions), and also seeing safe-mode after up to a dozen restarts.

But most significantly, I changed the name of one of the wifi parameters to `xCIRCUITPY_WIFI_SSID` to prevent the web workflow from starting up. After that, I see the exceptions printed, and am not seeing safe mode. So I think it may be related to wifi workflow.

czei, 2024-03-01:
Good progress!  Thanks.

tannewt, 2024-03-02:
Commenting out `_PM_row_handler` also prevents the problem. That is the interrupt that updates the matrix.

PR #9008 (2024-03-06): Improve RGBMatrix reliability
The guard against running when flash is disabled isn't perfect and may lead to calling the row handler anyway. We were crashing due to mp_hal_delay_us being in flash. So, move it to RAM.

Also move the panic handler to IRAM so we get nice output when this happens.

Turn off JTAG by default so its pins can be used by code.py.

Fixes #8987
commit e0c6302b (2024-03-06): Improve RGBMatrix reliability

The guard against running when flash is disabled isn't perfect and
may lead to calling the row handler anyway. We were crashing due
to mp_hal_delay_us being in flash. So, move it to RAM.

Also move the panic handler to IRAM so we get nice output when this
happens.

Turn off JTAG by default so its pins can be used by code.py.

Fixes #8987
closed 2024-03-06 by dhalbert