adafruit/circuitpython #8988: Timeout error went from `116: [Errno 116] ETIMEDOUT` to `-116`

justmobilize, 2024-02-28:
### CircuitPython version

```python
works: adafruit-circuitpython-unexpectedmaker_feathers3-en_US-20240215-no-branch-8291467.uf2
_note there are a few merges between these two_
doesn't work: adafruit-circuitpython-unexpectedmaker_feathers3-en_US-20240216-main-PR8935-696aa83.uf2
```

### Code/REPL

```python
# Found the issue re-testing my code for this PR: https://github.com/adafruit/Adafruit_CircuitPython_AzureIoT/pull/63
```

### Behavior

Updated `adafruit_minimqtt` to tring the error before trying to handle it

Previous handling:
```
25.242: INFO - Received registration results on topic $dps/registrations/res/200/?$rid=FeatherS3 - {...}
[Errno 116] ETIMEDOUT
[Errno 116] ETIMEDOUT
27.266: DEBUG - Loop timed out after 2 seconds
```
New Handling:
```
38.707: INFO - Received registration results on topic $dps/registrations/res/200/?$rid=FeatherS3 - {...}
-116
Traceback (most recent call last):
  File "code.py", line 41, in <module>
  File "/lib/adafruit_azureiot/iotcentral_device.py", line 163, in connect
  File "/lib/adafruit_azureiot/device_registration.py", line 214, in register_device
  File "/lib/adafruit_azureiot/device_registration.py", line 162, in _wait_for_operation
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 1060, in loop
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 1090, in _wait_for_msg
MMQTTException:
```

justmobilize, 2024-02-29:
Unsure. The test I was running fails on a Pico W:
```
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Raspberry Pi Pico W with rp2040
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WiFi...
Connected to WiFi!
Year seems good, skipping set time.
Connecting to Azure IoT Central...
89.781: DEBUG - Attempting to connect to MQTT broker (attempt #0)
89.784: DEBUG - Attempting to establish MQTT connection...
89.788: INFO - Establishing a SECURE SSL connection to global.azure-devices-provisioning.net:8883
Traceback (most recent call last):
  File "code.py", line 41, in <module>
  File "/lib/adafruit_azureiot/iotcentral_device.py", line 163, in connect
  File "/lib/adafruit_azureiot/device_registration.py", line 211, in register_device
  File "/lib/adafruit_azureiot/device_registration.py", line 109, in _connect_to_mqtt
  File "/lib/adafruit_minimqtt/adafruit_minimqtt.py", line 533, in connect
  File "/lib/adafruit_minimqtt/adafruit_minimqtt.py", line 591, in _connect
  File "/lib/adafruit_minimqtt/adafruit_minimqtt.py", line 352, in _get_connect_socket
MemoryError:

Code done running.
```

dhalbert, 2024-02-29:
@justmobilize The test might work on a Pico W if you mpy-cross it first to get an .mpy

@jepler I remember you doing something with negative error numbers in #8926 or elsewhere but I cannot find it.

justmobilize, 2024-02-29:
Okay, I can look at that 1st thing tomorrow, maybe tonight depending on how things go

justmobilize, 2024-02-29:
Actually, had just enough time. confirmed it fails the same way on a Pico W in both `9.0.0-beta.0` and `9.0.0-beta.2`

Also what's odd is it seems to be specific to some flow. when I get a NTP timeout, it's the right error...

dhalbert, 2024-02-29:
The error on line 1090 above is from this library code:
```py
            try:
                res = self._sock_exact_recv(1, timeout=timeout)
            except OSError as error:
                if error.errno in (errno.ETIMEDOUT, errno.EAGAIN):
                    # raised by a socket timeout if 0 bytes were present
                    return None
                raise MMQTTException from error   # <-- line 1090
```
Note that the library was just updated today to use ConnectionManager, so the line numbers will be different if you re-run with the new library, though I don't think the error handling will change.

The new library is smaller and might not cause a `MemoryError` when imported.

justmobilize, 2024-02-29:
Getting everything to mpy did work (as in no memory error), and failed with an MQTT exception in new and old versions (and it's a MQTT exception, since it's -116).

Sorry I wasn't clear in my previous message.

PR #9006 (2024-03-05): ssl: Give correct errno value for "timed out" exceptions
Incorrect error handling in send/recv would raise an OSError with an incorrect (negative) code.

It's likely that this bug was always happening in the Pico W implementation, which became the basis of the current shared implementation.

Push handling of WANT_{READ,WRITE} down into mbedtls_raise_error and use it in recv_into and send.

Tested by connecting to google.com:443, sending nothing, and trying to read a byte:

```py
import socketpool, ssl, time, wifi
socket = socketpool.SocketPool(wifi.radio)
ctx = ssl.SSLContext()
with ctx.wrap_s [... cut]
commit 3e029a9d (2024-03-05): ssl: Fix "timed out" exceptions

Incorrect error handling in send/recv would raise an OSError with
an incorrect (negative) code.

It's likely that this bug was always happening in the Pico W
implementation, which became the basis of the current shared
implementation.

Push handling of WANT_{READ,WRITE} down into mbedtls_raise_error
and use it in recv_into and send.

Tested by connecting to google.com:443, sending nothing, and trying
to read a byte:

```py
import socketpool, ssl, time, wifi
socket = socketpool.SocketPool(wifi.radio)
ctx = ssl.SSLContext()
with ctx.wrap_socket(socket.socket()) a [... cut]
closed 2024-03-05 by dhalbert