adafruit/circuitpython #8995: RuntimeError: buffer too small error during long-term bluetooth scan

mwrohloff, 2024-03-01:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit Feather nRF52840 Express with nRF52840
```

### Code/REPL

```python
import time
import adafruit_ble
import adafruit_ble.advertising

ble = adafruit_ble.BLERadio()

connection = None

while True:
    print("# ble scanner")
    for adv in ble.start_scan(timeout=5):
        if adv.complete_name:
            if "E14_" in adv.complete_name:
                print(f'{{"E14_2_NAME": "{adv.complete_name}", "RSSI": {adv.rssi}}}')
                break
            if "E-14" in adv.complete_name:
                print(f'{{"E-14_NAME": "{adv.complete_name}", "RSSI": {adv.rssi}}}')
                break
    time.sleep(1)

    # Stop scanning whether or not we are connected.
    ble.stop_scan()
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 17, in <module>
  File "adafruit_ble/init.py", line 289, in start_scan
  File "adafruit_ble/advertising/init.py", line 275, in init
  File "adafruit_ble/advertising/init.py", line 51, in decode_data
RuntimeError: buffer too small

### Description

the code runs fine for about half an hour and then crashes with the memory RuntimeError. Without the break statement the code crashes after some seconds.

dhalbert, 2024-03-04:
I tried this simpler example:
```py
import time
import adafruit_ble
import adafruit_ble.advertising

ble = adafruit_ble.BLERadio()

while True:
    print("# ble scanner")
    for adv in ble.start_scan(timeout=5):
        if adv.complete_name:
            print(adv.complete_name, adv.rssi)
    time.sleep(1)

    # Stop scanning whether or not we are connected.
    ble.stop_scan()
```
I ran it for a while and it did not crash. You are looking at advertisements from a particular device. Is it an easily obtained commercial device? Maybe that device has a malformed advertisement, with a very large length field of some kind?

dhalbert, 2024-03-04:
Immediately after posting the above, I saw this, after running for a while. Have you seen something similar?
```py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "i8995.py", line 9, in <module>
  File "adafruit_ble/__init__.py", line 275, in start_scan
MemoryError: memory allocation failed, allocating 51370 bytes
```

mwrohloff, 2024-03-04:
I think the "MemoryError: memory allocation failed, allocating 51370 bytes" or similar message appeared with CircuitPython 8.x that I tried before. Then I have read something about a memory fix in CircuitPython 9.0.0beta so I have tried. Here comes the "RuntimeError: buffer too small" error output. The device is a commercially available bike component based on a Würth Proteus 2 (nrf52832). Do you have some test code for me to get the raw advertisement data that might help localize the problem?

dhalbert, 2024-03-06:
I did some debugging and am seeing bad data showing up as part of an advertising packet., causing some data sizes that are much too large. This might be some kind of internal problem, or it might be a corrupted packet. Will continue on this.

PR #9014 (2024-03-06): shared-module/_bleio/ScanResults.c: make ScanEntry handling be atomic
- Fixes #8995 

Advertisement-data read/write `ScanEntry.c` was not atomic. One side of that was done during an interrupt.

Tested with last test program in #8995. Ran for half an hour or so without failure. Previously would fail within a minute or two.

Also made minor fix re signedness of RSSI: no behavior change.
closed 2024-03-06 by tannewt