adafruit/circuitpython #8996: http://<ip4_addr>/code/ does not connect, it drops back to the wifi/bluetooth/usb selection

davs2rt, 2024-03-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4
```

### Code/REPL

```python
# web workflow, not Python, not the REPL: 
http://10.0.0.1/code  # but use a valid IPv4 address for your unit.
```

### Behavior

Instead of going to the device, I am presented with the 
[image]
dialog which sends me to circuitpython.local if I select Wi-Fi

### Description

According to [this Adafruit example](https://learn.adafruit.com/circuitpython-with-esp32-quick-start/connecting-via-web-browser) it should be possible to connect to web workflow using an IP address.  Depending on the network, I find the mdns resolution of cpy-xxxxxx.local and circuitpython.local addresses to be intermittent at best, and total failures at worst.  It would be useful to bypass mdns in these cases and just use the known IP address, as the documentation indicates.

### Additional information

My C is lousy at best, but it looks like bool _origin_ok(_request *request), line 497 of     circuitpython/supervisor/shared/web_workflow/web_workflow.c does not perform this acceptance test, and this is why it fails.

davs2rt, 2024-03-02:
Added information: 
http://ipv4_address/  does go to the welcome page
http://ipv4_address/fs/ does go to the file browser
http://ipv4_address/cp/serial/ does go to the limited serial interface

tannewt, 2024-03-04:
Please try an absolute newest build. 8.2.9 is pretty old at this point.

davs2rt, 2024-03-09:
Retested on
Adafruit CircuitPython 9.0.0-rc.0 on 2024-03-08; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4
Board ID:seeed_xiao_esp32c3
UID:43588150EB0B

tested http://10.1.99.10 (the correct IP address) for /, /code/, /fs/ and /cp/serial/   everything works.  
tested http://cpy-50eb0b.local for /, /code/, /fs/, and /cp/serial/   everything works.  

Okay to close as fixed in 9.0

tannewt, 2024-03-09:
Thanks for retesting!

closed 2024-03-09 by tannewt