adafruit/circuitpython #8998: Atom S3 display has red and blue colors reversed

scarolan, 2024-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; M5Stack AtomS3 with ESP32S3
```

### Code/REPL

```python
import displayio
import board

# Display setup
display = board.DISPLAY
display_group = displayio.Group()
display.root_group = display_group

# Try this with a red image, it shows up blue
# Create a TileGrid to hold the bitmap
logo_tile_grid = displayio.TileGrid(logo_bitmap, pixel_shader=logo_palette)

# Position the image in the upper middle of the display
logo_tile_grid.x = 90
logo_tile_grid.y = 5  # Adjust this value based on your image size and desired position

# Add the image to the display group
display_group.append(logo_tile_grid)

# Load your image using adafruit_imageload
thermometer_bitmap, thermometer_palette = adafruit_imageload.load("/thermometer.gif", 
                                                      bitmap=displayio.Bitmap, 
                                                      palette=displayio.Palette)
```

### Behavior

It runs, but the colors are reversed

dhalbert, 2024-03-04:
This was added by @jins-tkomoda in #8474. @jins-tkomoda have you seen this issue?

jins-tkomoda, 2024-03-04:
@dhalbert 

I will check on it over the weekend.
I did not check to display images, so the process may be BGR order in the middle of the process.

scarolan, 2024-03-04:
菰田さん、ありがとうございました！

closed 2024-03-04 by dhalbert