adafruit/circuitpython #9010: watchdog.timeout changed from seconds (per docs) to milliseconds in CP 9

anecdata, 2024-03-06:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.2-21-gc91b20fd96 on 2024-03-05; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import time
from microcontroller import watchdog as w
from watchdog import WatchDogMode

time.sleep(3)  # wait for serial

print(f'{time.monotonic()=}')
print('setup watchdog...')
w.timeout = 3600  # should be 1 hour
w.mode = WatchDogMode.RAISE
while True:
    print(f'{time.monotonic()=}')
    time.sleep(1)
```

### Behavior

```
[tio 20:49:04] Connected
time.monotonic()=6.511
setup watchdog...
time.monotonic()=6.513
time.monotonic()=7.513
time.monotonic()=8.513
time.monotonic()=9.514
Traceback (most recent call last):
  File "code.py", line 13, in <module>
<WatchDogTimeout>

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
this used to work in 8.2.x (8.2.8 for sure)

### Description

Perhaps related to #7718

### Additional information

Also:
```py
>>> w.timeout=4294967295
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: timeout must be <= 4294967295
```

dhalbert, 2024-03-06:
(Paraphrased from my comment in discord:) Looking at https://github.com/adafruit/circuitpython/pull/7718/files#diff-d40fe0f9108540afc8b776d5f3294b72ef8826917062ce458558bffb9b398119 from #7718
it's setting a time in msec from a float secs. There's a cast on the left and no cast on the right, but it will be casting the float to a uint32_t in either case. It should be multiplying by 1000, but it's not.

vladak, 2024-03-06:
Ran into this on `Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit Feather ESP32 V2 with ESP32`. Which CP 9.0.x release this will be fixed in ?

dhalbert, 2024-03-06:
> Which CP 9.0.x release this will be fixed in ?

In the next release after beta.2.

vladak, 2024-03-06:
> It will also be available in "Absolute Newest" at https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin for the build that has `PR9012` in its name and any thereafter. The `PR9012` builds are in process right now.

Thanks, will give it a try. I don't mind running on the "bleeding edge".

PR #9012 (2024-03-06): Tweak watchdog
* Make setting timeout "feed" the watchdog.
* Fix #9010, timing on ESP