adafruit/circuitpython #9021: Unable to set simpleio.tone frequency to zero on RP2040

SalamCytron, 2024-03-08:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-rc.0 on 2024-03-08; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import board
import simpleio

# Define pin connected to piezo buzzer.
buzzer_pin = board.GP21

while True:
    simpleio.tone(buzzer_pin, 800, 0.5)
    simpleio.tone(buzzer_pin, 0, 0.5)
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "simpleio.py", line 74, in tone
ValueError: GP21 in use
```

### Description

Hi,

I am having a problem unable to set the simpleio.tone frequency to zero. On Circuitpython 8.2.x and 9.0.0 Beta 0, the code is working. 
May I know if it was on purpose in the latest update or if it will remain the same in future releases?

Your reply would greatly help standardize our code in the future.
Thank you!

kbsriram, 2024-03-08:
The trace above might be an issue with `pwmio` not releasing resources in its constructor (rather than a `simpleio` issue). The test case is in a file `tc.py`:
```
import board
import pwmio

def acquire():
    try:
        with pwmio.PWMOut(board.GP21, frequency=0) as p:
            pass
    except ValueError as ve:
        print(ve)
```
from the repl
```
>>> import tc
>>> tc.acquire()
Invalid frequency
>>> tc.acquire()
GP21 in use
>>> import gc
>>> gc.collect()
>>> tc.acquire()
Invalid frequency
```

I'm guessing this might be from how [`shared-binding/PWMOut.c`](https://github.com/adafruit/circuitpython/blob/60bd748b96dcc739dc27e0e21548613fa07f9dbb/shared-bindings/pwmio/PWMOut.c#L285) expects to be used - it doesn't expose an explicit `init` or `__enter__` to acquire resources, so any resources acquired when a constructor fails have to be handled directly within the constructor, as the `__exit__` won't get called, and it has to wait for `__del__` to actually release the resource.

The actual pin code seems to have been added quite [some time back](https://github.com/adafruit/circuitpython/commit/33bbb8b1f48ba1263bddb103f49720201a139a91) - perhaps it's been relying on GC to clean up resources in this situation all along, and surfaced with the recent GC changes?

dhalbert, 2024-03-08:
This is a bug in the CircuitPython caused by some recent revamping of PWMOut deinitalization. The constructor should clean up on error. Will fix.

PR #9024 (2024-03-08): raspberrypi/common-hal/pwmio/PWMOut.c: don't claim pin before error checking
- Fixes #9021
closed 2024-03-08 by tannewt