adafruit/circuitpython #9026: QT PY ESP32 Pico boot loop in 9.0.0 rc 0

RetiredWizard, 2024-03-09:
### CircuitPython version

```python
adafruit-circuitpython-adafruit_qtpy_esp32_pico-en_US-20240305-main-PR9004-4a335af.bin
```

### Code/REPL

```python
Boot loops, no code.py executed
```

### Behavior

Starting with the 20240305-main-PR9004-4a335af.bin build from the S3 list ESP32 boards that don't support CIRCUITPY over usb boot loop.

### Description

After the board is reset, the first boot messages appear and then every ~10-15 seconds the board restarts again. The "Serial console setup" message is never displayed.

```ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:312
load:0x40078000,len:14164
load:0x40080400,len:4
load:0x40080404,len:3268
entry 0x40080580
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:312
load:0x40078000,len:14164
load:0x40080400,len:4
load:0x40080404,len:3268
entry 0x40080580
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:312
load:0x40078000,len:14164
load:0x40080400,len:4
load:0x40080404,len:3268
entry 0x40080580
ets Jul 29 2019 12:21:46
```

### Additional information

I first noticed this while using the Huzzah32 breakout firmware on an unsupported ESP32 board and then confirmed the error occurred using the QT PY esp32 pico firmware on the actual device.

I was expecting to see the same issue on a QT PY esp32 C3, however that board/firmrware works fine with 9.0.0 rc 0.

vladak, 2024-03-09:
Happens to me as well on Feather ESP32 V2.

theBLUSpycrab, 2024-03-10:
Exactly the same issue on ItsyBitsy ESP32.

PR #9039 (2024-03-11): Fix bug in TLSF change that fails on ESP32 only
It is likely due to the multiple heap setup on ESP32. An allocation fails into one heap because the size is too large but caused an assertion instead of failing over to the next heap.

Fixes #9026
commit f3e6b82a (2024-03-11): Fix bug in TLSF change that fails on ESP32 only

It is likely due to the multiple heap setup on ESP32. An allocation
fails into one heap because the size is too large but caused an
assertion instead of failing over to the next heap.

Fixes #9026
closed 2024-03-11 by dhalbert