adafruit/circuitpython #9027: Build error CPY 9.0.0-rc0

PaulskPt, 2024-03-09:
### CircuitPython version

```python
rc0
```

### Code/REPL

```python
-
```

### Behavior

build crash

### Description

I downloaded and installed rc0. During build for a new board lolin S3 PRO the build process stops with error: without rule to make target "lib/tlsf/tlsf.c". (My translation from text in Portuguese). I checked the lib/tlsf folder. It is empty. How do I proceed to correct this error?

PaulskPt, 2024-03-09:
No problem to build with CPY 9.0.0-alpha.1

PaulskPt, 2024-03-09:
Solved: copied the library folder from here, in this repo on github

dhalbert, 2024-03-09:
You can do `make fetch-port-submodules` in the `port/espressif` directory or `make fetch-all-submodules` at the top.

PaulskPt, 2024-03-09:
Thank you Dan. I deleted Circuitpython, next cloned the latest. Did all the further steps. Then I was able to build BOARD for lolin_sr_pro. Flashed it. REPL shows: ` ` ` Adafruit CircuitPython 9.0.0-rc.0-3-gad6bc74f28 on 2024-03-09; LOLIN S3 PRO 16MB Flash 8MB PSRAM with ESP32S3` ` `

closed 2024-03-09 by PaulskPt