adafruit/circuitpython #9028: Linker error when compiling CP 8.2.10 with double (complex) float enabled

h-milz, 2024-03-09:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.10 on 2024-02-14; Adafruit Feather RP2040 with rp2040
(similarly on feather-m4-express so I suspect it's a general issue)

I am trying to build a custom image for the Feather RP2040, enabling MICROPY_FLOAT_IMPL_DOUBLE, MICROPY_PY_CMATH and ULAB_SUPPORTS_COMPLEX. My local changes are in boards/adafruit_feather_rp2040/mpconfigboard.h (see below), and I noticed that since double was ditched in 2018 (?), apparently the newer gcc-arm-none-eabi-10.3.1 (on Ubuntu 22.04 LTS) is more critical about various thing like implicit float to double conversions. I also noticed that in many source files, constants are deliberately declared as float, like 0.0f, which leads to many compiler errors when handling variables that are declared as mp_float_t. I do not think that hardcoding constants as fixed types is very clean, so in order to fix this I replaced a number (maybe not all) occurences of float by explicit type casts, e.g. MP_CONST_FLOAT(0.0), after which the code compiles cleanly with MICROPY_FLOAT_IMPL_DOUBLE set or not set. I also had to ignore some warnings like float-equal and overflow.

So far so good. But in the last stage, the linker now throws many errors like

/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: build-adafruit_feather_rp2040/extmod/ulab/code/ndarray.o: in function `ndarray_copy_view_convert_type':
ndarray.c:(.text.ndarray_copy_view_convert_type+0x76): undefined reference to `__wrap_round'

Apparently, this behaviour is triggered by setting 

#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)

A similar thing happens when compiling feather-m4-express: 

/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /tmp/firmware.elf.jy2M1p.ltrans113.ltrans.o:<artificial>:(.text.ndarray_binary_power+0x268): more undefined references to `pow' follow

My extra settings in mpconfigboard.h are: 

#ifdef MICROPY_FLOAT_IMPL
#undef MICROPY_FLOAT_IMPL
#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)
#endif

#ifdef MICROPY_PY_CMATH
#undef MICROPY_PY_CMATH
#define MICROPY_PY_CMATH  (1)
#endif

#ifdef ULAB_SUPPORTS_COMPLEX
#undef ULAB_SUPPORTS_COMPLEX
#define ULAB_SUPPORTS_COMPLEX  (1)
#endif

#pragma GCC diagnostic ignored "-Wfloat-equal"
#pragma GCC diagnostic ignored "-Woverflow"

Thank you!
```

### Code/REPL

```python
-- none --
```

### Behavior

-- none --

dhalbert, 2024-03-09:
Have you tried on `main` instead? 9.0.0 is imminent, and includes a big jump in gcc versions and MicroPython updates. I think that is where to spend the time to get this to work.

jepler, 2024-03-09:
Related to this, I have these two commits in one of my trees (port-posix): efcad6457b715dbdb3f38582467e1379ccc8b1b8 e6aac4014eb9b77c641d3fb3ccfa9c0580a3c584

Feel free to pull them into a separate branch and PR it, or use it as the basis of your own changes -- the port-posix work is not progressing at the moment.

h-milz, 2024-03-09:
@dhalbert, same thing with 9.0.0beta2, tried with m4 express: 

/usr/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /tmp/cctg8k9g.ltrans113.ltrans.o:<artificial>:(.text.ndarray_binary_power+0x268): more undefined references to `pow' follow

I also had issues with MP_STATIC_ASSERT in py.obj.h and commented it out - no idea if there are any side effects. 

@jepler, I had to change many more places. Raw git diff below, but it is very possible I did not catch them all. 

`diff --git a/ports/atmel-samd/boards/feather_m4_express/mpconfigboard.h b/ports/atmel-samd/boards/feather_m4_express/mpconfigboard.h
index 0744fc3f92..8ad0068d7d 100644
--- a/ports/atmel-samd/boards/feather_m4_express/mpconfigboard.h
+++ b/ports/atmel-samd/boards/feather_m4_express/mpconfigboard.h
@@ -23,3 +23,25 @@
 // USB is always used internally so skip the pin objects for it.
 #define IGNORE_PIN_PA24     1
 #define IGNORE_PIN_PA25     1
+
+/* local additions for double float and complex handling */
+
+#ifdef MICROPY_FLOAT_IMPL
+#undef MICROPY_FLOAT_IMPL
+#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)
+#endif
+
+#ifdef MICROPY_PY_CMATH
+#undef MICROPY_PY_CMATH
+#define MICROPY_PY_CMATH  (1)
+#endif
+
+#ifdef ULAB_SUPPORTS_COMPLEX
+#undef ULAB_SUPPORTS_COMPLEX
+#define ULAB_SUPPORTS_COMPLEX  (1)
+#endif
+
+#pragma GCC diagnostic ignored "-Wfloat-equal"
+#pragma GCC diagnostic ignored "-Woverflow"
+
+
diff --git a/ports/atmel-samd/common-hal/alarm/time/TimeAlarm.c b/ports/atmel-samd/common-hal/alarm/time/TimeAlarm.c
index a8f408181f..c929e669bc 100644
--- a/ports/atmel-samd/common-hal/alarm/time/TimeAlarm.c
+++ b/ports/atmel-samd/common-hal/alarm/time/TimeAlarm.c
@@ -78,7 +78,7 @@ void time_alarm_callback(void) {
 
 bool alarm_time_timealarm_woke_this_cycle(void) {
     if (SAMD_ALARM_FLAG_TIME_CHK) {
-        mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / 1000.0f;
+        mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / MICROPY_FLOAT_CONST(1000.0);
         if (now_secs > wakeup_time) {
             woke_up = true;
         }
@@ -116,7 +116,7 @@ void alarm_time_timealarm_set_alarms(bool deep_sleep, size_t n_alarms, const mp_
 
     // Compute how long to actually sleep, considering the time now.
     // At least 1 count = 1/16384 sec is necessary.
-    mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / 1000.0f;
+    mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / MICROPY_FLOAT_CONST(1000.0);
     uint32_t wakeup_in_counts = MAX(1, (uint32_t)((wakeup_time - now_secs) * 16384));
 
 
@@ -137,7 +137,7 @@ void alarm_time_timealarm_prepare_for_deep_sleep(void) {
     // In deep sleep mode, prescaler is set to 1024, so that 1 count = 1/32 s.
     // At least 1 count is necessary.
 
-    mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / 1000.0f;
+    mp_float_t now_secs = uint64_to_float(common_hal_time_monotonic_ms()) / MICROPY_FLOAT_CONST(1000.0);
     uint32_t wakeup_in_counts = MAX(1, (uint32_t)((wakeup_time - now_secs) * 32));
     RTC->MODE0.COMP[1].reg = wakeup_in_counts;
     while ((RTC->MODE0.SYNCBUSY.reg & (RTC_MODE0_SYNCBUSY_COMP1)) != 0) {
diff --git a/py/formatfloat.c b/py/formatfloat.c
[... 230 lines cut ...]
--- a/shared-module/usb_cdc/Serial.c
+++ b/shared-module/usb_cdc/Serial.c
@@ -33,8 +33,8 @@
 
 size_t common_hal_usb_cdc_serial_read(usb_cdc_serial_obj_t *self, uint8_t *data, size_t len, int *errcode) {
 
-    const bool wait_forever = self->timeout < 0.0f;
-    const bool wait_for_timeout = self->timeout > 0.0f;
+    const bool wait_forever = self->timeout < MICROPY_FLOAT_CONST(0.0);
+    const bool wait_for_timeout = self->timeout > MICROPY_FLOAT_CONST(0.0);
 
     // Read up to len bytes immediately.
     // The number of bytes read will not be larger than what is already in the TinyUSB FIFO.
@@ -79,8 +79,8 @@ size_t common_hal_usb_cdc_serial_read(usb_cdc_serial_obj_t *self, uint8_t *data,
 }
 
 size_t common_hal_usb_cdc_serial_write(usb_cdc_serial_obj_t *self, const uint8_t *data, size_t len, int *errcode) {
-    const bool wait_forever = self->write_timeout < 0.0f;
-    const bool wait_for_timeout = self->write_timeout > 0.0f;
+    const bool wait_forever = self->write_timeout < MICROPY_FLOAT_CONST(0.0);
+    const bool wait_for_timeout = self->write_timeout > MICROPY_FLOAT_CONST(0.0);
 
     // Write as many bytes as possible immediately.
     // The number of bytes written at once will not be larger than what can fit in the TinyUSB FIFO.
`

h-milz, 2024-03-10:
After a bit of code reading, I think I found an ansatz. py/mpconfig.h, if _FLOAT is set, defines: 

#define MICROPY_FLOAT_C_FUN(fun) fun##f

which chops the appended f from all function names in libm/*.c. Otherwise, if DOUBLE ist set, the define is

#define MICROPY_FLOAT_C_FUN(fun) fun 

which chops nothing. Consequently, the functions in libm/*.c are not found. But they should not be found in the first place, but the functions in libm_dbl. 

Later today, I'll introduce INTERNAL_LIBM = 2 as a hack, and will set up libm_dbl to be used in py/circuitpy_defns.mk:837 and following, and in ./ports/raspberrypi/Makefile:425. It's an ugly hack for now. Hints welcome for a better style. 

Any comments concerning the MP_STATIC_ASSERT thing mentioned before? 

(I also tried to use the toolchain libm saying INTERNAL_LIBM=0 and adding -L for the lib path, but I can't seem to make the linker to find it.)

h-milz, 2024-03-10:
OK it seems I should avoid this hack and try to use libm.a instead. Compiles now, but I'll need to throw out some unneeded modules on M4 Express. In RP2040, I don't get these -wrap-things to link with libm.a.

closed 2024-03-10 by h-milz