adafruit/circuitpython #9029: .devcontainer: Uses incorrect ARM toolchain version

eightycc, 2024-03-09:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 rc0
```

### Code/REPL

```python
n/a
```

### Behavior

n/a

### Description

The gcc compiler installed by `.devcontainer/cortex-m/on-create.sh` is an out of date version of the ARM toolchain for an `x86_64` build host. The version installed is `gcc-arm-none-eabi-10-2020-q4-major`, but should be `arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi` for an `x86_64` build host.

Additionally, support should be provided for other build hosts.

dhalbert, 2024-03-11:
@bablokb would you want to fix this?

bablokb, 2024-03-11:
I am already working on that. You can find a preliminary version here: https://github.com/bablokb/circuitpython/blob/devcontainer-9.x/.devcontainer/cortex-m/create_cortex_m_9.x.sh

Currently, I am trying to track down a problem with devcontainer creation: if I create a container with the default configuration and then start my script after first login manually, everything works. If I run the same script automatically as postCreateCommand, it fails, because the tags won't work as expected (`make fetch-tags` works, but `tools/describe` still gives no result and therefore the compilation of mpy-cross fails).

 Ultimately, I will also reorganize the structure by port (and not by architecture) and make use of `make fetch-port-submodules`, which really speeds up the devcontainer-creation. The above script does this for the raspberrypi-port.

bablokb, 2024-03-11:
@dhalbert : two minor notes regarding your "Building CircuitPython"-guide: it does not mention `make fetch-tags` at all. And secondly, the image on the very first page shows a `git submodule update --init` (which is more of a cosmetic problem).

bablokb, 2024-03-21:
@eightycc 

> Additionally, support should be provided for other build hosts.

This is for Github-codespaces, and as far as I know, they only provide x86_64 hosts.

eightycc, 2024-03-22:
@bablokb Thank you, it's good with me. Didn't know that GH-codespaces had that limitation.

PR #9068 (2024-03-22): Devcontainer update
This adds support for Github-codespaces for CP 9.x.x.

It fixes #9029.

Another major change is that the devcontainer-alternatives are now on a per port basis (major speedup in codespace creation since the codespace only fetches submodules relevant to the selected port). Currently only ports using espressif and cortex-m are supported.
closed 2024-03-22 by dhalbert