adafruit/circuitpython #9040: Web Workflow: SD Folder contents intermittantly only shows placeholder

makermelissa, 2024-03-11:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-rc.0 on 2024-03-08; Adafruit Camera with ESP32S3
Board ID:adafruit_esp32s3_camera
UID:CD4557ADE748
```

### Code/REPL

```python
Fancy Camera Example is running on Camera
```

### Behavior

N/A. No error messages. See Description for more info.

### Description

Occasionally, the SD folder shows incorrect files. I noticed this after a file operation especially when deleting or uploading a file, the only file that will show is placeholder.txt (and ._placeholder.txt). However, refreshing the browser shows the actual files there. It doesn't seem to affect the root file system and I noticed this both with a read only and non-read only file system. 

### Additional information

This could be related to https://github.com/adafruit/circuitpython/issues/8980, as it was also intermittent.

[image]
[image]

tannewt, 2024-03-11:
This happens if you request the sd directory while code is restarting. SD access depends on the user code being active.

makermelissa, 2024-03-11:
Ah ok. Yeah, I just checked and the file ops were causing a code restart. Do you know if there is a way to have it check if code is still restarting so that the correct contents are shown to the user. I could probably make something with the full code editor, but the problem would persist with the built-in one.

tannewt, 2024-03-11:
I don't think there is. My intent was to fix this by having SD cards mount automatically and persist on boards with it built in.

tannewt, 2024-03-12:
Closing in favor of https://github.com/adafruit/circuitpython/issues/8678

closed 2024-03-12 by tannewt