adafruit/circuitpython #9044: SPI Device Manager needs parameter for bit width

kamocat, 2024-03-13:
The [busio.SPI](https://docs.circuitpython.org/en/latest/shared-bindings/busio/#busio.SPI) class has a parameter to set the bit width, but this parameter is not available in the [SPI device manager](https://docs.circuitpython.org/en/latest/shared-bindings/adafruit_bus_device/spi_device/index.html)

kamocat, 2024-03-13:
It seems the implementation could be as simple as passing along the kwargs:

```python
class SPIDevice:
    def __init__(self, bus, cs:digitalio.DigitalInOut|None=None, cs_active_value=False, extra_clocks:int=0, **kwargs):
        self.bus = bus
        self.cs = cs
        self.active_val = cs_active_value
        self.extra_clocks = extra_clocks
        self.config = kwargs
    def __enter__(self):
        assert self.bus.try_lock()
        self.bus.configure(self.config)
        if self.cs is not None:
            self.cs.value = self.active_val
        # FIXME: insert extra clocks
    def __exit__(self):
        if self.cs is not None:
            self.cs.value = not self.active_val
            self.bus.unlock()
```

tannewt, 2024-03-13:
@kamocat PRs to the Python busdevice and the core C implementation would be welcome.

kamocat, 2024-03-13:
I'll consider it.
In the meantime, it seems the busio.SPI class only allows 8 or 9 bits, although the RP2040 supports anywhere from 4 to 16 bits and the ESP32 SPI has some really crazy settings.
I suppose it's because the SAMD21 Sercom peripheral only supports 8 or 9 bits and the NRF52840 *only* supports 8 bit words.

I wonder if the spi write and readinto functions handle 16-bit arrays transparently, without any need to set the bit width?

dhalbert, 2024-03-13:
This is also discussed in
- #8179

kamocat, 2024-03-13:
I'm writing a driver for ADS131M04, which uses 16 bit registers and supports 16, 24, or 32 bit words.

dhalbert, 2024-03-13:
> I'm writing a driver for ADS131M04, which uses 16 bit registers and supports 16, 24, or 32 bit words.

You can just send 8-bit-bytes concatenated, which I'm assuming you know already. Lots of our existing CircuitPython drivers do that for various SPI devices.  You can use `struct` or `.to_bytes` and .from_bytes` to convert to/from integers.

kamocat, 2024-03-13:
Unless anyone actually has a 9-bit SPI device, I think we can close this.

closed 2024-03-13 by kamocat