adafruit/circuitpython #9050: 01space_lcd042_esp32c3 crashing to safe mode on hard reset.

SeanTheITGuy, 2024-03-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.10; 01space_lcd042_esp32c3
Adafruit CircuitPython 9.0.0-rc.0-13-g39b16feaa0-dirty; 01space_lcd042_esp32c3
```

### Code/REPL

```python
print("Hello World!")
```

### Behavior

```ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xd0
load:0x403cc710,len:0x7c8
load:0x403ce710,len:0x2518
entry 0x403cc710

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Heap allocation when VM not running.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.```

### Description

After a fresh flash of 8.2.10 up to and including 9.0.0-rc0.13, CircuitPython will boot perfectly fine, displaying the logo and wifi off info on the LCD. After a power cycle, it also works fine. However, if you push the reset button, it boots with the following:

No amount of ctrl-D soft-boots bring it back to a functional state.  When the board is in this safe-mode crashed state, the LCD is still being updated. If I have pushed a settings.toml to the board with my wifi credentials, I can see the LCD update from No IP to the obtained IP.
![IMG_20240314_121004](https://github.com/adafruit/circuitpython/assets/3707163/21834f5d-a7bf-4ff8-b1d3-205a093dcff6)

tannewt, 2024-03-14:
Do you have an espressif devkit you can test on? I don't have this board to test with.

Are there any log statements before the build line? The actual error happened before the reset.

SeanTheITGuy, 2024-03-14:
I can test, yes.   There are no log statements before the build line, neither via usb uart or hooking an external usb serial device to pins 20/21.
```ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xd0
load:0x403cc710,len:0x7c8
load:0x403ce710,len:0x2518
entry 0x403cc710
``` is the entirety of it.

jepler, 2024-03-14:
you mentioned on discord that this same board works when flashed with the qtpy firmware.

I'd check whether removing display initialization fixes it; if so, it greatly narrows the range of stuff to look at.

jepler, 2024-03-14:
(that seems to be the main difference, aside from the content of the board module)

SeanTheITGuy, 2024-03-14:
In this variant, it looks like the entire board_init() function is dedicated to the construction and initialisation of the display. 
 Commenting out everything inside board_init() does seem to allow it to be reset without crashing, indicating that it is a problem talking to the display.

SeanTheITGuy, 2024-03-15:
This appears to be a hardware issue. As silly as it sounds, if I hold the board (thus giving a high resistance continuity to several pins) it boots stable every time. If I lay the board on a non-conductive surface and push the reset button with a probe, it crashes on reset every time. This is very repeatable.

dhalbert, 2024-03-15:
The schematic for the original board is here: https://github.com/01Space/ESP32-C3-0.42LCD/blob/main/Schematic/ESP32-C3-0.42OED%20Schematic.pdf. I don't see any pull-up resistors for SDA and SCL on the board. Ideally there should be 10k or so pullups somewhere. _Maybe_ they are on the display itself, maybe not. CircuitPython checks for pullups, and might initially fail creating the I2C bus because of that (though I'm not sure why the display would then sort of work).

Try setting `CIRCUITPY_I2C_ALLOW_INTERNAL_PULL_UP = 1` in `mpconfigboard.mk`. This may or may not help,

SeanTheITGuy, 2024-03-15:
It did not help. I've ordered a second board for testing to ensure it isn't a fluke occurrence. One thing I note is that, if one solders pin headers to this and puts it in a bread board, one loses access to the reset/boot buttons as they are on the bottom. I wonder if it does affect all of these units, but given it's relative lack of use and difficulty of actually using the reset buttons, it has gone unnoticed, or at least unreported.

SeanTheITGuy, 2024-03-17:
Update: Under further testing, putting 10K pull up resistors on I2C pins 5 and 6 resolves the issue.  I would consider this a closed matter filed under poor hardware design, however, as stated before, sketches compiled under the Arduino framework do work.  Could there be a difference in the way the CircuitPython vs Arduino bootup sequences do things that could explain it? I've tried enabling the internal pullups in board_init before initalizing the lcd, as well as in espressif_board_reset_pin_number. Additionally I added `#define CIRCUITPY_I2C_ALLOW_INTERNAL_PULL_UP (1)` to mpconfigboard.h.

jepler, 2024-03-17:
[image]
aren't these pull up resistors on the relevant lines? (it is weird that there are *3* R4s on the schematic though, there's another on on IO8)

dhalbert, 2024-03-17:
I missed those resistors the first time I looked, sorry. However, this board is a clone, not the original board, and maybe they omitted the pullup resistors.

Another possibility is that the resistors are there, but you made the pullups stronger, which may help.

Since `CIRCUITPY_I2C_ALLOW_INTERNAL_PULL_UP = 1` did not help originally, it might be because we are checking for pullups when creating the I2C object. If this is turned off, via `CIRCUITPY_REQUIRE_I2C_PULLUPS = 0` in `mpconfigboard.mk`, that might help as well.

You could measure the resistance from 3.3V to the SCL and SDA pins to see what is going on and whether the pullups are present. You will also be measuring some internal resistances on the chip, but you may get a general idea.

SeanTheITGuy, 2024-03-17:
The mystery's continue. Without my extra pull ups attached into the I2C plug on the board, there IS a 10K path from 3v to SDA/SCL. I did read somewhere about higher I2C frequencies requiring stronger pull ups. I don't know enough about the inner workings of CircuitPython vs a compiled Arduino sketch, as to whether it varies I2C clock, or has it fixed, but is that a possible reason why a 5K pull up (the built in pull ups plus my 10K additions) makes it function while a 10K does not?

dhalbert, 2024-03-17:
It's a "stronger" pullup, so the settling time will be shorter. Did you try `CIRCUITPY_REQUIRE_I2C_PULLUPS = 0` ?

SeanTheITGuy, 2024-03-17:
Confirmed. `CIRCUITPY_REQUIRE_I2C_PULLUPS = 0` and `CIRCUITPY_REQUIRE_I2C_PULLUPS = 1` both make no difference.

tannewt, 2024-03-18:
Build with `DEBUG=1` on the command line and you should get IDF logging out. It can give you more detail.

SeanTheITGuy, 2024-03-25:
Confirmed that this is a hardware error. Flashed a second identical unit in the default manner and it does not have this issue

closed 2024-03-25 by SeanTheITGuy