adafruit/circuitpython #9052: ulab.numpy.carray not available

h-milz, 2024-03-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-rc.1 on 2024-03-14; Adafruit Feather M4 Express with samd51j19

ulab.numpy does not seem to contain carray respectively it does not support complex arrays as documented. This is in the 9.0.0-rc.0 standard build as well as my custom build mentioned above with ULAB_SUPPORTS_COMPLEX explicitly set. Other functions are there, like conjugate: 

>>> import ulab.numpy as np
>>> dir (np)
['__class__', '__name__', 'all', 'any', 'bool', 'sort', 'sum', '__dict__', 'acos', 'acosh', 'arange', 'arctan2', 'argmax', 'argmin', 'argsort', 'around', 'array', 'asarray', 'asin', 'asinh', 'atan', 'atanh', 'bitwise_and', 'bitwise_or', 'bitwise_xor', 'ceil', 'clip', 'compress', 'concatenate', 'convolve', 'cos', 'cosh', 'cross', 'degrees', 'delete', 'diag', 'diff', 'dot', 'e', 'empty', 'equal', 'exp', 'expm1', 'eye', 'fft', 'flip', 'float', 'floor', 'frombuffer', 'full', 'get_printoptions', 'inf', 'int16', 'int8', 'interp', 'isfinite', 'isinf', 'left_shift', 'linalg', 'linspace', 'load', 'loadtxt', 'log', 'log10', 'log2', 'logspace', 'max', 'maximum', 'mean', 'median', 'min', 'minimum', 'nan', 'ndarray', 'ndinfo', 'nonzero', 'not_equal', 'ones', 'pi', 'polyfit', 'polyval', 'radians', 'right_shift', 'roll', 'save', 'savetxt', 'set_printoptions', 'sin', 'sinc', 'sinh', 'size', 'sqrt', 'std', 'tan', 'tanh', 'trace', 'trapz', 'uint16', 'uint8', 'vectorize', 'where', 'zeros']

But complex variable handling array is missing (1). My custom 9.0.0.rc0 build with ULAB_SUPPORTS_COMPLEX explicitly set does support e.g. np.conjugate as show below but complex arrays don't work. 

(1) I know that ULAB_SUPPORTS_COMPLEX is set to 0 in py/circuitpy_mpconfig.h but this setting seems to be checked nowhere in the entire source tree; at least git grep finds nothing. OTOH, ulab.h sets it to 1 by itself: 

#ifndef ULAB_SUPPORTS_COMPLEX
#define ULAB_SUPPORTS_COMPLEX               (1)
```

### Code/REPL

```python
>>> import ulab.numpy as np
>>> a=(1+4j)
>>> np.conjugate(a)
(1-4j)
>>> complex_array = np.array([1+2j, 3+4j, 5+6j])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert complex to float
```

### Behavior

see above

### Description

see above

dhalbert, 2024-03-15:
> I know that `ULAB_SUPPORTS_COMPLEX` is set to `0` in `py/circuitpy_mpconfig.h` but this setting
> seems to be checked nowhere in the entire source tree except in `ulab.h`:

I see ULAB_SUPPORTS_COMPLEX checked in many places:
```sh
~/repos/circuitpython$ ag '#if ULAB_SUPPORTS_COMPLEX' -l
extmod/ulab/code/ulab_tools.c
extmod/ulab/code/ndarray.c
extmod/ulab/code/ndarray_operators.c
extmod/ulab/code/numpy/numpy.c
extmod/ulab/code/numpy/carray/carray_tools.h
extmod/ulab/code/numpy/carray/carray_tools.c
extmod/ulab/code/numpy/carray/carray.c
extmod/ulab/code/numpy/compare.c
extmod/ulab/code/numpy/filter.c
extmod/ulab/code/numpy/create.c
extmod/ulab/code/numpy/poly.c
extmod/ulab/code/numpy/numerical.c
extmod/ulab/code/numpy/fft/fft_tools.c
extmod/ulab/code/numpy/fft/fft.h
extmod/ulab/code/numpy/fft/fft_tools.h
extmod/ulab/code/numpy/fft/fft.c
extmod/ulab/code/numpy/bitwise.c
extmod/ulab/code/numpy/vector.c
extmod/ulab/code/numpy/io/io.c
extmod/ulab/code/numpy/vector.h
extmod/ulab/code/utils/utils.c
extmod/ulab/code/scipy/signal/signal.c
extmod/ulab/code/ulab.c
extmod/ulab/code/ndarray_properties.c
extmod/ulab/code/ndarray.h
extmod/ulab/code/ulab_tools.h
```
I'm not clear if you did set it to `(1)` in `circuitpy_mpconfig.h` or not.

h-milz, 2024-03-15:
@dhalbert I did ... 

$ grep SUPPORTS_COM ../../py/circuitpy_mpconfig.h
#ifndef ULAB_SUPPORTS_COMPLEX
#define ULAB_SUPPORTS_COMPLEX (1)

dhalbert, 2024-03-15:
Do you see `carray` available in a MicroPython build with `ulab` on (I don't know which MicroPython builds it's turned on in).

Pinging @v923z here, as this may be a `ulab` issue.

h-milz, 2024-03-15:
No idea, I'm not into MP atm.

v923z, 2024-03-15:
On `micropython`, I can conjugate numbers. I settings are in https://github.com/v923z/micropython-ulab/blob/master/code/ulab.h, so I must assume that `ULAB_SUPPORTS_COMPLEX` is defined  somewhere by `circuitpython`.

dhalbert, 2024-03-23:
@h-milz This works for me, when `#define ULAB_SUPPORTS_COMPLEX (1)`. You have to give the `dtype` explicitly.

Adafruit CircuitPython 9.0.0-1-gdc69c29dac-dirty on 2024-03-22; Adafruit Feather RP2040 with rp2040
>>> import ulab.numpy as np
>>> np.array([1,2,3])
array([1.0, 2.0, 3.0], dtype=float32)
>>> np.array([1,2,3+0j])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
>>> np.array([1,2,3+0j], dtype=np.complex)
array([1.0+0.0j, 2.0+0.0j, 3.0+0.0j], dtype=complex)
>>>

dhalbert, 2024-03-23:
As a point of interest, turning on complex support made the RP2040 CircuitPython firmware about 12kB larger.

v923z, 2024-03-23:
> @h-milz This works for me, when `#define ULAB_SUPPORTS_COMPLEX (1)`. You have to give the `dtype` explicitly.

This is actually a good point. I should, perhaps, inspect the arguments of `array`, and depending on the results, decide on the `dtype`. But that raises a question about the design concept: `array` defaults to float, because float was the most-inclusive `dtype` at the beginning. We have added complex since, so, if we don't want to default to complex, then we can't really save the inspection.
 
> There is no `carray` in ulab specifically (I searched for that name).

No, it's the name of a file (that also appears in your listing https://github.com/adafruit/circuitpython/issues/9052#issuecomment-1999663626).

dhalbert, 2024-03-23:
> I should, perhaps, inspect the arguments of `array`, and depending on the results, decide on the `dtype`

@v923z That would make it work more like numpy, I suppose. You would know better if there are any traps introduced by such a change. It would require scanning all of the input values, since the user might have passed in, say, a mixed int/float list. I'm not sure if that counts as an expensive operation or not.

h-milz, 2024-03-23:
As a user, I'd say the current way it works is probably okay (float as a default), as long as the documentation is in sync (i.e., https://docs.circuitpython.org/en/latest/shared-bindings/ulab/numpy/index.html).

v923z, 2024-03-23:
> @v923z That would make it work more like numpy, I suppose. You would know better if there are any traps introduced by such a change. It would require scanning all of the input values, since the user [...]

It's not cheap. The float-int distinction wouldn't even cut it, because the magnitude of integers should also be inspected. 
I do see the point of `numpy`-compatibility, but I have the feeling that this would introduce a change that benefits few, but impacts everyone.

dhalbert, 2024-03-24:
@v923z I'm not pushing for it, just wondering. @h-milz I will close this now because I think it's all explained. If not, please reopen.

closed 2024-03-24 by dhalbert