adafruit/circuitpython #9053: Web Workflow uploads file to wrong folder if default folder has been changed using os.chdir()

RetiredWizard, 2024-03-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-rc.0-4-g04843e8597 on 2024-03-11; M5 Stack Cardputer with ESP32S3
Adafruit CircuitPython 9.0.0-rc.1 on 2024-03-14; Adafruit HUZZAH32 Breakout with ESP32
```

### Code/REPL

```python
import os
os.chdir('/')
if not 'test' in os.listdir():
    os.mkdir('test')

os.chdir('/test')
print('Directory set to test folder:',os.getcwd())
```

### Behavior

Using Web Workflow to upload a file to the root folder when the indicated code.py has been executed causes the file to uploaded to the /test folder rather than the root folder.

RetiredWizard, 2024-03-15:
I have been having intermittent issues moving the web workflow to subdirectories but hadn't been able to reliably reproduce the issue. I'm now thinking it's related to this issue and the web workflow file listing is also using the current folder of the device to look for selected target folders when attempting to change to a subfolder. Essentially the Web Worfklow file listing and the devices current folder are at different locations.

dhalbert, 2024-06-05:
#9224 (merged after this issue was reported) forces a `chdir` to top level when the VM is restarted. So that fixes it being "sticky", if that's the problem. But if you are using the web workflow while the program is running and has done an `os.chdir()`, I guess we need to fix that

PR #9311 (2024-06-06): Web workflow: use absolute paths to avoid paths being adjusted by current directory
- Fixes #9053 (opened by @RetiredWizard).

The web workflow was using relative paths internally when adjusting paths at the root `/`. This made them subject to whatever `os.chdir()` had been set to. Instead, use absolute paths to avoid any influence by the current directory.

I tested by doing `os.chdir("/lib")` in the REPL while using the web workflow. Previously this messed up being able to descend into directories or fetch files. Now it seems to work fine.
closed 2024-06-06 by tannewt