adafruit/circuitpython #9058: Web workflow problem with Lolin C3 Mini

ldrolez, 2024-03-17:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.10 on 2024-02-14; Wemos Lolin C3 Mini with ESP32-C3FH4
```

### Code/REPL

```python
none
```

### Behavior

The board does not connect to the AP, and web workflow is not started

### Description

- settings.toml is OK
- but it does not connect to the AP
- I have read about the tx power problem so I added 'wifi.radio.tx_power=8.5' to code.py => no improvement
- If I run wifi.radio.connect(...) it connects, but is there a way to restart the web workflow from here,  in command line?

dhalbert, 2024-03-17:
Please try 9.0.0-rc.1, and please show us your `settings.toml`, with the SSID and password redacted.

ldrolez, 2024-03-19:
```
CIRCUITPY_WIFI_SSID = "Myssid-ABCDEF"
CIRCUITPY_WIFI_PASSWORD = "pass with spaces"
CIRCUITPY_WEB_API_PASSWORD = "qwerqwer"
CIRCUITPY_WEB_API_PORT = 80
```
Ok I'll try the 9.0

ldrolez, 2024-03-20:
Adafruit CircuitPython 9.0.0 on 2024-03-19; Wemos Lolin C3 Mini with ESP32-C3FH4

Same problem, manual wifi.radio.connect() works

ldrolez, 2024-05-09:
Things seem to be worse with 9.0.4: if I have a `settings.toml`, Thonny cannot connect to the ESP.

ldrolez, 2024-05-10:
Ok with the proper boot.py, the web workflow starts:
```
import wifi
wifi.radio.tx_power=8.5
```