adafruit/circuitpython #9059: Using the keyboard blocks execution if there is no connection to USB

berkut0, 2024-03-19:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-rc.1 on 2024-03-14; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
import board 
import time
import neopixel
import usb_hid
import supervisor

from digitalio import DigitalInOut, Direction, Pull
from adafruit_debouncer import Debouncer

from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
 
keyboard = Keyboard(usb_hid.devices)

pixels = neopixel.NeoPixel(board.GP16, 1, brightness=1)

def make_pin_reader(pin):
    io = DigitalInOut(pin)
    io.direction = Direction.INPUT
    io.pull = Pull.UP
    return io

keymap = ((board.GP13, Keycode.ONE),
          (board.GP12, Keycode.TWO),
          (board.GP10, Keycode.THREE),
          (board.GP9, Keycode.FOUR))

keymap = [{Debouncer(make_pin_reader(pin)), keycode} for pin, keycode in keymap]

while True:
    for btn, key in keymap:
        btn.update()
        if btn.fell:
#             keyboard.press(key)
            pixels[0] = (255,0,255)
        if btn.rose:
#             keyboard.release(key)
            pixels[0] = (0,0,0)
```

### Behavior

I'm not sure if it's a bug or a behaviour. But it took me quite a while to find the problem. The problem goes something like this: I'm using keyboard emulation, and I've noticed that without a computer connection, the code just stops somewhere in the bowels of the keyboard constructor - when running with powerbank, the code runs before the keyboard is created, but not afterwards. Why is this important? I want to preserve the functionality of the device when it is powered by a power bank. But now it turns out that the device stops working when it is not connected to a computer.

### Description

- Code blocks while Keyboard not connected to USB

dhalbert, 2024-03-19:
```py
keyboard = Keyboard(usb_hid.devices)
```
will raise an exception if you are not connected to USB. So catch that. Maybe wrap your entire program  in a try-catch, which is perhaps most easily done by turning the program into a `def main()` and then wrapping a call to that.

berkut0, 2024-03-19:
Yes, I tried, but I don't understand what to wrap. I wrapped the constructor (of the Keyboard class) and the import. I also wrapped everything after the keyboard was created. No response at all. Also, I have the LED flashing red when an exception occurs, but it is flashing yellow as if it is just loading. Maybe I'm not using Try correctly?

berkut0, 2024-03-19:
something like this doesn't work:
```Python
try:
    from adafruit_hid.keyboard import Keyboard
    from adafruit_hid.keycode import Keycode
    keyboard = Keyboard(usb_hid.devices)
except:
    pixels[0] = (255,255,0)
```

I didn't just mention that the code stops running. Before the keyboard is created, the code is executed, after it is not. And the exception is not caught.

dhalbert, 2024-03-19:
I'm sorry, I led you astray.

It's possible to create `Keyboard()` before USB is connected. The `adafruit_hid` library will then wait for USB to connected before proceeding. By default it waits forever. I forgot the timeout was indefinite. See https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/01610da60bcf4ac23304b6ee410b46b8bbb35cd7/adafruit_hid/keyboard.py#L41 for the documentation. (There is some stuff missing in the online ReadTheDocs documentation: I will fix that.)

You can add a timeout by doing something like:

```py
    keyboard = Keyboard(usb_hid.devices, timeout=2)
```
This will `raise OSError` after two seconds if it cannot connect to USB. You can catch that with a `try`-`catch`.

An alternative, which is basically the same thing, but doesn't raise an error is:
```py
import supervisor
import time
# ...

# Wait a bit for USB to finish connecting if it is slow.
time.sleep(2)

if supervisor.runtime.usb_available():
    # initialize HID
```

berkut0, 2024-03-19:
@dhalbert thank you so much for explaining!
It explains the whole situation perfectly!

closed 2024-03-19 by dhalbert

berkut0, 2024-03-19:
But I should mention that:
```
AttributeError: 'Runtime' object has no attribute 'usb_available'
```
Probably you meant:
```Python
if supervisor.runtime.usb_connected:
    pass
```
I was very close to this option, using it as an indication, but your explanation cleared things up. But it seemed to me that this option was not very stable and that you had to adjust the delay time. Perhaps the problem is with my cable.

dhalbert, 2024-03-19:
Sorry for the typo. `.usb_connected` is what is used internally when you specify a timeout: https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/01610da60bcf4ac23304b6ee410b46b8bbb35cd7/adafruit_hid/__init__.py.

The delay time depends on how fast USB enumeration happens with the host. So you need to allow the maximum possible time. We can't determine whether the host is slow or not there (well, maybe we could test whether enumeration is in progress, but that would be more complicated.) A few seconds should be fine.