adafruit/circuitpython #9063: synthio LFO retrigger does not immediately affect value

kevinjwalters, 2024-03-19:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import time
import board
import audiomixer
import synthio
import audiopwmio
import ulab.numpy as np

SAMPLE_RATE = 32_000
MIXER_BUFFER_SIZE = 2048

audio = audiopwmio.PWMAudioOut(board.GP18, right_channel=board.GP19)
mixer = audiomixer.Mixer(channel_count=1,
                         sample_rate=SAMPLE_RATE,
                         buffer_size=MIXER_BUFFER_SIZE)
synth = synthio.Synthesizer(channel_count=1,
                            sample_rate=SAMPLE_RATE)

audio.play(mixer)
mixer.voice[0].play(synth)
mixer.voice[0].level = 0.75

lfo_waveform = np.array([0, 32767], dtype=np.int16)

#### 1 second count down
lfo_test1 = synthio.LFO(rate=1,
                        waveform=lfo_waveform,
                        scale=1000.0,
                        interpolate=True,
                        once=True)

synth.blocks.append(lfo_test1)
audio.play(mixer)
mixer.voice[0].level = 0.75
mixer.voice[0].play(synth)
start_ns = time.monotonic_ns()

for _ in range(5):
    print((time.monotonic_ns() - start_ns) / 1e9, lfo_test1.value)
    time.sleep(0.099)

print((time.monotonic_ns() - start_ns) / 1e9, "retrigger")
lfo_test1.retrigger()

for _ in range(20):
    print((time.monotonic_ns() - start_ns) / 1e9, lfo_test1.value)
    time.sleep(0.001)

for _ in range(15):
    print((time.monotonic_ns() - start_ns) / 1e9, lfo_test1.value)
    time.sleep(0.099)

while True:
    pass
```

### Behavior

The retrigger does not have an immediate effect on the LFO `value`.  Here you can see the LFO being retriggered at half a second when it's approximately 500 in value and that value continues for a few ms (~15ms for this example/run).

```
0.000976562 7.99976
0.100586 95.9971
0.200195 191.994
0.299805 287.991
0.400391 399.988
0.5 retrigger
0.5 495.984
0.501953 495.984
0.50293 495.984
0.503906 495.984
0.504883 495.984
0.505859 495.984
0.507813 495.984
0.508789 495.984
0.509766 495.984
0.510742 495.984
0.511719 495.984
0.514648 15.9995
0.516602 15.9995
```

### Description

I suspect this is because the LFO values are only calculated periodically rather than every sample and the previous value is cached and that cache is _not_ invalidated by the retrigger.

https://docs.circuitpython.org/en/latest/shared-bindings/synthio/index.html says

> In the current implementation, LFOs are updated every 256 samples.

kevinjwalters, 2024-03-19:
BTW, has there been any discussion about making the "update ever N" samples value configurable per LFO?

kevinjwalters, 2024-03-20:
It looks like there's a similar issue for newly created LFOs.

```
WAVEFORM_HALFLEN = 2048
lfo_test2 = synthio.LFO(rate=1, scale=WAVEFORM_HALFLEN * 15//16 // 2, offset=WAVEFORM_HALFLEN - WAVEFORM_HALFLEN * 15//16 // 2)
```

I'd expect that to start around 1024 but there are a load of (out of range) 0 values first before it gets going, see output below for printing the value in a loop doing some other stuff. 

```
0
0
0
0
0
1114
1114
1114
1139
1139
1139
1139
1139
1139
1139
1139
1139
1139
1165
1165
1165
1165
1165
1165
1165
1165
1165
1165
1190
```

jepler, 2024-03-20:
I took a quick look at this and I don't think there's an easy fix. This also applies to arithmetic blocks, and changing other inputs such as phase or offset.

Here's why: The values are lazily calculated during the generation of audio in the background. There's not even code to enable it to happen at other times (it's tied up with the idea that a LFO gets updated with a "tick").

In the short term it would be good to mention this in the documentation because I'm not surprised that *you* were surprised.

PR #9197 (2024-04-29): LFO: document some things that trip up users
Closes: #9073

Closes: #9063

ping @kevinjwalters
commit a4e60ec6 (2024-04-29): LFO: document some things that trip up users

Closes: #9073

Closes: #9063
closed 2024-04-29 by dhalbert