adafruit/circuitpython #9067: Immediate wake from deepsleep when upgrading to 9.0.0

dtorres-sf, 2024-03-21:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Board ID:adafruit_feather_esp32s3_4mbflash_2mbpsram
UID:CD4557DDA3C4
```

### Code/REPL

```python
import alarm

# Full program here, everything else is working fine except for sleep below...

alarmA = alarm.pin.PinAlarm(pin=board.D12, value=True, pull=True)
alarm.exit_and_deep_sleep_until_alarms(alarmA)
```

### Behavior

I have not included all the code, but abbreviated to just the part that I think has an issue. On CP 8 the above alarm worked as expected. I have a button hooked up to pin D12, that is connected 3V. When the button is pushed 3V is applied to D12 which should wake from sleep.

Instead, the board immediately wakes, without ever sleeping (or if it does sleep it is for a very short interval, less than 1 second). 

I have tested on two boards and same behavior on both. I have doubled checked wiring as well. Downgrading back to CP 8, without changing my code.py (only changing CP and libs) fixes the issue.

dhalbert, 2024-03-21:
Two questions:

1. Is this while you are battery powered and not connected to USB, or connected to USB? In the latter case it will be a "fake" deep sleep.
2. The example above does not include `import board`, so it will fail immediately (and not because of sleeping).

dhalbert, 2024-03-22:
@dtorres-sf I reproduced this problem, and started to track it down, as noted above. I think it may be a bug in ESP-IDF, not in our code. However, we are going to upgrade to ESP-IDF v5.2.1 in CircuitPython 9.1.0. When I test with ESP-IDF v5.2.1, the problem goes away.

So at this point, I don't think it is worth a temporary fix for this.

We should have a beta 9.1.0 fairly soon. In the meantime, could you stick with 8.x.x? Another alternative, if easy for you, is to make your pushbutton pull the pin low, instead of high. That works fine in 9.0.x.

dtorres-sf, 2024-03-22:
@dhalbert thank you for the quick investigation, root cause, and suggested workarounds. We can definitely stay on 8.x.x and then retest on 9.1.0 when that is ready and upgrade our boards. Thanks again!

dhalbert, 2024-03-27:
Closing because this will be fixed in 9.1.0-beta.0 due to the ESP-IDF update.

closed 2024-03-27 by dhalbert