adafruit/circuitpython #9073: synthio LFO once changes the waveform

kevinjwalters, 2024-03-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import math
import time
import gc

import board
import audiomixer
import synthio
import audiopwmio
import ulab.numpy as np

SAMPLE_RATE = 32_000
MIXER_BUFFER_SIZE = 2048

audio = audiopwmio.PWMAudioOut(board.GP18, right_channel=board.GP19)
mixer = audiomixer.Mixer(channel_count=1,
                         sample_rate=SAMPLE_RATE,
                         buffer_size=MIXER_BUFFER_SIZE)
synth = synthio.Synthesizer(channel_count=1,
                            sample_rate=SAMPLE_RATE)

audio.play(mixer)
mixer.voice[0].play(synth)
mixer.voice[0].level = 0.75

lfo_waveform = np.array([0, 32767, 0, -32767], dtype=np.int16)
lfo_waveform_extrazero = np.concatenate((lfo_waveform,
                                         np.array([0], dtype=np.int16)))

### 0.2Hz LFOs ...
lfo_inf = synthio.LFO(rate=1/5,
                      waveform=lfo_waveform,
                      scale=500.0,
                      offset=1500.0,
                      interpolate=True,
                      once=False)
lfo_oneshot1 = synthio.LFO(rate=1/5,
                           scale=500.0,
                           offset=500.0,
                           interpolate=True,
                           once=True)
lfo_oneshot2 = synthio.LFO(rate=1/5,
                           waveform=lfo_waveform,
                           scale=500.0,
                           offset=-500.0,
                           interpolate=True,
                           once=True)
lfo_oneshot3 = synthio.LFO(rate=1/5,
                           waveform=lfo_waveform_extrazero,
                           scale=500.0,
                           offset=-1500.0,
                           interpolate=True,
                           once=True)

print_every_ns = 100_000_000  ### 0.1s
rows = 75  ### 7.5 seconds worth, 1.5 cycles ideally
                      
synth.blocks.append(lfo_inf)
synth.blocks.append(lfo_oneshot1)
synth.blocks.append(lfo_oneshot2)
synth.blocks.append(lfo_oneshot3)
audio.play(mixer)
mixer.voice[0].level = 0.75
gc.collect()
mixer.voice[0].play(synth)
start_ns = time.monotonic_ns()
next_ns = start_ns + print_every_ns

for idx in range(rows):
    print((lfo_inf.value, lfo_oneshot1.value, lfo_oneshot2.value, lfo_oneshot3.value))
    while time.monotonic_ns() < next_ns:
        pass
    next_ns += print_every_ns
```

### Behavior

LFOs which use `once=True` and use the same waveform as a repeating LFO do not have the same effective frequency as they do not return to the initial value to complete a full cycle of the waveform.

### Description

This is going to be unintuitive for anyone who doesn't ponder (and look at) what the code does and it's unlike a real synthesizer/module with a oneshot mode.

If the current behaviour is viewed as correct then documentation with an example would be helpful.

### Additional information

Mu plotting the output

[image]

From top to bottom, all at 0.2 Hz.

1. cyan - normal LFO using [0, 32767, 0, -32767] waveform
2. green - `once=True` with default waveform
3. orange - `once=True` with [0, 32767, 0, -32767]
4. blue - `once=True` with [0, 32767, 0, -32767, 0]

jepler, 2024-03-22:
A "once" waveform cannot in general interpolate between the last and first entry. This would, for instance, make a sweep with waveform [0, 37678] be wrong: it would become a triangle waveform. So, something "had to give".

My choice was to change how a `once_lfo=LFO(once=True)` interpolates across its cycle, compared to a `normal_lfo=LFO(once=False)`: Consider the case when phase is exactly .999: `normal_lfo.value` should be almost exactly `waveform[0]` (plus a tiny bit of `waveform[3]` in this case). While the `once_lfo` is almost at the end of its single cycle, so its value has to be almost exactly `waveform[3]` with a tiny hint of `waveform[2]` thrown in.

That's why using the waveform with last element repeated changes the behavior.

A side effect of this is that merely toggling "once" on/off dynamically causes a step change in the output, because the way that the `phase` is translated to a sample index changes. :-/

Is there something better to do here? Having an `lfo(once=True)` arrive at its final value when phase reaches `1-1/waveform_len` doesn't seem like a great idea, but that's the only alternative I thought of.

Once again, without an easy solution, a documentation improvement is my suggested fix. I'd love to hear I'm overlooking something, though.

kevinjwalters, 2024-04-02:
This is more idea than a reference implementation but maybe if waveforms had some indication of whether they are normally used as repeating waveforms could provide a hint on how to turn them into one shot waveforms?

https://gist.github.com/kevinjwalters/ff004b1b1acab9f70b9d237765acb3b3

The output from this set is shown below

https://gist.github.com/kevinjwalters/f0f7028c4b88dab56cf218ccbe0382b6

[image]

PR #9197 (2024-04-29): LFO: document some things that trip up users
Closes: #9073

Closes: #9063

ping @kevinjwalters
commit a4e60ec6 (2024-04-29): LFO: document some things that trip up users

Closes: #9073

Closes: #9063
closed 2024-04-29 by dhalbert