adafruit/circuitpython #9078: Strange issue when doing storage.mount with no terminal connection.

kwpodolsky, 2024-03-23:
I posted this issue up in the forum (https://forums.adafruit.com/viewtopic.php?t=209383), but have gotten no responses, so I figured I would post the issue here.

Currently running CircuitPythone 8.2.6 on Raspberry Pico W.
My code looks like the following snippet:

===================

import adafruit_sdcard
import storage
....

sdcard = adafruit_sdcard.SDCard(spi1, digitalio.DigitalInOut(sd_cs))

vfs = storage.VfsFat(sdcard)

//# The following line FAILS if there is NO Terminal connection to the board
//# I know because I put a try/except around it in my real code.
storage.mount(vfs, "/sd")

stats = os.statvfs('/sd')

=================

If I have a terminal app connected to the usb on the board and run the code the "storage.mount" code works fine.
IF, I try to run without a terminal connection, the "storage.mount" line fails.

Anyone have ANY idea as to WHY a terminal connection is required to allow this code to work/fail?

See screen shots of my OLED display output on working and fail.

[image]
[image]

dhalbert, 2024-03-23:
Could you try 9.0.0?

RetiredWizard, 2024-03-23:
I don't think this is a generic CP issue, I no longer have a keyboard feather wing but I'm pretty sure I used to mount SD cards on the keyboard feather wing with CP 7.x and 8.x using a Raspberry Pi Pico without a terminal connected. I do the same thing regularly using Espressif chips/boards using CP 9.x. 

I have seen intermittent mount failures with some SD cards after soft board resets and have taken to ejecting and reinserting the card and/or power cycling the board if I encounter mount problems using one of those cards. If you have a second card, and can format it with a FAT file system I'd try making sure the issue wasn't somehow associated with the particular card you're using. 

If that doesn't help identify the issue, I'm wondering if connecting the terminal is feeding power and regulating the voltage so that the SD card behaves better. How is the board powered when not connected to a terminal?

Is there a schematic for the board you're using available?

dhalbert, 2024-03-23:
Which SD breakout or accessory are you using? Could you show the wiring in a clear picture? (The current pictures are unfortunately pretty blurry.)

Also try `sdcardio` instead of `adafruit_sdcard`, though that may not be the issue. `sdcardio` is native and will be faster.

dhalbert, 2024-05-24:
I tested this on a Pico W with CircuitPython 9.1.0-beta.3, with this SD card breakout: https://www.adafruit.com/product/4682, a 4GB card, and the test program below. I plugged it in to a host computer or to a USB 5V adapter. A fast blink means there's trouble. A slow blink is fine.

I could not reproduce the problem. If this is still an issue for you, could you try with 9.1.0-beta.3, and using `sdcardio` instead of `adafruit_sdcard`?

Interestingly, I had a bad jumper wire, which caused problems powering the SD card breakout. So I wonder if you are seeing some kind of power problem.

I'm going to close this for now, but we can reopen it if this is still an issue for you.

```py
import board
import busio
import digitalio
import sdcardio
import storage
import time

def blink(led, delay):
    while True:
        led.value = not led.value
        time.sleep(delay)

led = digitalio.DigitalInOut(board.LED)
led.switch_to_output()

spi1 = busio.SPI(clock=board.GP10, MOSI=board.GP11, MISO=board.GP12)

sd = sdcardio.SDCard(spi1, cs=board.GP13)

vfs = storage.VfsFat(sd)

try:
    storage.mount(vfs, "/sd")
    blink(led, 1)
except Exception as e:
    print(type(e), e)
    blink(led, 0.1)
```

closed 2024-05-24 by dhalbert