adafruit/circuitpython #9080: Wifi Radio Takes Longer to Connect than in early 8.x builds

DJDevon3, 2024-03-24:
### CircuitPython version

```python
S3 Wifi vs Airlift WiFi initial negotiation timing.
I first noticed this issue around 8.2.3 to 8.2.5 where connecting to wifi (without web workflow env enabled) was taking 5-10 seconds vs almost instantly within milliseconds in prior versions.  ESP32SPI (airlift) in particular was much much faster to connect to wifi.  

Was told by Danh this was abnormal behavior and that both should be connecting at about the same instantaneous speed... at least to wifi.  I do not use web workflow, serial USB only.

I figured this issue would get cleaned up eventually for the S3 but it has persisted into 9.0 where there is a considerable latency between attempting to connect to wifi and confirming connecting.  Because I primarily only connect to wifi for requests API projects the code to test can be [any of the API examples](https://github.com/adafruit/Adafruit_CircuitPython_Requests/tree/main/examples/wifi/expanded).
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2024 DJDevon3
# SPDX-License-Identifier: MIT
# Coded for Circuit Python 8.2.x
"""Premiere League Total Players API Example"""
# pylint: disable=import-error

import os
import time

import adafruit_connection_manager
import adafruit_json_stream as json_stream
import wifi

import adafruit_requests

# Public API. No user or token required

# Get WiFi details, ensure these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

# API Polling Rate
# 900 = 15 mins, 1800 = 30 mins, 3600 = 1 hour
SLEEP_TIME = 900

# Initalize Wifi, Socket Pool, Request Session
pool = adafruit_connection_manager.get_radio_socketpool(wifi.radio)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(wifi.radio)
requests = adafruit_requests.Session(pool, ssl_context)

# Publicly available data no header required
PREMIERE_LEAGUE_SOURCE = "https://fantasy.premierleague.com/api/bootstrap-static/"

def time_calc(input_time):
    """Converts seconds to minutes/hours/days"""
    if input_time < 60:
        return f"{input_time:.0f} seconds"
    if input_time < 3600:
        return f"{input_time / 60:.0f} minutes"
    if input_time < 86400:
        return f"{input_time / 60 / 60:.0f} hours"
    return f"{input_time / 60 / 60 / 24:.1f} days"

while True:
    # Connect to Wi-Fi
    print("\nConnecting to WiFi...")
    while not wifi.radio.ipv4_address:
        try:
            wifi.radio.connect(ssid, password)
        except ConnectionError as e:
            print("❌ Connection Error:", e)
            print("Retrying in 10 seconds")
    print("✅ Wifi!")

    try:
        print(" | Attempting to GET Premiere League JSON!")

        # Set debug to True for full JSON response.
        # WARNING: may include visible credentials
        # MICROCONTROLLER WARNING: might crash by returning too much data
        DEBUG_RESPONSE = False

        try:
            PREMIERE_LEAGUE_RESPONSE = requests.get(url=PREMIERE_LEAGUE_SOURCE)
            pl_json = json_stream.load(PREMIERE_LEAGUE_RESPONSE.iter_content(32))
        except ConnectionError as e:
            print(f"Connection Error: {e}")
            print("Retrying in 10 seconds")
        print(" | ✅ Premiere League JSON!")

        print(f" | Total Premiere League Players: {pl_json['total_players']}")
        PREMIERE_LEAGUE_RESPONSE.close()
        print("✂️ Disconnected from Premiere League")

        print("\nFinished!")
        print(f"Board Uptime: {time.monotonic()}")
        print(f"Next Update: {time_calc(SLEEP_TIME)}")
        print("===============================")

    except (ValueError, RuntimeError) as e:
        print(f"Failed to get data, retrying\n {e}")
        time.sleep(60)
        break
    time.sleep(SLEEP_TIME)
```

### Behavior

Connecting to wifi is delayed compared to prior versions of 8.x  This issue pre-dates Connection Manager possibly even as early as some 8.x betas.  I never reported it because I thought it was a temporary issue but here we are in 9.0 and it hasn't been a temporary issue.

### Description

The time between `print("\nConnecting to WiFi...")` and `print("✅ Wifi!")` should be instantaneous but it takes a couple seconds.

It could very well be only my Wifi AP responsible for the slow connection but when I used an ESP32SPI (airlift) it was instantaneous.  There's a difference in connection time between the Airlift and S3 modules.

RetiredWizard, 2024-03-24:
Using CP 9.x I tried the wifi.radio.connect on an ESP32 Pico, ESP32-S2 and ESP32-S3 all of which executed instantly. I also tried connecting to WiFi using a Nano Connect RP2040 w/ESP32SPI and it did take a second or two to connect using a while loop.
```py
while not esp.connect_AP(os.getenv('CIRCUITPY_WIFI_SSID'),os.getenv('CIRCUITPY_WIFI_PASSWORD')):
    pass
```
I also threw the code.py example above onto an ESP32-S2 board and the first three output lines all printed with no noticeable delay between them
```
Connecting to WiFi...
✅ Wifi!
 | Attempting to GET Premiere League JSON!
 | ✅ Premiere League JSON!
```
There was a second or two delay before the fourth line printed.

DJDevon3, 2024-03-25:
That was a quick example and probably a bad one as that particular API does take a while to pull the JSON that's normal.  

Would probably be better to use the standard [wifi connect simpletest](https://learn.adafruit.com/adafruit-esp32-s3-feather/circuitpython-internet-test) than an API.  I think Anecdata might have some kind of testing tool for wifi socket connection timing.

anecdata, 2024-03-25:
I don't have any ready-made performance tests, but a couple of notes...

Airlift isn't CircuitPython core, so any issues there should be filed in the ESP32SPI library or NINA repo. NINA uses a much older version of the ESP-IDF, so it wouldn't be surprising if the performance is different (also, there is a Python library and an SPI transaction before it gets to the actual wifi connect C code, but those shouldn't be noticeable).

I've always assumed that wifi connection can take up to a couple of seconds, and expect variance due to equipment, network config, current network conditions, etc.

Maybe something like this just to accumulate some wifi connection times:
```py
import os
import wifi
import traceback

time.sleep(3)  # wait for serial

# choose your own credentials method...
ssid = os.getenv('WIFI_SSID')
password = os.getenv('WIFI_PASSWORD')

while True:
    wifi.radio.enabled = True
    print('Connecting to wifi...', end=' ')
    # choose your preferred time units (ns, ms, s)...
    start_wifi_connect = time.monotonic_ns()
    while not wifi.radio.connected:
        try:
            wifi.radio.connect(ssid, password)
        except Exception as ex:
            traceback.print_exception(ex, ex, ex.__traceback__)
    print(f'{wifi.radio.ipv4_address} {time.monotonic_ns() - start_wifi_connect}ns')
    wifi.radio.enabled = False
    time.sleep(5)
```

DJDevon3, 2024-03-25:
Nice, thank you.  I'm averaging 3.4s.  Maybe it's just me but in the past it seemed like the connection time was more like .4 seconds and now it's taking 3 seconds longer.  

It could be explained as simply as having my connection print statements right next to each other and what I thought was a faster connection was just the arrangement of my prints?

anecdata, 2024-03-25:
Yeah, keep the timing block as simple as possible.

_If_ (web workflow) auto-connect is enabled ('CIRCUITPY_WIFI_SSID'), connection time in `code.py` should be virtually zero, since it's already connected. The core connection code checks for an existing connection and exits if it's already connected, so there is minimal code to run and the timing should be negligible.

For non-(web workflow)-autoconnect, a lot of things could be different over such a long span of time: RSSI, core code changes, esp-idf changes, changes in measurement code, network environment, etc. We'd have to go back and test older versions. My expectation has always been a couple / few seconds.

DJDevon3, 2024-03-25:
Ah ok. I had a fundamental misunderstanding of how web workflow launches then.  It has to be just the way I was setting up prints back then.  I've been using the same code snippets for wifi connection for a while.  Perhaps the changes in connection speed prior to then was simply due to the way I used to do wifi connection prints. 

We can go ahead and close this one unless anyone wants to look into it deeper?  I'm not as confident this is a real issue anymore.

dhalbert, 2024-03-25:
I'll close this for now. If someone wants to do a timing test matrix like, that would settle it one way or the other, I think.

|                 | 8.2.0 | 8.2.10 | 9.0.0 |
|-----------------|-------|--------|-------|
| web workflow    |       |        |       |
| no web workflow |       |        |       |

closed 2024-03-25 by dhalbert