adafruit/circuitpython #9081: httpserver simpletest fails (cp9 on Pico W)

studiostephe, 2024-03-25:
### CircuitPython version

```python
dafruit CircuitPython 9.0.0 on 2024-03-19; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import socketpool
import wifi

from adafruit_httpserver import Server, Request, Response

ssid = "Unintended Consequences"
password = "xxxx"

print("Connecting to", ssid)
wifi.radio.connect(ssid, password)
print("Connected to", ssid)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)

@server.route("/")
def base(request: Request):
    """
    Serve a default static plain text message.
    """
    return Response(request, "Hello from the CircuitPython HTTP Server!")

server.serve_forever(str(wifi.radio.ipv4_address))
```

### Behavior

code.py output:
Connecting to Unintended Consequences
Connected to Unintended Consequences
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "adafruit_httpserver/server.py", line 185, in serve_forever
  File "adafruit_httpserver/server.py", line 228, in start
  File "adafruit_httpserver/server.py", line 207, in _create_server_socket
OSError: [Errno 95] EOPNOTSUPP

### Description

Test code taken from first example here: https://docs.circuitpython.org/projects/httpserver/en/latest/examples.html

anecdata, 2024-03-25:
I think this is a known mismatch with the library version. Try this library branch:
https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/tree/reuseaddr

studiostephe, 2024-03-25:
Either way, I get the same error when I get to "sock.setsockopt":

Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "/lib/adafruit_httpserver/server.py", line 186, in serve_forever
  File "/lib/adafruit_httpserver/server.py", line 227, in start
  File "/lib/adafruit_httpserver/server.py", line 205, in _set_socket_level_to_reuse_address
OSError: [Errno 95] EOPNOTSUPP

DJDevon3, 2024-03-25:
Same, tried the library you posted and getting the same error.
```py
Adafruit CircuitPython 9.0.0 on 2024-03-19; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
```
```py
Connecting to WiFi...
✅ Wifi!
Traceback (most recent call last):
  File "code.py", line 28, in <module>
  File "/lib/adafruit_httpserver/server.py", line 186, in serve_forever
  File "/lib/adafruit_httpserver/server.py", line 227, in start
  File "/lib/adafruit_httpserver/server.py", line 205, in _set_socket_level_to_reuse_address
OSError: [Errno 95] EOPNOTSUPP
```

DJDevon3, 2024-03-25:
Tried using Connection Manager just to see if the socket management might change something
```py
import os

import adafruit_connection_manager
import wifi

from adafruit_httpserver import Server, Request, Response

# Get WiFi details, ensure these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

print("Connecting to WiFi...")
wifi.radio.connect(ssid, password)
print("✅ Wifi!")

# Initalize Wifi, Socket Pool, Request Session
pool = adafruit_connection_manager.get_radio_socketpool(wifi.radio)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(wifi.radio)
server = Server(pool, "/static", debug=True)

@server.route("/")
def base(request: Request):
    """Serve a default static plain text message"""
    return Response(request, "Hello from the CircuitPython HTTP Server!")

server.serve_forever(str(wifi.radio.ipv4_address))
``` Still same error.

studiostephe, 2024-03-25:
Interesting update: Since my use case is actually making an AP, I tried this code.py:
`import socketpool
import wifi
from adafruit_httpserver import Server, Request, Response, SSEResponse, GET

wifi.radio.start_ap('MyAP')

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool,debug=True)
print(wifi.radio.ap_active)

@server.route("/")
def base(request: Request):
    """
    Serve a default static plain text message.
    """
    return Response(request, "Hello from the CircuitPython HTTP Server!")

server.serve_forever(str(wifi.radio.ipv4_address_ap))`

and I modified server.py thusly (starting at line 198):
`    def _set_socket_level_to_reuse_address(self) -> None:
        """
        On systems that have SO_REUSEADDR, prevents "Address already in use"
        error when restarting the server.
        """
        try:
            self._sock.setsockopt(
                self._socket_source.SOL_SOCKET, self._socket_source.SO_REUSEADDR, 1
            )
        except Exception as e:
            print("oh nooooo:", e)`

and got a successful result despite still having the error: 
code.py output:
True
oh nooooo: [Errno 95] EOPNOTSUPP
Started development server on http://192.168.4.1:80

And I can even get a successful response in my browser: 
192.168.4.16 -- "GET /" 382 -- "200 OK" 125 -- 236ms

DJDevon3, 2024-03-25:
I'm getting the same result for that too.  Interesting error message lol.

studiostephe, 2024-03-25:
I think the way its written in the library, the `except` only runs if the type of error is AttributeError. Since the type of error actually being thrown is `OSError `, the the error is just raised instead of handled. 

As to knowing the actual cause of the error, I'm way out of my depth. I just wanted to display a lil' sensor reading, and I thought this would be faster than hooking up a display 💀

re: non-existent variable- Try power cycling the pico, the wifi hardware might still be connected to your network from a previous run

DJDevon3, 2024-03-25:
That makes sense, all of it.  Thanks.  Can always add AttibuteError as a separate exception. I'm not sure if that will have any negative effects but for now it is a workaround for this specific issue.

```py
def _set_socket_level_to_reuse_address(self) -> None:
        """
        On systems that have SO_REUSEADDR, prevents "Address already in use"
        error when restarting the server.
        """
        try:
            self._sock.setsockopt(
                self._socket_source.SOL_SOCKET, self._socket_source.SO_REUSEADDR, 1
            )
        except AttributeError:
            pass
        except Exception:
            pass
```  Just add both ;)

anecdata, 2024-03-25:
This should get sorted out in the library. If I understand it correctly, if the `.setsockopt` call can't be run (by removing it, or catching any exceptions), the server should still work, but there may be edge cases where it will run out of sockets.

studiostephe, 2024-03-25:
Out of curiosity when did you see the `[Errno 112] EADDRINUSE`? I can hit the server a bunch of times successfully:
```
10.0.1.33 -- "GET /" 380 -- "200 OK" 125 -- 63ms
10.0.1.33 -- "GET /" 380 -- "200 OK" 125 -- 190ms
10.0.1.33 -- "GET /" 380 -- "200 OK" 125 -- 95ms
10.0.1.33 -- "GET /" 380 -- "200 OK" 125 -- 44ms
10.0.1.33 -- "GET /" 380 -- "200 OK" 125 -- 144ms
and lots more...
```

DJDevon3, 2024-03-25:
When I hit file save in Mu.  Apparently once you get the server up and running you shouldn't attempt to save code.py again?  If I leave the server running yes I can refresh the browser as many times as I want without issue.

anecdata, 2024-03-25:
Looks like this is more than a library issue... I don't think `SO_REUSEADDR` is implemented in the `raspberrypi` port
https://github.com/adafruit/circuitpython/blob/ac66da21fe33567c50f5716bf040ce5c88e1e12b/ports/raspberrypi/common-hal/socketpool/Socket.c#L1186
Probably the testing so far has been with `espressif`

dhalbert, 2024-03-25:
#9084 is meant to fix this. Could you someone test the build artifact? Thanks.
https://github.com/adafruit/circuitpython/actions/runs/8421732172?pr=9084

DJDevon3, 2024-03-26:
Artifact works.  Can soft-reload without issue.  Simpletests work.  Good fix. 👍

anecdata, 2024-03-26:
Closing, fixed by https://github.com/adafruit/circuitpython/pull/9084

Dan [commented](https://github.com/adafruit/Adafruit_CircuitPython_ConnectionManager/issues/3#issuecomment-1999901399) in another thread that adafruit_httpsrver does not need to use ConnectionManager.

closed 2024-03-26 by anecdata

dhalbert, 2024-03-26:
> Dan [commented](https://github.com/adafruit/Adafruit_CircuitPython_ConnectionManager/issues/3#issuecomment-1999901399) in another thread that adafruit_httpsrver does not need to use ConnectionManager.

I meant that the actual `adafruit_httpserver` library does not need ConnectionManager internally. But it's fine and useful to use ConnectionManager in a test program. ConnectionManager makes it easy to get the socketpool and the ssl context.

If you want to open a new PR to use ConnectionManager in a simpletest or other example, I think that's a fine idea.

anecdata, 2024-03-26:
Thanks for clarifying, Dan

PR #9084 (2024-03-26): raspberrypi: implement setsockopt(SOL_SOCKET, SO_REUSEADDR)
untested

edit by @dhalbert:
Fixes #9081.