adafruit/circuitpython #9086: ItsyBitsy ESP32 can't use onboard NeoPixel in v9.0.0

tyeth, 2024-03-25:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit ItsyBitsy ESP32 with ESP32
```

### Code/REPL

```python
# import time
# import board
# import neopixel

# pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)

# while True:
#     pixels.fill(0xADAF00)
#     time.sleep(1)
#     pixels.fill(0)
#     time.sleep(1)

import board
import digitalio
import neopixel
import time

print("START")
# On MagTag, enable power to NeoPixels.
# Remove these two lines on boards without board.NEOPIXEL_POWER.
# np_power = digitalio.DigitalInOut(board.NEOPIXEL_POWER)
# np_power.switch_to_output(value=False)

np = neopixel.NeoPixel(board.NEOPIXEL, 1)

np[0] = (50, 50, 50)
time.sleep(1)
np[0] = (0, 0, 0)
```

### Behavior

```python
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
START
Traceback (most recent call last):
  File "code.py", line 24, in <module>
  File "/lib/neopixel.py", line 141, in __init__
ValueError: NEOPIXEL in use

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Following the code on the circuitpython-neopixels page from the itsybitsy esp32 learn guide, similar pin in use if trying to use neopixel_power. I was originally using external neopixels so didn't notice, but now that I've got a hot-glue light shaft I wanted to take over the status light / onboard neopixel.
Same code works fine on esp32c6

https://learn.adafruit.com/adafruit-itsybitsy-esp32/neopixel-example

RetiredWizard, 2024-03-25:
I don't have an ItsyBitsy to test with but it looks to me like this is a Pin/GPIO miss-assignment. Can you test the attached firmware and if it resolves the issue I'll post a PR?
[firmware.zip](https://github.com/adafruit/circuitpython/files/14749227/firmware.zip)

tannewt, 2024-03-26:
@tyeth Please test with 9.0.x

dhalbert, 2024-03-29:
Reproduced with 9.0.0. Tested with 9.0.2 and problem is now gone.

Fixed by #9087.

PR #9087 (2024-03-26): ItsyBitsy ESP32 Neopixel Pin/Power GPIO number fix
Possibly fixes #9086 

I don't have the hardware but according the the learning guide pinouts the neopixel/power are on GPIO0 & GPIO2.
closed 2024-03-29 by dhalbert