adafruit/circuitpython #9088: Memory Allocation Fail with Neo Trinkey

mrklingon, 2024-03-25:
### CircuitPython version

```python
Memory problem with CP 9.0.0 - code fails that works with CP 8.2.10

code.py output:
]0;🐍code.py | 9.0.0\Traceback (most recent call last):
File "code.py", line 1, in <module>
MemoryError: memory allocation failed, allocating 1016 bytes
]0;🐍1@code.py MemoryError | 9.0.0\
Code done running.

-------

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍code.py | 8.2.10\TREK
-  .-.  .  -.-  

Code:
9.0.0

code.py: https://github.com/mrklingon/neotrinkey/blob/main/TOS3.py

Uses these modules:
https://github.com/mrklingon/neotrinkey/blob/main/morse.py
https://github.com/mrklingon/neotrinkey/blob/main/prt.py
```

### Code/REPL

```python
code.py output:
]0;🐍code.py | 9.0.0\Traceback (most recent call last):
File "code.py", line 1, in <module>
MemoryError: memory allocation failed, allocating 1016 bytes
]0;🐍1@code.py MemoryError | 9.0.0\
Code done running.

-------

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍code.py | 8.2.10\TREK
-  .-.  .  -.-
```

### Behavior

MemoryError: memory allocation failed, allocating 1016 bytes
]0;🐍1@code.py MemoryError | 9.0.0\

### Description

Code:

code.py: https://github.com/mrklingon/neotrinkey/blob/main/TOS3.py

Uses these modules:
https://github.com/mrklingon/neotrinkey/blob/main/morse.py
https://github.com/mrklingon/neotrinkey/blob/main/prt.py

dhalbert, 2024-04-02:
I reproduced this with 9.0.2 and 9.1.0-beta.0.

I tried this recent change to MicroPython, which allows some compilation garbage to be collected more reliably: https://github.com/micropython/micropython/pull/14136.

If I comment out the last four or so routines in `TOS3.py` (`code.py`), then I don't get the storage problem. So we are just over the edge.

If I use `mpy-cross` on `TOS3.py`, `morse.py`, and `prt.py`, replace the .py files with those .mpy files, and replace code.py with:
```py
from TOS3 import *
```
then the program works. So I think it is probably fragmentation from code compilation. This makes sense given that we removed long-lived objects in 9.0.0.

@mrklingon So you could:
- stay on 8.2.10, and it will work fine.
- use `mpy-cross` to avoid having to compile your code on-chip: https://learn.adafruit.com/welcome-to-circuitpython/library-file-types-and-frozen-libraries#what-is-an-mpy-file-3117820
- try squeezing the program down a bit. For example, there are many calls to `prt(..., REPL)`. You could make a `prtr()` routine that supplies the second argument as `REPL` .

dhalbert, 2024-04-02:
I'm impressed by what you've been able to squeeze onto this tiny board. You might consider an RP2040 Trinkey if you want more room.

mrklingon, 2024-04-02:
Thanks, I get a kick out of squeezing as much as possible out of The NeoTrinkey. I do have one of the rp2040s - but I really like the NeoTrinkey’s four pixels for output and two pads for input :)

mrklingon, 2024-04-02:
Thanks again! I tried mpy-cross and got it working! As silly as my project may seem, I appreciate getting it to work with current CP, and as a bonus learning to use mpy-cross!!

dhalbert, 2024-04-02:
Great!

dhalbert, 2024-04-02:
I am going to close this, but thank you for the report -- it helps us understand how much the change for 8.x to 9.x affected programs on small boards.

closed 2024-04-02 by dhalbert