adafruit/circuitpython #9089: Can't vary frequency of PWM pin on ItsyBitsy ESP32 - assignment ignored

tyeth, 2024-03-25:
### CircuitPython version

```python
9.0.0 itsybitsy esp32
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2024 Tyeth Gundry
# SPDX-License-Identifier: MIT

#TODO: Add Adafruit IO feeds for BME688.
#TODO: Add scaled output to air quality feed from scaled air sensor (CO2/PM2.5/gas-ohms)
#TODO: swap to using rotary switch inputs ( max_events=1 ) with keypad.keys:
# https://docs.circuitpython.org/en/latest/shared-bindings/keypad/index.html#keypad.Keys

#NO LONGER TRUE, WAS FOR QTPY-S2, NOW ITSYBITSY ESP32 - different pinouts
# the dotstar should use SPI pins, they'll be natively better setup for it.
# Neopixel on A3, POT on A2, dotstar on gpio 35+36, tacho on MISO,  4PST on
# A0(GPIO18),A1(GPIO17),SDA (GPIO7),SCL, mosfet on TX (GPIO5)
#
# NeoPixel on A3 (GPIO8)
# Potentiometer on A2 (GPIO9)
# DotStar on SPI pins:
# Data on GPIO35 (SCK)
# Clock on GPIO36 (MOSI)
# Tachometer Input on MISO (GPIO37)
# 4PST Rotary Switch:
# Pole 1 on A0 (GPIO18)
# Pole 2 on A1 (GPIO17)
# Pole 3 on SDA (GPIO7)
# Pole 4 on SCL (GPIO8)
# MOSFET Output on TX (GPIO5)

import time
import board
from rainbowio import colorwheel
import neopixel
import asyncio
import analogio
import digitalio
import os
import busio
# import adafruit_dotstar as dotstar
import countio
import pwmio
import traceback
import sys
import math
import socketpool
import ssl
import wifi
import adafruit_requests

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError
import sys
CIRCUITPYTHON_AIO_USERNAME=os.getenv("CIRCUITPY_AIO_USERNAME", "YOUR_USERNAME_HERE")
CIRCUITPYTHON_AIO_KEY=os.getenv("CIRCUITPY_AIO_KEY", 'YOUR_KEY')
ADAFRUIT_IO_USERNAME = os.getenv('ADAFRUIT_IO_USERNAME', CIRCUITPYTHON_AIO_USERNAME)
ADAFRUIT_IO_KEY = os.getenv('ADAFRUIT_IO_KEY', CIRCUITPYTHON_AIO_KEY)

# STATUS_NEOPIXEL_PIN = board.NEOPIXEL
# STATUS_NEOPIXEL_NUMPIXELS = 1
# STATUS_NEOPIXEL_BRIGHTNESS = 1.0
# #turn on neopixel power
# NEOPIXEL_POWER = digitalio.DigitalInOut(board.NEOPIXEL_POWER)
[... 303 lines cut ...]
import time
import board
import pwmio

# For the M0 boards: ## now itsybitsy esp32
piezo = pwmio.PWMOut(board.D13, duty_cycle=0, frequency=100, variable_frequency=True)

# For the M4 boards:
# piezo = pwmio.PWMOut(board.A1, duty_cycle=0, frequency=440, variable_frequency=True)

while True:
    for f in (100, 200, 300):
        piezo.frequency = f
        print(f"on at {f}(true hz:{piezo.frequency}")
        piezo.duty_cycle = 65535 // 2  # On 50%
        time.sleep(15)  # On for 1/4 second
        piezo.duty_cycle = 0  # Off
        print(f"off at {f}(true hz:{piezo.frequency}")
        time.sleep(15)  # Pause between notes
    time.sleep(0.5)
```

### Additional information

I can't vary the pwm freqency for a pin. I have duty cycle of 50%(~32k) or 0, and variable_frequency true, but my original frequency of 100 seems to remain despite assigning a new one in the `set_fan_frequency` method.

tyeth, 2024-03-25:
output of small test case:
[image]

tyeth, 2024-03-25:
strangely my other longer code is working now having switched the base frequency from 100 to 300, the auto-adjusting code is now kicking in as expected and my frequency is varying. Not retested the test case, but it's in the same code.py just commented out, no power off of the device has happened yet.

I'm now turning off the itsy (its in a device). I'll look properly tomorrow evening

tyeth, 2024-03-26:
Just to add it's still set at 300Hz and switched on fine this morning.

Leads me to believe it may have been from initialising the pwm pin at a fixed frequency of 100Hz when testing code initially (no variable frequency or False effectively), then soft rebooting and initialising at 100Hz with variable frequency True (no deinit performed?), at which point variable frequency was ignored.

tannewt, 2024-03-26:
> Just to add it's still set at 300Hz and switched on fine this morning with speed varying as expected.
> 
> Leads me to believe it may have been from initialising the pwm pin at a fixed frequency of 100Hz when testing code initially (no variable frequency or False effectively), then soft rebooting and initialising at 100Hz with variable frequency True (no deinit performed?), at which point variable frequency was ignored.
> 
> Power cycling obviously cleared the state / masked the problem

Please post a small test case if you can reproduce it again. Otherwise, please close.