adafruit/circuitpython #9098: AnalogOut NotImplementedError on Adafruit ESP32-S3 Feather

DJDevon3, 2024-03-27:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Board ID:adafruit_feather_esp32s3_4mbflash_2mbpsram
```

### Code/REPL

```python
# This used to work fine in 8.0.0 beta on an Adafruit ESP32-S2 Feather
from analogio import AnalogOut
# Adafruit arcade button with LED. Only supplies power to the LED.
Reset_LED = AnalogOut(board.A0)
Reset_LED.value = 65535

I switched boards to the ESP32-S3 feather working on a different bug and ran into this.
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 79, in <module>
NotImplementedError: 

### Description

This worked fine on the Adafruit ESP32-S2 Feather (with BME280) board in 8.0.0 beta (last time I used it).  I'm using pwmout now instead. I would prefer using analogout just to power the LED.  I have no idea if this is a difference between the S2 & S3.  Analog voltage out from pin A0 should just work?

anecdata, 2024-03-27:
An annotation should probably be added to the Essentials learn guide Analog out page to advise users that there is no DAC on ESP32-S3.

dhalbert, 2024-03-27:
> An annotation should probably be added to the Essentials learn guide Analog out page to advise users that there is no DAC on ESP32-S3 and they should use PWM.

The Essentials Analog Out section was written way before non-atmel-samd boards, and could use some cleanup to be less specific. The board-specific Learn Guides for ESP32-S3 do not include an Analog Out page (or at least the ones I looked at don't).

DJDevon3, 2024-03-27:
I thought Todbot was joking when he mentioned the DAC.  There really is no analogout on this board?  The pinouts reference GPIO pins for A0-A5.  I feel like not being able to do AnalogOut should be mentioned in pinout diagram.  I thought the DAC was only needed for music. 🤦

dhalbert, 2024-03-27:
That is correct, the ESP32-S3 does not have DAC. See https://learn.adafruit.com/adafruit-esp32-s3-feather/pinouts#logic-pins-3115989. The pinout diagram on that same page shows ADC channels on the `An` pins; it does not show DAC.

DJDevon3, 2024-03-27:
Sometimes fast iteration by using only repos and not reading the learn guides after a while has its drawbacks.  This is not an issue, it's user error.  Closing.

closed 2024-03-27 by DJDevon3