adafruit/circuitpython #9103: Automatically run repl.py after code.py finishes without press any key prompt

RetiredWizard, 2024-03-28:
I would like to use `repl.py` to run a "virtual repl" after code.py runs on boards that aren't connected to a host PC and have an alternate input method not supported by the core (serial matrix/i2c keyboard, touch screen, etc) but currently after `code.py` finishes the 
```
Press any key to enter the REPL. Use CTRL-D to reload.
```
message is displayed and unless an input device supported by the core is connected Circuit Python is done.

I think the advantage of this would be that as I understand it `code.py` runs in it's own "memory space"? which get's reset when it's done running so the "virtual repl" would be running in a new clean memory space without any overhead from the original `code.py` program.

tannewt, 2024-03-28:
Why do you need it to run outside of code.py? I'd just make code.py a repl. This library may help with it: https://github.com/adafruit/Adafruit_CircuitPython_Prompt_Toolkit

You could also use `supervisor.set_next_code_file()` to run a subsequent VM with another python file.

RetiredWizard, 2024-03-28:
I'm doing this on something like the cardputer or Hacktablet not connected to a host so all input goes though a virtual keyboard on the tablet or the cardputer keyboard. 

The device boots into PyDOS through an import in code.py. The idea is that if PyDOS is exited or crashes a python coded version of the REPL would be available and started via repl.py. I'll look at set_next_code_file, I've tried using that for launching programs from PyDOS that wouldn't fit in memory along with PyDOS although there was some issue that caused me to go another route. One problem with that approach would be the requirement for a some sort of reset.

I'm thinking if CircuitPython continues moving in the direction of standalone devices with attached input devices, prompting for a key press from a CP core supported input device is going to be problematic in general.

RetiredWizard, 2024-03-28:
The prompt_toolkit looks interesting, I'll have to see if I can simplify my ui abstraction using it.

The DOCS link seems to be broken for the Adafruit_CircuitPython_Prompt_Toolkit library though, fortunately it's not to bad figuring it out from the code.