adafruit/circuitpython #9109: 9.0.x: nRF: AnalogIn.value always returns 0

dhalbert, 2024-03-28:
Starting with early 9.0.0 releases, `AnalogIn` on nordic boards returns zero all the time, no matter what input is applied to the pin. For instance, even with `A0` tied to 3.3V, you see:

```py
>>> import analogio,board
>>> a0 = analogio.AnalogIn(board.A0)
>>> a0.value
0

```
The underlying code has not changed in some time.

We are currently using gcc13. I recompiled a `nordic` build 9.0.x with gcc10, and the problem goes away (!).

dhalbert, 2024-04-01:
Fixed by #9114.

closed 2024-04-01 by dhalbert

kevinjwalters, 2024-04-07:
I was just scratching my head over this in 9.0.2 and was about to try 8.x then found this!

Digital inputs are kaput too https://github.com/adafruit/circuitpython/issues/9154

PR #9114 (2024-04-01): Fix always zero AnalogIn on nordic
- Fixes #9109.

This is quite mysterious. The underlying code hasn't changed in quite a while. nordic `AnalogIn.value` was always returning zero on 9.x.x builds, including some early betas. The problem starts appearing when gcc13 starts being used. Compiling with gcc10 or gcc12 causes the problem to go away.

While debugging, I added a printf which also caused the problem to go away. Adding  tiny delay calls almost anywhere also helped. Eventually I realized it may have been due to these calls acting as an optimization memory fence. Adding `asm volatile("": : [... cut]