adafruit/circuitpython #9111: Pico W UF2 missing core modules on 9.0.2

seanritner, 2024-03-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.2 on 2024-03-28; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
```

### Code/REPL

```python
>>> help("modules")

Please wait a moment while I gather a list of all available modules...

/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
__future__          _weakrefset         getpass             resource
_abc                _xxsubinterpreters  gettext             rlcompleter
_aix_support        _xxtestfuzz         glob                runpy
_ast                _yaml               graphlib            sched
_asyncio            _zoneinfo           grp                 secrets
_bisect             abc                 gzip                select
_blake2             adafruit_board_toolkit hashlib             selectors
_bootsubprocess     aifc                heapq               send2trash
_bz2                antigravity         hmac                serial
_cffi_backend       argparse            html                setuptools
_codecs             array               http                shelve
_codecs_cn          ast                 imaplib             shlex
_codecs_hk          astroid             imghdr              shutil
_codecs_iso2022     asttokens           imp                 signal
_codecs_jp          asynchat            importlib           site
_codecs_kr          asyncio             inspect             six
_codecs_tw          asyncore            io                  smtpd
_collections        atexit              ipaddress           smtplib
_collections_abc    audioop             isort               sndhdr
_compat_pickle      base64              itertools           socket
_compression        bcrypt              jedi                socketserver
_contextvars        bdb                 json                sqlite3
_crypt              binascii            keyword             sre_compile
_csv                binhex              lazy_object_proxy   sre_constants
_ctypes             bisect              lib2to3             sre_parse
_ctypes_test        bitarray            linecache           ssl
_curses             bitstring           locale              stat
_curses_panel       builtins            logging             statistics
_datetime           bz2                 lzma                string
_dbm                cProfile            mailbox             stringprep
_decimal            calendar            mailcap             struct
_distutils_hack     certifi             marshal             subprocess
_elementtree        cffi                math                sunau
_functools          cgi                 mccabe              symtable
_hashlib            cgitb               mimetypes           sys
_heapq              chunk               mmap                sysconfig
_imp                cmath               modulefinder        syslog
_io                 cmd                 multiprocessing     tabnanny
_json               code                mypy                tarfile
_locale             codecs              mypy_extensions     telnetlib
_lsprof             codeop              mypyc               tempfile
_lzma               collections         nacl                termios
_markupbase         colorsys            netrc               textwrap
_md5                compileall          nis                 this
_multibytecodec     concurrent          nntplib             thonny
_multiprocessing    configparser        ntpath              threading
_opcode             contextlib          nturl2path          time
_operator           contextvars         numbers             timeit
_osx_support        copy                opcode              tkinter
_pickle             copyreg             operator            token
_posixshmem         cp_back             optparse            tokenize
_posixsubprocess    cp_launcher         os                  tomli
_py_abc             cp_tracers          paramiko            tomlkit
_pydecimal          crypt               parso               trace
_pyio               cryptography        pathlib             traceback
_queue              csv                 pdb                 tracemalloc
_random             ctypes              pickle              tty
_scproxy            curses              pickletools         turtle
_sha1               dataclasses         pip                 turtledemo
_sha256             datetime            pipes               types
_sha3               dbm                 pkg_resources       typing
_sha512             decimal             pkgutil             typing_extensions
_signal             difflib             platform            unicodedata
_sitebuiltins       dill                platformdirs        unittest
_socket             dis                 plistlib            urllib
_sqlite3            distutils           poplib              uu
_sre                doctest             posix               uuid
_ssl                docutils            posixpath           venv
_stat               ecdsa               pprint              warnings
_statistics         email               profile             wave
_string             encodings           pstats              weakref
_strptime           ensurepip           pty                 webbrowser
_struct             enum                ptyprocess          websockets
_symtable           errno               pwd                 wheel
_sysconfigdata__darwin_darwin espefuse            py_compile          wrapt
_testbuffer         espsecure           pyclbr              wsgiref
_testcapi           esptool             pycparser           xdrlib
_testclinic         faulthandler        pydoc               xml
_testimportmultiple fcntl               pydoc_data          xmlrpc
_testinternalcapi   filecmp             pyexpat             xxlimited
_testmultiphase     fileinput           pylint              xxlimited_35
_thread             fnmatch             queue               xxsubtype
_threading_local    fractions           quopri              yaml
_tkinter            ftplib              random              zipapp
_tracemalloc        functools           re                  zipfile
_uuid               gc                  readline            zipimport
_warnings           genericpath         reedsolo            zlib
_weakref            getopt              reprlib             zoneinfo

Enter any module name to get more help.  Or, type "modules spam" to search
for modules whose name or summary contain the string "spam".
```

### Behavior

there are missing core modules for `board` and `wifi` modules.

### Description

When installing a new UF2 of version 9.0.2, there are missing core modules for `board` and `wifi` modules.

I can see that they are supported in the documentation, but I am unsure on how to solve this issue myself.
[image]

anecdata, 2024-03-28:
Thonny seems to be showing you your host computer. Can you directly connect to the Pico W serial REPL with Thonny, or over web workflow, or over USB with a different serial client?

seanritner, 2024-03-28:
> Thonny seems to be showing you your host computer. Can you directly connect to the Pico W serial REPL with Thonny, or over web workflow, or over USB with a different serial client?

I am still kind of new to this. How do I connect to the Pico W serial REPL specifically? I thought that I was accessing the pico via thonny when I was using the code.py file open. Is that not the same REPL?

seanritner, 2024-03-28:
> Thonny seems to be showing you your host computer. Can you directly connect to the Pico W serial REPL with Thonny, or over web workflow, or over USB with a different serial client?

Wait I may have just figured it out based on your comment. I was able to change the intepretor language from local Python 3 to Circuitpython and it found my device:

```
Adafruit CircuitPython 8.2.10 on 2024-02-14; Raspberry Pi Pico W with rp2040
>>> help ("modules")
__future__        builtins          msgpack           sys
__main__          busio             neopixel_write    terminalio
_asyncio          collections       nvm               time
_bleio            countio           onewireio         touchio
_eve              cyw43             os                traceback
_pixelmap         digitalio         paralleldisplay   ulab
adafruit_bus_device                 displayio         picodvi           ulab.numpy
adafruit_bus_device.i2c_device      errno             pulseio           ulab.numpy.fft
adafruit_bus_device.spi_device      floppyio          pwmio             ulab.numpy.linalg
adafruit_pixelbuf fontio            qrio              ulab.scipy
aesio             framebufferio     rainbowio         ulab.scipy.linalg
alarm             gc                random            ulab.scipy.optimize
analogbufio       getpass           re                ulab.scipy.signal
analogio          gifio             rgbmatrix         ulab.scipy.special
array             hashlib           rotaryio          ulab.utils
atexit            i2cperipheral     rp2pio            usb_cdc
audiobusio        i2ctarget         rtc               usb_hid
audiocore         imagecapture      sdcardio          usb_midi
audiomixer        io                select            uselect
audiomp3          ipaddress         sharpdisplay      vectorio
audiopwmio        json              socketpool        watchdog
binascii          keypad            ssl               wifi
bitbangio         math              storage           zlib
bitmaptools       mdns              struct
bitops            microcontroller   supervisor
board             micropython       synthio
Plus any modules on the filesystem
>>> 
```

seanritner, 2024-03-28:
User error

closed 2024-03-28 by seanritner