adafruit/circuitpython #9113: os.getenv("A_NUMBER") can return an int

dhalbert, 2024-03-28:
If `settings.toml` contains:
```
A_NUMBER = 123
```
`os.getenv("A_NUMBER")`  will return the `int` `123`.

This is non-standard compared with CPython, where `os.getenv()` always returns a string or `None`.

This came up because I was thinking about adding TOML bool support (`true` and `false`) to what we can parse in `settings.toml`.

I'm not sure this is worth fixing, but I think it's possible to fix.

jepler, 2024-03-29:
It's worth noting that the formula `x = int(getenv("VAR"))` will be portable between circuitpython & standard python.

dhalbert, 2025-06-14:
See
https://github.com/adafruit/circuitpython/pull/10422#issuecomment-2972938020:
from @Neradoc:

> If we are completely disallowing ints then it's a breaking change that we need to document clearly with the mention that we only accept strings in our subset of toml (and that they need double quotes).
> https://docs.circuitpython.org/en/latest/docs/environment.html#details-of-the-toml-language-subset
> ```
> • The supported data types are string and integer
> ...
> • Only integers supported by strtol. (no 0o, no 0b, no underscores 1_000, 011 is 9, not 11)
> ```
> Alternatively we could still allow reading ints but return them as strings (and have a better error) but that could be confusing.
> 
> Also, it looks like this preserves `common_hal_os_getenv_int`, but does that mean that the user code can't read some of the environment variables used by Circuitpython ?
> ```toml
> CIRCUITPY_WEB_API_PORT=8000
> CIRCUITPY_DISPLAY_WIDTH=640
> CIRCUITPY_DISPLAY_COLOR_DEPTH=1
> ```

and
https://github.com/adafruit/circuitpython/pull/10422#issuecomment-2972951411
from @jepler:

> I have no HW so I'm not able to test on HW, but under this PR the intent is that internal routines that get integers will continue to work as before.
> 
> It is only intended to change the os.getenv API so that it can no longer return integers (cpython compatibility) and will issue a better error message in the case that a non-quoted value is encountered with a key passed to os.getenv.
> 
> I'm not sure whether I'm in favor of this change myself, because as you say it ought to go through deprecation and does remove the ability to read integer-syntax values from settings.toml into python code without adding it back in some hidden way. It's just that I was still being copied on comments on https://github.com/adafruit/circuitpython/issues/9113 and noticed a recent one go by. I decided to check into the coding required to implement CPython compatibility and this is what I came up with. But I'm not actually itching to see it merged, to be honest.

dhalbert, 2026-02-14:
I am implementing 3 and 4 above:

### **os.getenv(key, default)**

- `os.getenv(key, default)` always returns a string, which is the value on the right side of the `key =` in `settings.toml`. Leading and trailing whitespace is removed. Surrounding double-quotes are removed. The value may be invalid in TOML, but no error is raised. If there is no `settings.toml` or the key is not found, the default value is returned.

### **supervisor.get_setting(key, default)**

`supervisor.get_setting(key, default)` returns a `str`, an `int`, or a `bool`. If the key is not found or `settings.toml` is not present, then the default value is returned. If the value is present but is malformed and cannot be parsed as one of those types, a `ValueError` is raised. (`KeyError` is not raised; this is more like `dict.get()`.

If you have an API suggestion other than the above, I'm interested in hearing it. This needs to be a small API so it can fit on small boards.