adafruit/circuitpython #9115: Adafruit ESP32-S2 Feather ESPIDF Memory & Socket Errors

DJDevon3, 2024-03-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2

Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```

### Code/REPL

```python
import gc
import traceback
import os
import time
import wifi
import ssl
import socketpool
import adafruit_requests

start_mem = gc.mem_free()
print(f"Prior to Script Configurations: {start_mem}") 

# Initialize WiFi Pool (There can be only one pool)
pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

# No login necessary for Public API. Drastically reduced daily limit vs Private
# OpenSky-Networks.org REST API: https://openskynetwork.github.io/opensky-api/rest.html
# All active flights JSON: https://opensky-network.org/api/states/all PICK ONE!
# JSON order: transponder, callsign, country
# ACTIVE transpondes only, for multiple "c822af&icao24=cb3993&icao24=c63923"
TRANSPONDER = "88044d"

# Ensure ALL of these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")
GITHUB_UID = os.getenv("GITHUB_USERNAME")
GITHUB_TOKEN = os.getenv("GITHUB_TOKEN")

MASTODON_SERVER = os.getenv("MASTODON_INSTANCE")  # Set server instance
MASTODON_UID = os.getenv("MASTODON_USERNAME")  # Your server Username
# Requires Steam Developer API key
STEAM_UNUM = os.getenv("STEAM_ID")
STEAM_KEY = os.getenv("STEAM_API_KEY")

# API Polling Rate
# 900 = 15 mins, 1800 = 30 mins, 3600 = 1 hour
SLEEP_TIME = 900

# Set DEBUG to True for full JSON response.
# STREAMER WARNING: Credentials will be viewable
DEBUG = False

def time_calc(input_time):
    """Converts seconds to minutes/hours/days"""
    if input_time < 60:
        return f"{input_time:.0f} seconds"
    if input_time < 3600:
        return f"{input_time / 60:.0f} minutes"
    if input_time < 86400:
        return f"{input_time / 60 / 60:.0f} hours"
    return f"{input_time / 60 / 60 / 24:.1f} days"

def _format_datetime(datetime):
    """F-String formatted struct time conversion"""
    return (
        f"{datetime.tm_mon:02}/"
[... 302 lines cut ...]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 231, in <module>
  File "adafruit_requests.py", line 711, in get
  File "adafruit_requests.py", line 650, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed

Code done running.
```

### Description

Failed in 9.0 so I got out another ESP32-S2 Feather and tried on 8.2.5 with these results.  It seems like this might have been broken on the S2 for quite some time unreported?

This is in 8.2.5 but fails with similar error messages in 9.0.  It works fine on the ESP32-S3 Feather on 9.0.
Referenced issue: https://github.com/adafruit/Adafruit_CircuitPython_ConnectionManager/issues/6

I've been bouncing back and forth between the Adafruit ESP32-S2 & S3 feathers in both 9.0 and 8.2.5 trying to track this issue down.  There's some kind of memory and socket issue here that is beyond my understanding.

DanH requested I slim down my 500 line script to barebones, this is the slimed down version.  I've swapped out a lot of API's that required tokens to use public API's instead so the reviewer won't have to setup as many tokens and logins.  Some will still be required with logins.

It always fails on lines with requests.get(url=source).  I've spent a couple days tracking this thing down with no progress. Time to hand it over to the devs. Good luck.

anecdata, 2024-03-29:
I see `espidf.MemoryError: ` on ESP32-S2 during requests `_get_socket` (8.x.x), even after a fresh reset. My code automatically retries a few times, and if those fail, resets (microcontroller.reset). It may go a few cycles like that, then be successful either after the retries or after one or more resets. I have suspected something variable not quite ready in the ESP-IDF on startup, though sometimes it can happen well into a run. No idea if this is comparable to your situation.

tannewt, 2024-03-29:
This happens. I don't think it is a CP bug.

The ESP-IDF keeps all TLS related buffers in internal RAM and, as a result, the S2 will run out before the S3 does. The S3 has more internal RAM. Use it instead.

DJDevon3, 2024-03-30:
It's possible that's why I stopped using it on the S2 around 8.0 beta.  It would work for a while then crash.  At the time it was suspected that there was an ESP-IDF memory leak.  Very likely why I haven't used this project since.  At the time the ESP32-S2 was far more stable than the S3.  The S3 is now so stable that it works like a dream for most of my projects.  

I can switch to an S3 but that's not really the point.  Yes Anecdata the behavior you're seeing is similar. It's random and the amount of error handling I'd have to write in to catch every possible situation would be mind boggling.

One second I have 2.1MB of memfree then immediately afterward on the next line 1 single call to requests.get throws a memory error.  It doesn't seem like a lack of memory or at least not according to the behavior reported by memfree.  It's not like I have 10KB free... I have around 2MB free when it crashes. The API call I'm making when it crashes most times should only take about 200 bytes...

Is there a way I can see how big the TLS buffer is? Track its size as the script goes?  Why isn't that buffer reported in memfree if it's running out of buffer?  Can we increase the buffer?

somenice, 2024-03-31:
I can report similar behaviour switching from 8 to 9 on Feather ESP32 S2
I can successfully requests.get() an https address but fail with IO send_data() using IO_HTTP

`Adafruit CircuitPython 9.0.2 on 2024-03-28; Adafruit Feather ESP32S2 with ESP32S2`

`Traceback (most recent call last):
  File "code.py", line 313, in <module>
  File "adafruit_io/adafruit_io.py", line 624, in send_data
  File "adafruit_io/adafruit_io.py", line 564, in _post
  File "adafruit_requests.py", line 595, in post
  File "adafruit_requests.py", line 525, in request
  File "adafruit_connection_manager.py", line 264, in get_socket
MemoryError:`

`313.   io.send_data('temperature', temperature, precision=1)`

Please let me know if you need any other information. thanks

DJDevon3, 2024-03-31:
Here's a revised test script that requires no logins for (8.2.5). Anyone with an S2 should be able to run this script.

```py
Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```

```py
import gc
import os
import time
import wifi
import ssl
import socketpool
import adafruit_requests

print("-" * 40)
start_mem = gc.mem_free()
print(f"Prior to Script Config: {start_mem}")

# Initialize WiFi Pool (There can be only one pool)
pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

# All active flights JSON: https://opensky-network.org/api/states/all (PICK ONE!)
TRANSPONDER = "88044d"

# Ensure ALL of these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

# API Polling Rate
# 900 = 15 mins, 1800 = 30 mins, 3600 = 1 hour
SLEEP_TIME = 900

def time_calc(input_time):
    """Converts seconds to minutes/hours/days"""
    if input_time < 60:
        return f"{input_time:.0f} seconds"
    if input_time < 3600:
        return f"{input_time / 60:.0f} minutes"
    if input_time < 86400:
        return f"{input_time / 60 / 60:.0f} hours"
    return f"{input_time / 60 / 60 / 24:.1f} days"

# Requests URL - icao24 is their endpoint required for a transponder
# example https://opensky-network.org/api/states/all?icao24=a808c5
OPENSKY_SOURCE = "https://opensky-network.org/api/states/all?" + "icao24=" + TRANSPONDER

ROCKETLAUNCH_SOURCE = "https://fdo.rocketlaunch.live/json/launches/next/1"

JSON_GET_URL = "https://httpbin.org/get"
STATUS_TEST = "https://httpbin.org/status/"

# Originally attempted to use SVG. Found JSON exists with same filename.
# https://img.shields.io/discord/327254708534116352.svg
ADA_DISCORD_JSON = "https://img.shields.io/discord/327254708534116352.json"

def print_http_status(code, description):
    """Returns HTTP status code and description"""
    if "100" <= code <= "103":
        print(f" | ✅ Status Test: {code} - {description}")
    elif "200" <= code <= "299":
        print(f" | ✅ Status Test: {code} - {description}")
    elif "300" <= code <= "600":
        print(f" | ❌ Status Test: {code} - {description}")
    else:
        print(f" | Unknown Response Status: {code} - {description}")

# All HTTP Status Codes
[... 340 lines cut ...]
Memory Used: 2224
----------------------------------------
Prior to RocketLaunch.Live: 2015456
----------------------------------------
 | Attempting to GET RocketLaunch.Live JSON!
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 244, in <module>
  File "adafruit_requests.py", line 711, in get
  File "adafruit_requests.py", line 650, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
memfree reports 2015456 available prior to the errors.  Not getting Memory Errors here just `Failed SSL handshake` and `sending request failed`.

I've tested it with 1 URL and can hammer on it for hours with no errors.  The errors happen when you attempt to use multiple connections to different servers.  This test script is about as refined as I can make it.  Next will test this same script on the S2 running 9.0

DJDevon3, 2024-03-31:
```py
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```

```
code.py output:
----------------------------------------
Prior to Script Config: 2039824
----------------------------------------
After Script Config: 2038848
Memory Used: 976
----------------------------------------
Prior to WiFi: 2038848
✅ Wifi!
----------------------------------------
After Wifi: 2038848
Memory Used: 0
----------------------------------------
Prior to Open-Sky Public API: 2038848
----------------------------------------
 | Attempting to GET OpenSky-Network Single Public Flight JSON!
 | ✅ OpenSky-Network Public JSON!
This flight has no active data or you're polling too fast.
Public Limits: 10 second max poll & 400 weighted calls daily
✂️ Disconnected from OpenSky-Network API
----------------------------------------
After Open-Sky: 2034848
Memory Used: 4000
----------------------------------------
Prior to HTTPBin.org: 2034848
----------------------------------------
 | GET JSON: https://httpbin.org/get
Traceback (most recent call last):
  File "code.py", line 200, in <module>
  File "/lib/adafruit_requests.py", line 715, in get
  File "/lib/adafruit_requests.py", line 656, in request
  File "/lib/adafruit_requests.py", line 520, in _get_socket
MemoryError: 

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
In 9.0 it fails much sooner, on the 2nd API, reliably, every time.

tannewt, 2024-04-01:
> One second I have 2.1MB of memfree then immediately afterward on the next line 1 single call to requests.get throws a memory error. It doesn't seem like a lack of memory or at least not according to [...]

memfree is telling you how much is available for MP heap. It isn't accounting for the IDF memory requirements. TLS related buffers aren't put in PSRAM so there is much less available.

> Is there a way I can see how big the TLS buffer is? Track its size as the script goes? Why isn't that buffer reported in memfree if it's running out of buffer? Can we increase the buffer?

It is a function of the IDF settings. I believe it is lowered already. Some of it changed with the 4 -> 5 transition and I'm not sure there is a lot we can do. To make more space for it we'd need to allow TLS buffers on PSRAM or optimize our internal RAM use. It isn't a simple fix.

DJDevon3, 2024-04-02:
I didn't think it would be since this one seems to have continually affected the S2 since its debut.  memfree reporting everything is good to crashing without warning is a bit jarring.  

Having a TLS buffer error would be welcome so at least users will know it's TLS buffer related and not due to lack of PSRAM.  After looking into it more yesterday and testing it in different ways all HTTP sites can be sequentially loaded without issue.  It's specifically HTTPS responses that causes this one which points to a TLS issue but the error never says that. The general MemoryError had me thinking there was something physically wrong with the memory on my board at first.  It's not a nice error to get.  Requesting more descriptive error for TLS buffer limit reached (if possible).

tannewt, 2024-04-02:
> Requesting more descriptive error for TLS buffer limit reached (if possible).

It isn't very easy to do because the underlying `ESP_ERR_NO_MEM` is generated by the IDF. We don't know how big the allocation was or what it was for because we didn't attempt it ourselves. That's why it is `espidf.MemoryError` vs the built in `MemoryError` which usually includes the size that failed to allocate.

DJDevon3, 2024-05-15:
This issue is now fixed as of 9.0.4 stable with 9.x bundle as of `adafruit-circuitpython-bundle-9.x-mpy-20240514`  The test script here finished flawlessly first attempt.

dhalbert, 2024-05-15:
Fantastic! I'll close.

closed 2024-05-15 by dhalbert