adafruit/circuitpython #9119: Web Workflow Serial screen on Full Editor page crashing on non-focus input

RetiredWizard, 2024-03-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
9.0.2
9.1.0 beta 0
```

### Code/REPL

```python
a = ''
while a != 'q':
    a = input('_$ ')
```

### Behavior

```>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
_$ Traceback (most recent call last):
  File "code.py", line 3, in <module>
EOFError: 

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

In the Serial screen on the Web Workflow Full Editor site, if you use the Reset button to perform a soft reboot and then don't click within the serial screen, the serial screen won't have input focus but it still attempts to respond to keystrokes so if you press the space bar/enter key/or any key, `code.py` (or the REPL if no `code.py` exists) will attempt to read the input but since the window doesn't have focus there's actually no input there and CircuitPython errors out with an EOFError

### Additional information

Once you know what's going on this isn't really a problem but when it first happened to me, the `code.py` file I was running didn't handle the error well and the terminal behavior was very confusing.

dhalbert, 2024-03-30:
@makermelissa I'm not sure if this is a core issue or a https://github.com/circuitpython/web-editor issue.

RetiredWizard, 2024-03-30:
Sorry, I should have realized there was a web-editor repo. I should have opened this there. I'll close this one and open a new issue in the web-editor repo.

closed 2024-03-30 by RetiredWizard