adafruit/circuitpython #9120: M5Stack Dial with ESP32S3 memory seems not right

tonyhuang1962, 2024-03-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; M5Stack Dial with ESP32S3
```

### Code/REPL

```python
import gc       
gc.collect()
start_mem = gc.mem_free()
print( "Available memory: {} bytes".format(start_mem) )
```

### Behavior

Available memory: 162912 bytes

### Description

This memory seems not right for  M5Stack Dial with ESP32S3 ?

tonyhuang1962, 2024-03-30:
When I run Adafruit CircuitPython 9.0.0 with M5Stack Dial with ESP32S3 to load Image Bitmap using adafruit_imageload.load, it always give me error  like 'MemoryError: memory allocation failed, allocating 57600 bytes', thus I checked memory at the beginning of my progtram. To my surprise, it showed 'Available memory: 162912 bytes'. Is this correct?

tannewt, 2024-04-01:
This can happen due to memory fragmentation. The M5Stack doesn't have external PSRAM so ~150k seems reasonable for the mem_free() value.

tannewt, 2024-04-01:
It is possible that the TLSF allocation under the hood isn't performing very well when running out of memory. That's why I marked this as a bug (not support.)